esp8266/Arduino #504: Problems with WiFi.softApConfig

slaw54, 2015-07-02:
Using latest updates the code below sets the IP to 1.0.0.0 the first digit changes dependant on the last digit of the config IP i.e.: 192.168.1.8 gives 8.0.0.0.  The AP works fine with no config and the default IP. This is with Arduino 1.6.4,  Windows or Mac and with esp 01 any suggestions appreciated.

#include "ESP8266WiFi.h"
void setup() {
  IPAddress Ip = (192, 168, 3, 1);
  IPAddress NMask = (255, 255, 255, 0);
  WiFi.softAPConfig(Ip, Ip, NMask);
  WiFi.softAP("ESP8266_AP", "12345678", 6);
}

Accesse, 2015-08-11:
Try:

#include "ESP8266WiFi.h"
void setup() {
IPAddress Ip(192, 168, 3, 1);
IPAddress NMask = (255, 255, 255, 0);
WiFi.softAPConfig(Ip, Ip, NMask);
WiFi.softAP("ESP8266_AP", "12345678", 6);
}

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004