esp8266/Arduino #506: WiFiClient.connect slow the first time

torntrousers, 2015-07-03:
I wonder if there could be a bug in WiFiClient.connect that makes it slow to connect the first time its used?

I'm trying to have a battery powered ESP wakeup from deep sleep connect to a wifi access point, do an http get to publish some data, and then go back to sleep. And do that as quickly as possible to conserve battery power. Have it down to around 2 seconds now and adding in some printlns of millis() at various places in the code shows most of that time is taken up  by the WiFiClient.connect call which takes about 1649 milliseconds.

That 1649ms time is fairly consistent too, occasionally its higher but often exactly 1649, which seems interesting.

I've tried running the ESP at both 80Mhz and 160 Mhz and that doesn't change that connect time significantly so its not cpu thats causing that time.

To avoid internet delays i've set up a test service on a local pc on the same wifi network, so its not some internet delay, and a Java http client running on a different pc on the same wifi network can connect to it in just a millisecond or so so its not caused by a slow network or wifi.

Also doing multiple WiFiClient.connect's on the ESP its only the first time that its >= 1649  and subsequent ones take just 12 milliseconds or so. This is using an IPAddress too not a domain name so i don't think its some dns caching thing.

Any ideas what could be causing this or what i could try to reduce that first connect time?

chaeplin, 2015-07-04:
check this -->  https://github.com/esp8266/Arduino/issues/261

Links2004, 2015-07-04:
the most time takes the DHCP request. switching to static IP speed it up.
if possible also provide the BSSID mac and the channel to begin.

cimba007, 2015-07-04:
Maybe your first connect makes a DNS resolve call which might take time too. Try to connect to IP-Address directly.

chaeplin, 2015-07-04:
Static ip + channel + BSSID :
first time :  takes 3.2 second
with deep sleep reset : 1.1 second

torntrousers, 2015-07-04:
Thanks for the comments. Some follow up replies:
- this is already using a static ip not dhcp. I've not tried channel and bssid, will look into that out of interest, but i don't think its relevent - the 1649 time is just the time around the WiFiClient.connect, its already connected to the access point and gets connected with the static ip in around 200 milliseconds which seem ok, the 1649ms WiFiClient.connect is after that.
- this is using the ip address of the remote service not a domain name so its not a dns resolve being slow.

torntrousers, 2015-07-05:
It turns out this is caused by not calling WiFi.status() to see if its connected before starting the WiFiClient.connect. More about that [here](http://www.esp8266.com/viewtopic.php?p=22449#p22449).

closed 2015-07-05 by torntrousers

cimba007, 2015-07-08:
@chaeplin: Can u provide a quick and dirty example how to **pre-set** bssid and channel?

torntrousers, 2015-07-09:
Looks like you can specify them on the WiFi.begin call. Here's the function in ESP8266WiFi.h:
```
    /**
     * Start Wifi connection
     * if passphrase is set the most secure supported mode will be automatically selected
     * @param ssid const char*          Pointer to the SSID string.
     * @param passphrase const char *   Optional. Passphrase. Valid characters in a passphrase must be between ASCII 32-126 (decimal).
     * @param bssid uint8_t[6]          Optional. BSSID / MAC of AP
     * @param channel                   Optional. Channel of AP
     * @return
     */
    int begin(const char* ssid, const char *passphrase = NULL, int32_t channel = 0, uint8_t bssid[6] = NULL);
```