esp8266/Arduino #507: base64_encode and base64_decode

rakeshpai, 2015-07-04:
It appears that [the ROM already has these functions](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/tools/sdk/ld/eagle.rom.addr.v6.ld#L99-L100). A nice clean wrapper would make using it very easy.

In [this thread in the forum](http://www.esp8266.com/viewtopic.php?f=24&t=3289#p18751), cal says that the signature of base64_encode is
```c
extern unsigned char * base64_encode(const unsigned char *src, size_t len, size_t *out_len);
```
However, using this doesn't seem straightforward, for reasons mentioned in the forum post.

I'm still just getting my feet wet, so I don't have the chops to do this myself, sorry. However, on the surface, it seems like it should be rather easy to do? I'm guessing it could be made to look similar to `String sha1(String sha1)` in Hash.h. Maybe it even belongs in Hash.h?

tim-in-oakton, 2015-07-14:
How about something like this if you are in a hurry:
//start with what are Bbase64 charset members - will use this to make a string from our array of sextets
static const char PROGMEM b64chars[] =
  "ABCDEFGHIJKLMNOPQRSTUVWXYZabcdefghijklmnopqrstuvwxyz0123456789+/";

String Hash_base64( uint8_t *in, int hashlength) {
  int i, out;
  char b64[80];  // working byte array for sextets....
  String base64;
  for (i = 0, out = 0 ;; in += 3) { // octets to sextets
    i++;
    b64[out++] =   b64chars[in[0] >> 2];

    if (i >= hashlength ) { // single byte, so pad two times
      b64[out++] = b64chars[((in[0] & 0x03) << 4) ];
      b64[out++] =  '=';
      b64[out++] =  '=';
      break;
    }

    b64[out++] = b64chars[((in[0] & 0x03) << 4) | (in[1] >> 4)];
    i++;
    if (i >= hashlength ) { // two bytes, so we need to pad one time;
      b64[out++] =  b64chars[((in[1] & 0x0f) << 2)] ;
      b64[out++] =  '=';
      break;
    }
    b64[out++] = b64chars[((in[1] & 0x0f) << 2) | (in[2] >> 6)];
    b64[out++] =   b64chars[in[2] & 0x3f];

    i++;
    if (i >= hashlength ) { // three bytes, so we need no pad - wrap it;
      break;
    }
  } // this should make b64 an array of sextets that is "out" in length
  b64[out] = 0;
  base64 = b64;
  return base64;
}

igrr, 2016-06-06:
libb64 is now part of the core.

closed 2016-06-06 by igrr

Humancell, 2017-03-04:
Can someone post a complete example of how to use libb64 in a ESP app?

What do I include?  How can I access the encode() function?  I can't seem to get this working ...

vicnevicne, 2017-03-04:
Hi, humancell, 
Please refrain from cross-posting. You seem to have an issue in #1758 but asking the same question here and "a complete ESP example" in #1727 will not motivate developers to help you.
This issue tracking system is meant for problems and bugs with the Arduino environment. If you really need help of a first ESP example, please head to the esp8266.com forums where many people will be able to help you.
KR,

Humancell, 2017-03-06:
Sorry ... I had found several people asking the same question as this post, and although this issue was closed there was no answer given for the original question ... instead the poster was pointed at a different library.

I'll refrain from cross posting in the future, but was not clear if the closed issues generate the same notifications.  In addition, I'll post an answer to the original posters question, as I realized my error was the same as his.

To use the Base64 library included, if you don't want to use libb64, you can do the following:

```
#include "base64.h"

// base64-encoding of Aladdin:OpenSesame, or QWxhZGRpbjpPcGVuU2VzYW1l
String basicAuth = "Aladdin:OpenSesame";
Serial.print("Base64: ");
Serial.println(base64::encode(basicAuth));
Serial.print("Match:  ");
Serial.println("QWxhZGRpbjpPcGVuU2VzYW1l");

```
The issue that both the original poster and myself did was to use the '.' instead of '::' when referencing the class method.  The error message was just throwing me off.  Hopefully this will assist someone in the future.

Sorry for the extra noise ...