esp8266/Arduino #510: ESP-01 IS RANDOMLY REBOOTED

kumarmohan7, 2015-07-06:
MY ESP -01 IS RANDOMLY REBOOTED

I USING PubSubClient AND MQTT 

MY CODE IS 

#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ATT_IOT.h>                            //AllThingsTalk IoT library

const char* ssid     = "*****";
const char* password = "******";

char deviceId[] = "mf933FcGydT1awvrRwQOVIl"; // Your device id comes here
char clientId[] = "kumarmohan7"; // Your client id comes here;
char clientKey[] = "o2yi2qdo1xo"; // Your client key comes here;

ATTDevice Device(deviceId, clientId, clientKey);            //create the object that provides the connection to the cloud to manager the device.
char httpServer[] = "api.smartliving.io";                   // HTTP API Server host                  
char mqttServer[] = "broker.smartliving.io";		    // MQTT Server Address

// Define PIN numbers for assets
int temp=0;
int humi=1;
int light=2;
int sw1 = 3; 
int sw2 = 4; 
int sw3 = 5; 
int sw4 = 6; 
int sw5 = 7; 
int sw6 = 8; 
int sw7 = 9; 
int sw8 = 10; 

String inputString = "";         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete

//required for the device
void callback(char* topic, byte* payload, unsigned int length);
WiFiClient wificlient; 
PubSubClient pubSub(mqttServer, 1883, callback, wificlient);

void setup()
{
  Serial.begin(9600);
  inputString.reserve(200);
  							            // init serial link for debugging
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  delay(1000);                                          	
  
  if(Device.Connect(&wificlient, httpServer))					           
  {
//    Device.AddAsset(0, "Temperature", "Temperature", false, "integer");  
//    Device.AddAsset(1, "Humidity", "Humidity", false, "integer"); 
//    Device.AddAsset(2, "Light", "Light", false, "integer");   
//    Device.AddAsset(3, "SW-1", "SW-1", true, "boolean");
//    Device.AddAsset(4, "SW-2", "SW-2", true, "boolean");
//    Device.AddAsset(5, "SW-3", "SW-3", true, "boolean");
//    Device.AddAsset(6, "SW-4", "SW-4", true, "boolean");
//    Device.AddAsset(7, "SW-5", "SW-5", true, "boolean");
//    Device.AddAsset(8, "SW-6", "SW-6", true, "boolean");
//    Device.AddAsset(9, "SW-7", "SW-7", true, "boolean");
//    Device.AddAsset(10, "SW-8", "SW-8", true, "boolean");
    Device.Subscribe(pubSub);						           
  }
  else 
    while(true);                                                                
}
unsigned long time;
int i=0;	
String rx;						       
void loop()
{
  Serial.println("REQ");
  serialEvent();
  if (stringComplete) 
  {
    inputString.replace("\r\n","");
    Serial.println(inputString);
//    Serial.println(", OK");
    Device.Send(String(inputString), 0);                       
    Device.Send(String(inputString), 1);
    Device.Send(String(inputString), 2);
    inputString = "";
    stringComplete = false;
  }
  Device.Process(); 
  delay(2000);
}

// Callback function: handles messages that were sent from the iot platform to this device.
void callback(char* topic, byte* payload, unsigned int length) 
{ 
  String msgString; 
  {                                                           //put this in a sub block, so any unused memory can be freed as soon as possible, required to save mem while sending data
	char message_buff[length + 1];	                      //need to copy over the payload so that we can add a /0 terminator, this can then be wrapped inside a string for easy manipulation
	strncpy(message_buff, (char*)payload, length);        //copy over the data
	message_buff[length] = '\0';		              //make certain that it ends with a null			
		  
	msgString = String(message_buff);
	msgString.toLowerCase();			      //to make certain that our comparison later on works ok (it could be that a 'True' or 'False' was sent)
  }
  int* idOut = NULL;
  {	                                                      //put this in a sub block, so any unused memory can be freed as soon as possible, required to save mem while sending data
	int topicLength = strlen(topic);
	int pinNr = Device.GetPinNr(topic, strlen(topic));
//	Serial.print("Payload: ");                            //show some debugging
//	Serial.println(msgString);
//	Serial.print("Pin: ");
//	Serial.println(pinNr);
        
        
	if (pinNr == sw1)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw1 Off");		      
            idOut = &sw1;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw1 On");
            idOut = &sw1;
	  }
	}

        if (pinNr == sw2)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw2 Off");		      
            idOut = &sw2;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw2 On");
            idOut = &sw2;
	  }
	}

        if (pinNr == sw3)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw3 Off");		      
            idOut = &sw3;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw3 On");
            idOut = &sw3;
	  }
	}

        if (pinNr == sw4)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw4 Off");		      
            idOut = &sw4;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw4 On");
            idOut = &sw4;
	  }
	}

        if (pinNr == sw5)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw5 Off");		      
            idOut = &sw5;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw5 On");
            idOut = &sw5;
	  }
	}

        if (pinNr == sw6)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw6 Off");		      
            idOut = &sw6;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw6 On");
            idOut = &sw6;
	  }
	}

        if (pinNr == sw7)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw7 Off");		      
            idOut = &sw7;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw7 On");
            idOut = &sw7;
	  }
	}

        if (pinNr == sw8)
	{
	  if (msgString == "false") 
          {
//            digitalWrite(RELAY, LOW);
            Serial.println("sw8 Off");		      
            idOut = &sw5;		                        
	  }
	  else if (msgString == "true") 
          {
//	    digitalWrite(RELAY, HIGH); 
            Serial.println("sw8 On");
            idOut = &sw8;
	  }
	}
  }
  if(idOut != NULL)                                           //Let the iot platform know that the operation was succesful
    Device.Send(msgString, *idOut);    
}
void serialEvent() {
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read(); 
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '\n') {
      stringComplete = true;
    } 
  }
}

canadaduane, 2015-07-22:
I'm sorry, but this is totally unreadable. Use triple backticks around your code if you'd like to post it here. (Google "github flavored markdown" to see what I'm talking about if this isn't clear).

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004