esp8266/Arduino #513: ADC forcing WIFI to disconnect and lose SSID

iisfaq, 2015-07-06:
I have been following the ADC thread at https://github.com/esp8266/Arduino/issues/338

I implemented the change to the analaogRead() function

    extern int __analogRead(uint8_t pin) {
        if (pin == 17){
           return 0xffff & system_adc_read();
        }
       return digitalRead(pin) * 1023;
    }

This gave me much more stable ADC readings when ADC is pinned to GND or 3.3V

But I have now found another issue.

My code is using the ADC along with an ACS712 Current Sensor to measure current in a 240V supply.

The code is pretty simple

Basically Connect to WIFI, loop around calling analogRead and once per minute upload this data to a web site, and just display it every second to the serial port.

When I am calling analogRead I get this type of output

Connecting to CROWE
......
WiFi connected
IP address: 
192.168.2.128
Interval=1, WiFi.status=WL_CONNECTED=3
Interval=2, WiFi.status=WL_CONNECTED=3
Interval=3, WiFi.status=WL_CONNECTED=3
Interval=4, WiFi.status=WL_CONNECTED=3
Interval=5, WiFi.status=WL_DISCONNECTED=6
Interval=6, WiFi.status=WL_DISCONNECTED=6
Interval=7, WiFi.status=WL_DISCONNECTED=6
Interval=8, WiFi.status=WL_DISCONNECTED=6
Interval=9, WiFi.status=WL_DISCONNECTED=6
Interval=10, WiFi.status=WL_DISCONNECTED=6
Interval=11, WiFi.status=WL_DISCONNECTED=6
Interval=12, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=13, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=14, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=15, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=16, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=17, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=18, WiFi.status=WL_NO_SSID_AVAIL=1

if I simply do not call analogRead(A0) in the loop it stays connected forever.

Here is the code I am using

```
#include <ESP8266WiFi.h>

const char* ssid     = "CROWE";
const char* password = "abcd";

const char* host = "192.168.2.122";

String Statuses[] =  { "WL_IDLE_STATUS=0", "WL_NO_SSID_AVAIL=1", "WL_SCAN_COMPLETED=2", "WL_CONNECTED=3", "WL_CONNECT_FAILED=4", "WL_CONNECTION_LOST=5", "WL_DISCONNECTED=6"};
unsigned long lastMillis = 0;
unsigned long lastMillis2 = 0;
int IntervalInSeconds = 1;
int Errors = 0;
int min = 1024;
int max = -1024;
int samples = 0;
void setup() {

  Serial.begin(115200);
  Serial.println("ESP8266 Power Monitor");

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();

  Connect();
}

void Connect()
{
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  int Attempt = 0;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Attempt++;
    Serial.print(".");
    if (Attempt == 50)
    {
      Serial.println();
      Serial.println("Could not connect to WIFI");
      return;
    }
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop()
{
  yield() ; // Allow WIFI to do stuff

  unsigned long CurrentMillis = millis();
  if (CurrentMillis - lastMillis2 > 3)
  {
    lastMillis2 = CurrentMillis;

    double sensorValue = 0;
    for (int reading = 0; reading < 5; reading++)
    {
      sensorValue += analogRead(A0);
    }
    sensorValue /= 5;
    if ((int)sensorValue < min)
      min = (int)sensorValue;
    if ((int)sensorValue > max)
      max = (int)sensorValue;
    samples++;
  }

  if (CurrentMillis - lastMillis > 900)
  {
    lastMillis = CurrentMillis;

    Serial.print("Interval=");
    Serial.print(IntervalInSeconds);
    Serial.print(", WiFi.status=");
    Serial.print(Statuses[WiFi.status()]);

    if (IntervalInSeconds >= 60)
    {
      IntervalInSeconds = 1;

      Serial.print("connecting to ");
      Serial.println(host);

      if (WiFi.status() != WL_CONNECTED)
        Connect();

      if (WiFi.status() == WL_CONNECTED)
      {
        // Use WiFiClient class to create TCP connections
        WiFiClient client;
        const int httpPort = 8080;
        if (!client.connect(host, httpPort)) {
          Serial.println("connection failed");
          return;
        }

        // We now create a URI for the request
        String url = "/Power/?Watts=";
        url += "SOMENUMBER";
        url += "&DeviceID=HomeTheatre";

        Serial.print("Requesting URL: ");
        Serial.println(url);

        // This will send the request to the server
        client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                     "Host: " + host + "\r\n" +
                     "Connection: close\r\n\r\n");
        delay(50);

        // Read all the lines of the reply from server and print them to Serial
        while (client.available()) {
          String line = client.readStringUntil('\r');
          Serial.print(line);
        }
        Serial.println();
      }
      else
      {
        Serial.print("Error Not Connected To WIFI, Error Count: ");
        Serial.println(++Errors);
      }
    }
    else
      Serial.println();

    IntervalInSeconds = IntervalInSeconds + 1;
    min = 1024;
    max = -1024;
    samples = 0;
  }
}
```

iisfaq, 2015-07-06:
Interestingly I have taken my laptop and ESP8266 hardware to work and it is working fine from there. 

So the same code just a different access point!

I have not seen any disconnecting of the WIFI at all.

The AP is the only difference.

drmpf, 2015-07-06:
Also interesting, I had exactly this problem last night. First time sending readings via WiFi, running ESP8266 as a server and in Station (STS) mode.

In my application I am using my Android pfodApp (www.pfod.com.au) to collect and plot the position of the garage door by reading the values from an accelerometer ADXL335 and sending them to my mobile.  The completed code will show the position on a slider so I can monitor the door via the internet. 

What I found was that if I tried to read the ADC at 1mS intervals I could not connect to the ESP8266.
2mS intervals same issue
10mS intervals works fine.

While 10mS works for me it seem woefully slow for an 80Mhz chip. 

So for the moment, I am using 10mS intervals, but would really like some polling version of ADC read so that I could start a conversion and then keep going and come back later and pick up the result.  

See http://www.forward.com.au/pfod/ArduinoProgramming/pollingAnalogReadLibrary/index.html
for the library I have written to do this for Arduino boards, Uno/Mega

iisfaq, 2015-07-07:
drmpf are you using the same modified analogRead() code that I was using?

````
extern int __analogRead(uint8_t pin) {
   if (pin == 17){
      return 0xffff & system_adc_read();
   }
   return digitalRead(pin) * 1023;
}
````

drmpf, 2015-07-07:
yes exactly the same code

iisfaq, 2015-07-07:
Back at home now and exactly the same problem occurring again.

WiFi connected
IP address: 
192.168.2.116
Interval=1, WiFi.status=WL_CONNECTED=3
Interval=2, WiFi.status=WL_CONNECTED=3
Interval=3, WiFi.status=WL_CONNECTED=3
Interval=4, WiFi.status=WL_CONNECTED=3
Interval=5, WiFi.status=WL_DISCONNECTED=6
Interval=6, WiFi.status=WL_DISCONNECTED=6
Interval=7, WiFi.status=WL_DISCONNECTED=6
Interval=8, WiFi.status=WL_DISCONNECTED=6
Interval=9, WiFi.status=WL_DISCONNECTED=6
Interval=10, WiFi.status=WL_DISCONNECTED=6
Interval=11, WiFi.status=WL_DISCONNECTED=6
Interval=12, WiFi.status=WL_NO_SSID_AVAIL=1
Interval=13, WiFi.status=WL_NO_SSID_AVAIL=1

iisfaq, 2015-07-07:
I adjusted my analogRead code to the following (snippet)

~~~~
  unsigned long CurrentMillis = millis();
  if (CurrentMillis - lastMillis2 > 10)
  {
    double sensorValue = 0;
    for (int reading = 0; reading < 5; reading++)
    {
      sensorValue += analogRead(A0);
      delay(3);
    }
    sensorValue /= 5;

    lastMillis2 = CurrentMillis;
    sensorValue += analogRead(A0);
    if ((int)sensorValue < min)
      min = (int)sensorValue;
    if ((int)sensorValue > max)
      max = (int)sensorValue;
    samples++;
  }

~~~~

It is not dropping and staying connected and giving me about 50 samples per second still. Which may be enough to get the values I am looking for.

It has been oK for 30 minutes now.

Testato, 2015-07-08:
Yiur analogog patch is the old way. In the development .json you receive de development version of the core, whit a completeky rewrited analog section and new Apu for correctly neasure A0 or Vdd voltage.
It is solved also the wifi disconnection problem

iisfaq, 2015-07-08:
Hi Testato

Sorry I take it English is a 2nd or 3rd language for you.

I really could not understand what you wrote.

It appears there is a new version along with a re-written analog function that resolves the issue.

But I do not know how I would use this.

Testato, 2015-07-08:
Third language plus cellphone autocotrection :-)

Are you installed this core whit the .json method writed in readme.md ?
If yes, there is a development json link for install the actual core.
Search it in the issue opened and closed section

iisfaq, 2015-07-08:
No worries, I only know English..... You are far better than me!!! 

I am using this for the board

http://arduino.esp8266.com/package_esp8266com_index.json

so I take it there is a different url that I should use which is newer.

Any chance you know the URL already?

igrr, 2015-07-08:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

Note that this version is often broken for various reasons.

Testato, 2015-07-08:
@igrr maybe it is time to update the stable version ?
There is to many difference actually from develop and stable

Testato, 2015-07-08:
pull request on readme update https://github.com/esp8266/Arduino/pull/520

closed 2015-07-23 by igrr

k-a-z-u, 2016-10-28:
Today I had exactly the same issue as mentioned by " iisfaq commented on 7 Jul 2015 "
using Stable version 2.3.0 and system_adc_read()
Am I missing something?

krzychb, 2016-10-28:
Hi @k-a-z-u,

Wi-Fi connection may get lost depending on how often you sample (read) analog input.
For quantification of "how often" please see https://github.com/krzychb/EspScopeA0/tree/master/Bravo

Krzysztof