esp8266/Arduino #515: Android + esp8266 softap + GET request.

zoutepopcorn, 2015-07-07:
When connecting from a android device to the ESP8266, sometimes a get request is not responing. Especially when it connect the first time. I have to forget the network and reconnect again. 

It is the same issue is also in the older original SDK and was fixed in 1.1.1. See:
http://www.esp8266.com/viewtopic.php?f=6&t=3422&hilit=softap+arduino

They have a fix for older firmware, by putting this before main():
void user_rf_pre_init(void)
{
   system_phy_set_rfoption(2);
}

I'm new to the Arduino esp8266, can anybody tell me howto debug / implement this?

Links2004, 2015-07-07:
the ```user_rf_pre_init``` function is existing here:
https://github.com/esp8266/Arduino/blob/ea302aab05480ad36c076b618abb642b1eb1893b/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_phy.c#L243
you can do a locale change.

the files con normally find on the pc here:
```
Windows: C:\Users\[username]\AppData\Roaming\Arduino15\packages\esp8266\hardware
Linux: /home/[username]/.arduino15/packages/esp8266/hardware
```

but i think you have a power supply problem.

```system_phy_set_rfoption(2);``` from doku:
[image]

zoutepopcorn, 2015-07-08:
Thank you. Now I have to find a way to get it in the code..

I dont think its a power supply issue. I have tried it on esp8266 with good power supply. Also at nodemcu with usb power and my ipad charger... The problem is only on android.

zoutepopcorn, 2015-07-13:
Links2004, thanks so much. 
I edit the file: core_esp8266_main.cpp
void user_init(void) {
    system_phy_set_rfoption(2);
    struct rst_info *rtc_info_ptr = system_get_rst_info();
    ...... 
}

Now the get request get a response the first time. It's a perfect solution for me. But I do think it has to be fixed in the firmware. Softap + Android is unstable in the release 1.6.5-804-g2d340c7 I tested.. 

With my program skills I don't know howto find a clean fix for the firmware.... Maybe some other people with Android devices can test it.

I was on S5 (android 4.4) and Moto G (android 5.0.2).

closed 2015-11-27 by Links2004