esp8266/Arduino #517: 80% missed OTA with the latest changes

hallard, 2015-07-07:
Hi guys,
I just pulled latest repo, just saw OTA has changed for better integration with IDE, that's pretty cool.
So I migrated my program to use new version of OTA. I was using old @igrr one on command line.
What I noticed despite the fact that I do not see ESP port in IDE (I'm on windows and already add lot's of problem with bonjour with Yun compatible board called Dragino, so I won't spend time on this, Command line + up array + enter to run python OTA script is okay for me),
Anyway, my return on this new version is that I miss more OTA than the old version. With old version I was missing about 1/5 OTA and with this version I'm missing about 4/5 (lot errors with "lmac.c" seen on Serial Debug)

I succeeded to decrease OTA errors reducing chunk size from 4096 to 1024 in python script and adding 5ms delay between each chunk. Seems better for me, far from ideal but better here are my changes. 

    chunk = f.read(1024)
      if not chunk: break
      try:
        sys.stderr.write('.')
        sys.stderr.flush()
        connection.sendall(chunk)
        # wait 5ms
        time.sleep (0.005)

May be a chunk test/valid send from the node after each chunk received to tell python script to continue if okay or send back the chunk could solve more problems.

Anyway, excellent job guys, just my 2 cents ;-)

Cheers
Charles

me-no-dev, 2015-07-07:
yeah... made things faster and hit the error. That lmac.c thing is the only problem with OTA as of now.
I want OTA to be more synchronous also, even though update time will go up, less packets will go through and we might not get into this problem.
Python is not the issue here. The delay and packet size just help a bit, but not really a solution.
Had the same exact problem using the web update. Did not have it when i took the esp far from the AP, even though update time went up quite. We are consuming the incoming data fast enough as I have tested it and see the buffer emptied quite often.
Will try synchronous and see how long it will take and will I hit the lmac.c again

hallard, 2015-07-08:
Humm, good news to see ESP is fast enough  ;-) 

Oh sorry to mention, I also made a change in the demo sketch because when OTA fails, we're unable to try another OTA until we push reset button.

Since `WiFiUDP::stopAll();` is called just before update, we need to enable back UDP in case of failure with `OTA.begin(aport);` 

    if(Update.end()){
        client.println("OK");
        Serial.printf("Update Success: %u\nRebooting...\n", millis() - startTime);
        ESP.restart();
     } else {
        Update.printError(client);
        Update.printError(Serial);
    }
becomes

    if(Update.end()){
        client.println("OK");
        Serial.printf("Update Success: %u\nRebooting...\n", millis() - startTime);
        ESP.restart();
     } else {
        Update.printError(client);
        Update.printError(Serial);
        OTA.begin(aport);
    }

me-no-dev, 2015-07-08:
I actually do not stop it at all
and to be honest the only fail is with lmac.c and that triggers reset

hallard, 2015-07-08:
@me-no-dev correct!!
in case of lmac error it reboot so no problem, but in case of checksum error (rarely) it does not, so I'm stuck to push reset button to try again OTA ;-)
of course we can also change `OTA.begin(aport);` by a `ESP.restart();` to force reboot

igrr, 2015-07-08:
Had a look at the libraries yesterday, found that lmac.c 662 is printed when something goes wrong during TX (not RX!). Perhaps we are turning off interrupts for too long while flashing, which affects WiFi connectivity?

me-no-dev, 2015-07-08:
maybe we can let the ESP do it's thing between read/write? enable interrupts and give it an yield :) then write the data

me-no-dev, 2015-07-08:
old eboot... just looked at the bin you posted, maybe other things are old as well

me-no-dev, 2015-07-08:
currently running this:
```cpp

bool UpdaterClass::_writeBuffer(){
  WDT_FEED();
  noInterrupts();
  int rc = SPIEraseSector(_currentAddress/FLASH_SECTOR_SIZE);
  interrupts();
  
  if(!rc){
    WDT_FEED();
    noInterrupts();
    rc = SPIWrite(_currentAddress, _buffer, _bufferLen);
    interrupts();
  }
  if (rc){
    _error = UPDATE_ERROR_WRITE;
    _currentAddress = (_startAddress + _size);
#ifdef DEBUG_UPDATER
    printError(DEBUG_UPDATER);
#endif
    return false;
  }
  _currentAddress += _bufferLen;
  _bufferLen = 0;
  return true;
}
```
can you give this a shot? no lmac.c yet, but a bit slow

me-no-dev, 2015-07-08:
And @hallard there is no need to stop UDP or anything else while updating :) so even if it fails, you can try again

igrr, 2015-07-08:
I have to disagree regarding stopAll: I can easily make it fail with
LmacRxBlk if I send lots of TCP or UDP traffic while update happens. So in
real life scenario all UDP and TCP connections which aren't handled
asynchronously have to be stopped. In this particular case we only have
mDNS which is async, so shouldn't be an issue.

hallard, 2015-07-08:
Guys,
I never said we need (or not) stopAll, it was already in the demo sketch I just let it in. I just said in case of OTA failure where no reset are fired (so no lmac error), we need to enable back, if not, you can't do another OTA try because it still disable. And there are 2 points where we need to enable back or fire a reset ;-) 

    if(!Update.begin(size))
    {
      Debugln("Update Begin Error");
      // ======================
      OTA.begin(8266);
      // ======================
      return;
    }
    WiFiClient client;
    if (client.connect(remote, port)) 
    {
      Serial1.setDebugOutput(true);
      while(!Update.isFinished()) 
        Update.write(client);
      Serial1.setDebugOutput(false);

      if(Update.end())
      {
        client.println("OK");
        Debugf("Update Success in %u ms\nRebooting...\n", millis() - startTime);
        ESP.restart();
      } 
      else 
      {
        Update.printError(client);
        Update.printError(Serial);
        // ======================
        OTA.begin(8266);
        // ======================
      }
    }
    else 
    {
      Debugf("Connect Failed: %u\n", millis() - startTime);
    }
  }

me-no-dev, 2015-07-08:
@igrr it's not hard to ddos an ESP :D but in reality you would not have such traffic going to it while updating. Also, usually the packets tat do come in will be small. 
OTA is nothing different than uploading a file to SPIFFS over HTTP POST :) and those did not fail :)
Try my proposed edit a couple of comments above. Besides the speed, it seems working just fine

hallard, 2015-07-08:
Ok guys, got a bad news, my OTA does not work anymore. Just amazing was working so fine, the only thing I changed since yesterday is a `ant clean ; ant` to help @igrr to solve #518 (but I already did `ant clean ; ant` yesterday). I did not fetch/pull anything from repo since.
Now, OTA is working fine, just once!!. Let me explain :
* From IDE, I compile only the sample OTA Sketch and will use the .bin for all following flash tests
* I burn it with esptool.exe -> OK
* I OTA burn same file again with espota.py -> OK
* I OTA burn same file again from espota.py -> KO

I need to flash it again from esptool to be able to use OTA again (but works only one time)
Take a look at the command below (I waited 10s between each command)
[image]

When 2nd OTA fails, the debug log is

    Update Start: ip:192.168.1.226, port:48266, size:318480
    Update Begin Error

Who voodoed my computer using github, just spent hours thinking I've done error in my sketch, but it's not, even with OTA sample it does not work  ;-) ?

Any idea will greatly appreciated.

me-no-dev, 2015-07-08:
try this :) https://github.com/esp8266/Arduino/pull/523

me-no-dev, 2015-07-08:
oops I forgot to update the sketch...
```cpp
      Serial.setDebugOutput(true);
      uint32_t written;
      while(!Update.isFinished()){
        written = Update.write(client);
        if(written > 0) client.print(written, DEC);
      }
      Serial.setDebugOutput(false);
```

hallard, 2015-07-08:
@me-no-dev 
thanks, I will try your fix, but my problem wasn't there. I did not even entered "writing" OTA process on ESP, got update begin error !!!
So I recompiled IDE from scratch, flashed and all is back again, really strange !!!

anyway will now try your fix ;-)

me-no-dev, 2015-07-08:
you know it takes a bit of time after the OK for the device to actually reboot, reflash and boot the new sketch

me-no-dev, 2015-07-08:
what's in the "fix" is making the whole process synchronous, and I added a "missing" eboot.elf
i have not gotten lmac.c yet, nor any other failures

hallard, 2015-07-08:
Yeah I know, I wrote 10s but maybe more, I've blinked the onboard led at the end of setup to have visual indication when all is okay and can start OTA;-)

hallard, 2015-07-08:
@me-no-dev 
thanks, I will try, not sure what to do with eboot.elf ? this file is not generated when compiling sketch or ide ? If not where do I need to put it ?

me-no-dev, 2015-07-08:
its in the esp8266 folder into bootloader/eboot/eboot.elf
not sure if it's generated since makefile is more linux/mac oriented i guess
you can run the make command in the eboot root and see if it compiles
eboot is what boots and switches the firmwares

me-no-dev, 2015-07-08:
also on the update begin failed thing... you sure you have selcted the correct flash size, speed and so on? Since recompile those might have reset

hallard, 2015-07-08:
@me-no-dev 
just tested your changes (copied the eboot.elf, changed updater.cpp, espota.py and sketch) no luck unable to do a correct OTA, here the results
[image]
the debug on ESP same time
[image]

I'm leaving back with delay in my python script for now

me-no-dev, 2015-07-09:
ha... ?? is it not sending the correct packets? mine looks nothing like yours...

me-no-dev, 2015-07-09:
and are you sending 1460 bytes packets like I changed in the python?

hallard, 2015-07-09:
Yeah, for sure
What I do not Know is when eboot.elf is used, in IDE link state ?

igrr, 2015-07-09:
eboot.elf is used at the binary image generation stage.

https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/platform.txt#L78

EUA, 2015-07-15:
OTA still doesn't work with latest git.

```
Arduino OTA Test
Sketch size: 304468
Free size: 741376
IP address: 192.168.1.247
Update Start: ip:192.168.1.2, port:48266, size:304480

Exception (0):
epc1=0x40235c0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40217055 3fff1848 3ffec6c0 000000f5  
3fffff80:  3ffebea0 00000000 00000002 00000610  
3fffff90:  40216a3c 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffebea0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

me-no-dev, 2015-07-15:
example sketch has changes. get the latest one

EUA, 2015-07-15:
Yes, I just cloned main git repo half an hour before and used example that served with it. 
Also WebUpdate sketch doesn't work. It's even doesn't boot, but makes a bootloop!

me-no-dev, 2015-07-15:
will run those again fresh, but to let you know I've had 0 problems with the latest OTA code and uploader. I really mean 0 issues. here I will paste you the OTA base for any sketch that you want to use it in:
```cpp
WiFiUDP OTA;

void initOTA(){
  MDNS.begin(host);
  MDNS.addService("arduino", "tcp", ota_port);
  OTA.begin(ota_port);
}

void checkOTA(){
  if (OTA.parsePacket()) {
    IPAddress remote = OTA.remoteIP();
    int cmd  = OTA.parseInt();
    int port = OTA.parseInt();
    int size   = OTA.parseInt();

    Serial.printf("Update Start: %d\n", size);
    if(!Update.begin(size)){
      Update.printError(Serial);
      return;
    }
    
    WiFiClient client;
    if (client.connect(remote, port)) {
      Serial.setDebugOutput(true);
      uint32_t written;
      while(!Update.isFinished()){
        written = Update.write(client);
        if(written > 0) client.print(written, DEC);
      }
      Serial.setDebugOutput(false);
      if(Update.end()){
        client.print("OK");
        Serial.printf("Update Success\n");
        ESP.restart();
      } else {
        Update.printError(client);
        Update.printError(Serial);
      }
    } else {
      Serial.printf("Connect Failed\n");
    }
  }
}
```

Imroy, 2015-07-17:
I whipped up a sketch to do OTA updates via MQTT:
https://github.com/Imroy/pubsubclient/blob/master/examples/ESP8266-OTA/ESP8266-OTA.ino

MQTT messages can be up to 256 MiB in size, so there's plenty of room for a sketch. I just had to modify PubSubClient to handle "large" messages.

I also get the "lmac.c 662" message sometimes.

hallard, 2015-07-21:
Hu guys,
sorry went away for other stuff, I'm back testing
I've rebuilt from scratch, git clone + ant (just one hour ago)
then ran Arduino OTA test, I'm sorry but in my case it does not works

here the ouputs from console and serial 

[image]

[image]

To be sure I copied .bin and espota.py to a linux box and tried with python 2.6 and python 2.7, same results 

[image]

Then modified espota.py to Python3 as told by @kzyapkov
https://github.com/kzyapkov/Arduino/commit/f5653529ecbad1a29631198e2740ed9395e247f3

Once again same results

[image]

Here the full serial output for all these tests

```
Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
Update Start: ip:192.168.1.226, port:48266, size:304688

ctx: cont 
sp: 3ffea840 end: 3ffeaa80 offset: 01b0

>>>stack>>>
3ffea9f0:  e201a8c0 00000000 3ffeab10 402038eb  
3ffeaa00:  3ffeab10 0000bc8a 3ffeaae0 40202433  
3ffeaa10:  3ffe8d00 00000000 000003e8 4020734c  
3ffeaa20:  3ffe9698 3fff4bf0 3ffe8f48 e201a8c0  
3ffeaa30:  3ffe8f48 e201a8c0 3ffeaae0 402022d0  
3ffeaa40:  00004138 00000000 00000016 40101e31  
3ffeaa50:  402017f9 00000000 00000000 3ffeaaac  
3ffeaa60:  3fffdc20 00000000 3ffeaaa4 40201856  
3ffeaa70:  00000000 00000000 3ffe9a60 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld

Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
Update Start: ip:192.168.1.21, port:48266, size:304688

ctx: cont 
sp: 3ffea850 end: 3ffeaa80 offset: 01b0

>>>stack>>>
3ffeaa00:  3ffeab10 0000bc8a 3ffeaae0 40202433  
3ffeaa10:  3ffe8d00 00000000 000003e8 4020734c  
3ffeaa20:  3ffe9698 3fff4b30 3ffe8f48 1501a8c0  
3ffeaa30:  3ffe8f48 1501a8c0 3ffeaae0 402022d0  
3ffeaa40:  000f01c1 00000000 00000016 40101e31  
3ffeaa50:  402017f9 00000000 00000000 3ffeaaac  
3ffeaa60:  3fffdc20 00000000 3ffeaaa4 40201856  
3ffeaa70:  00000000 00000000 3ffe9a60 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld

Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
Update Start: ip:192.168.1.21, port:48266, size:304688

ctx: cont 
sp: 3ffea860 end: 3ffeaa80 offset: 01b0

>>>stack>>>
3ffeaa10:  3ffe8d00 00000000 000003e8 4020734c  
3ffeaa20:  3ffe9698 3fff4780 3ffe8f48 1501a8c0  
3ffeaa30:  3ffe8f48 1501a8c0 3ffeaae0 402022d0  
3ffeaa40:  00003380 00000000 00000016 40101e31  
3ffeaa50:  402017f9 00000000 00000000 3ffeaaac  
3ffeaa60:  3fffdc20 00000000 3ffeaaa4 40201856  
3ffeaa70:  00000000 00000000 3ffe9a60 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld

Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
Update Start: ip:192.168.1.21, port:48266, size:304688

ctx: cont 
sp: 3ffea860 end: 3ffeaa80 offset: 01b0

>>>stack>>>
3ffeaa10:  3ffe8d00 00000000 000003e8 4020734c  
3ffeaa20:  3ffe9698 3fff4bf0 3ffe8f48 1501a8c0  
3ffeaa30:  3ffe8f48 1501a8c0 3ffeaae0 402022d0  
3ffeaa40:  000451f2 00000000 00000016 40101e31  
3ffeaa50:  402017f9 00000000 00000000 3ffeaaac  
3ffeaa60:  3fffdc20 00000000 3ffeaaa4 40201856  
3ffeaa70:  00000000 00000000 3ffe9a60 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld

Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
Update Start: ip:192.168.1.21, port:48266, size:304688

ctx: cont 
sp: 3ffea860 end: 3ffeaa80 offset: 01b0

>>>stack>>>
3ffeaa10:  3ffe8d00 00000000 000003e8 4020734c  
3ffeaa20:  3ffe9698 3fff4780 3ffe8f48 1501a8c0  
3ffeaa30:  3ffe8f48 1501a8c0 3ffeaae0 402022d0  
3ffeaa40:  00002737 00000000 00000016 40101e31  
3ffeaa50:  402017f9 00000000 00000000 3ffeaaac  
3ffeaa60:  3fffdc20 00000000 3ffeaaa4 40201856  
3ffeaa70:  00000000 00000000 3ffe9a60 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld

Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
```

Any ideas (Generic ESP12 80MHz/4Mb) ?

igrr, 2015-07-21:
Changes related to optimisitc_yield have broken much of the network functions. Could you please try the potential fix from this branch: https://github.com/igrr/Arduino/commits/optimistic-yield ?

hallard, 2015-07-22:
@igrr 
To be sure It was not related with my 8.1 x64 computer, I tried rebuilding from scratch on a updated Linux Debian Jessie laptop, 
It's exactly the same problem does not work.

But these tests were not in vain, for those interested I noticed two things over Windows
- Serial Upload at High Speed (>=512000) is very random under my windows computer. Even if 115200 works every time, over this speed, sometimes it works and other not (same board/same wiring). But works all the time under linux
- MDNS advertised service board are displayed correctly in Arduino IDE so you can select the network port corresponding to OTA test sketch (it's not shown on my windows due to bonjour service related voodo, and I gave up trying to resolve this)

By the way, I spent too much time trying things to get working with OTA for now. And it's not critical for my project, so I will focus back on my project code and will keep an eye on the new versions.

@igrr the link above does not seems to be working, is that the same branch optimistic-yield than the one into arduino-esp8266 repo ? FYI I do not have any network related problem with the current branch except with OTA, but may be the problems occurs on UDP only ?

Riggeotto, 2015-07-23:
@igrr @hallard @me-no-dev 
I also am having issues with doing OTA updates as well.

Working from https://github.com/esp8266/Arduino/tree/esp8266 I get these results:

```
07-22-2015 07:41:22 PM [RX]: Arduino OTA Test 1.0.0
07-22-2015 07:41:22 PM [RX]: Sketch size: 306420
Free size: 741376
IP address: 192.168.1.62

07-22-2015 07:41:41 PM [RX]: Update Start: ip:192.168.1.93, port:48266, size:306432
Update Success: 9513
Rebooting...

07-22-2015 07:41:51 PM [RX]:  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
07-22-2015 07:41:51 PM [RX]: 
07-22-2015 07:41:51 PM [RX]: load 0x4010f000, len 1240, room 16 
07-22-2015 07:41:51 PM [RX]: tail 8
07-22-2015 07:41:51 PM [RX]: chksum 0x04
07-22-2015 07:41:51 PM [RX]: csum 0x04
07-22-2015 07:41:51 PM [RX]: @cp:
07-22-2015 07:41:58 PM [RX]:  ets Jan  8 2013,rst cause:4, boot mode:(3,6)
07-22-2015 07:41:58 PM [RX]: 
07-22-2015 07:41:58 PM [RX]: wdt reset

07-22-2015 07:41:58 PM [RX]: load 0x4010f000, len 1240, room 16 
07-22-2015 07:41:58 PM [RX]: tail 8
07-22-2015 07:41:58 PM [RX]: chksum 0x04
07-22-2015 07:41:58 PM [RX]: csum 0x04
07-22-2015 07:41:58 PM [RX]: ~ld

07-22-2015 07:41:58 PM [RX]: Arduino OTA Test 1.0.0
07-22-2015 07:41:58 PM [RX]: Sketch size: 306420
Free size: 741376
IP address: 192.168.1.62
```

And the output from the espota python script indicates that the transfer was successful:

```
PS E:\Development\Workspace\Repositories\esp8266\Arduino\build\windows\work\hardware\esp8266comesp8266\tools> .\espota.cmd 192.168.1.62 8266 esp_ota.cpp_1.bin
E:\Development\Workspace\Repositories\esp8266\Arduino\build\windows\work\hardware\esp8266com\esp8266\tools>python espota.py 192.168.1.62 8266 E:\Development\Workspace\Repositories\esp8266\Arduino\build\windows\work\hardware\esp8266com\esp8266\tools\esp_ota.cpp_1.bin
Starting on 0.0.0.0:48266
Upload size: 306432
Sending invitation to: 192.168.1.62
Waiting for device...

Uploading..................................................................................................................................................................................................................
Waiting for result...

Result: OK
```

I was initially working on doing this through a Java application and wasn't sure if it was something specific to my application so I tried the python script and the DNS_SD_Arduino_OTA sketch to no avail.

Looking at the bootloader, it never seems to output the return value from the copy_raw function, but rather resets before it gets there.  It looks as though the watch dog gets tripped and resets the chip before it finishes copying the binary in flash.

Looking into this further, I downloaded the latest Espressif SDK and tested this theory with the attached binary which is presumably blank and the output is as follows:

```
07-22-2015 09:33:02 PM [RX]: Arduino OTA Test 1.0.0
07-22-2015 09:33:02 PM [RX]: Sketch size: 306452
Free size: 741376
IP address: 192.168.1.62

07-22-2015 09:33:14 PM [RX]: Update Start: ip:192.168.1.93, port:48266, size:4096
Update Success: 311
Rebooting...

07-22-2015 09:33:14 PM [RX]:  ets Jan  8 2013,rst cause:2, boot mode:(3,7)

07-22-2015 09:33:14 PM [RX]: load 0x4010f000, len 1240, room 16 
07-22-2015 09:33:14 PM [RX]: tail 8
07-22-2015 09:33:14 PM [RX]: chksum 0x04
07-22-2015 09:33:14 PM [RX]: csum 0x04
07-22-2015 09:33:14 PM [RX]: @cp:0
ld

07-22-2015 09:33:15 PM [RX]: Arduino OTA Test 1.0.0
07-22-2015 09:33:15 PM [RX]: Sketch size: 306452
Free size: 741376
IP address: 192.168.1.62
```

This leads me to believe that possibly disabling or feeding the watch dog during the copy process would circumvent this issue.

I'll look into making trying and making the changes myself but C isn't exactly my forte.  If someone else tries, I would be curious to also test said changes.

This could explain why some people run into this issue while others don't, assuming the binary is small enough the bootloader might have enough time to finish copying the binary, whereas if the binary is large, this situation might occur.

Riggeotto, 2015-07-23:
Just an update, I modified eboot.c to disable and reenable the watch dog before and after copying the binary and the OTA updates are now working for me, no more getting interrupted during the middle of the copy operation.

```
Arduino OTA Test 1.0.0
Sketch size: 306420
Free size: 741376
IP address: 192.168.1.62
Update Start: ip:192.168.1.93, port:48266, size:306432
Update Success: 10755
Rebooting...

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0xb4
csum 0xb4
@cp:0
ld

Arduino OTA Test 1.0.1
Sketch size: 306420
Free size: 741376
IP address: 192.168.1.62
```

Modified eboot.c:
```
/* Copyright (c) 2015 Ivan Grokhotkov. All rights reserved. 
 * This file is part of eboot bootloader.
 *
 * Redistribution and use is permitted according to the conditions of the
 * 3-clause BSD license to be found in the LICENSE file.
 */

#include <stddef.h>
#include <stdint.h>
#include <stdbool.h>
#include "flash.h"
#include "eboot_command.h"

#define SWRST do { (*((volatile uint32_t*) 0x60000700)) |= 0x80000000; } while(0);

int load_app_from_flash_raw(const uint32_t flash_addr)
{
    image_header_t image_header;
    uint32_t pos = flash_addr + APP_START_OFFSET;

    if (SPIRead(pos, &image_header, sizeof(image_header))) {
        return 1;
    }
    pos += sizeof(image_header);
    

    for (uint32_t section_index = 0;
        section_index < image_header.num_segments; 
        ++section_index)
    {
        section_header_t section_header = {0};
        if (SPIRead(pos, &section_header, sizeof(section_header))) {
            return 2;
        }
        pos += sizeof(section_header);

        const uint32_t address = section_header.address;

        bool load = false;

        if (address < 0x40000000) {
            load = true;
        }

        if (address >= 0x40100000 && address < 0x40108000) {
            load = true;
        }

        if (address >= 0x60000000) {
            load = true;
        }

        if (!load) {
            pos += section_header.size;
            continue;
        }

        if (SPIRead(pos, (void*)address, section_header.size))
            return 3;

        pos += section_header.size;
    }

    register uint32_t sp asm("a1") = 0x3ffffff0;
    register uint32_t pc asm("a3") = image_header.entry;
    __asm__  __volatile__ ("jx a3");

    return 0;
}

int copy_raw(const uint32_t src_addr, 
             const uint32_t dst_addr, 
             const uint32_t size)
{
    // require regions to be aligned
    if (src_addr & 0xfff != 0 ||
        dst_addr & 0xfff != 0) {
        return 1;
    }

    const uint32_t buffer_size = FLASH_SECTOR_SIZE;
    uint8_t buffer[buffer_size];
    uint32_t left = ((size+buffer_size-1) & ~(buffer_size-1));
    uint32_t saddr = src_addr;
    uint32_t daddr = dst_addr;

    ets_wdt_disable();
    
    while (left) {

      if (SPIEraseSector(daddr/buffer_size)) {
          return 2;
      }
      if (SPIRead(saddr, buffer, buffer_size)) {
        return 3;
      }
      if (SPIWrite(daddr, buffer, buffer_size)) {
        return 4;
      }
      saddr += buffer_size;
      daddr += buffer_size;
      left  -= buffer_size;
    }

    ets_wdt_enable();

    return 0;
}

void main()
{
    int res = 9;
    struct eboot_command cmd;

    if (eboot_command_read(&cmd)) {
        cmd.action = ACTION_LOAD_APP;
        cmd.args[0] = 0;
        ets_putc('~');
    } else {
        ets_putc('@');
    }
    eboot_command_clear();
    
    if (cmd.action == ACTION_COPY_RAW) {
        ets_putc('c'); ets_putc('p'); ets_putc(':');
        res = copy_raw(cmd.args[0], cmd.args[1], cmd.args[2]);
        ets_putc('0'+res); ets_putc('\n');
        if (res == 0) {
            cmd.action = ACTION_LOAD_APP;
            cmd.args[0] = cmd.args[1];
        }
    }

    if (cmd.action == ACTION_LOAD_APP) {
      ets_putc('l'); ets_putc('d'); ets_putc('\n');
      res = load_app_from_flash_raw(cmd.args[0]);
      //we will get to this only on load fail
      ets_putc('e'); ets_putc(':'); ets_putc('0'+res); ets_putc('\n');
    }

    if (res) {
      SWRST;
    }

    while(true){}
}
```

Not sure if this is the proper solution to this, but it fixed the issue I was running into.

av1024, 2015-07-23:
Got boot failure after OTA update with latest git version c6e2a290d1 and code one-by-one based on sample sketch. Looks like Flash writing issue.

There is no errors when updated same binary as flashed by serial.
But after OTA update got various failures - boot loop:
```
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld
```
or fatal exception:
```
 * OTA UPDATE *
 OTA size: 349472
 OTA Update Done. Rebooting.
rm 0
pm close 7 0 0/25760005
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
@cp:
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld
----- RESET BUTTON PRESS -----
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld
Fatal exception (0):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):
Fatal exception (3):

```

igrr, 2015-07-23:
@Riggeotto thanks for the observation, I think your solution makes sense. Updated eboot.c as per your suggestion.

hallard, 2015-07-24:
@Riggeotto 
Good catch for the watchdog ;-) Unfortunatly this does not solve my issue because I'm not going so far, I think I do not getting any data sent by python script. Transfer even does not start, look below I should have a lot of dot displaying on transfer (like in your example). After `Uploading` word, I got only one (the time between display of `Uploading` and `Error Uploading` is about 4/5 second, time from ESP to reboot itself after `ctx:cont` message.
````
build5457492820131988925.tmp # python ~/espota.py 192.168.1.51 8266 DNS_SD_Arduino_OTA.cpp.bin
Starting on 0.0.0.0:48266
Upload size: 304688
Sending invitation to: 192.168.1.51
Waiting for device...

Uploading.
Error Uploading
build5457492820131988925.tmp #
````
and as soon as I push enter key here (of above espota.py command) I got this on ESP Serial
````
Update Start: ip:192.168.1.226, port:48266, size:304688

ctx: cont 
sp: 3ffea860 end: 3ffeaa80 offset: 01b0

>>>stack>>>
3ffeaa10:  3ffe8d00 00000000 000003e8 4020734c  
3ffeaa20:  3ffe9698 3fff4780 3ffe8f48 e201a8c0  
3ffeaa30:  3ffe8f48 e201a8c0 3ffeaae0 402022d0  
3ffeaa40:  00001b3a 00000000 00000016 40101e31  
3ffeaa50:  402017f9 00000000 00000000 3ffeaaac  
3ffeaa60:  3fffdc20 00000000 3ffeaaa4 40201856  
3ffeaa70:  00000000 00000000 3ffe9a60 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1240, room 16 
tail 8
chksum 0x04
csum 0x04
~ld

Arduino OTA Test
Sketch size: 304680
Free size: 741376
IP address: 192.168.1.51
````
it does not upload anything. So I would believe I'm doing something wrong, but I've done so much tests by :
* git clone from scratch from my 8.1 X64 Computer
* git clone from scratch from my Linux debian Laptop
* ant on both to rebuild IDE from scratch 
* launched IDE on both select the correct board on each (NodeMCU on Windows, Generic ESP on linux) and loaded example `DNS_SD_Arduino_OTA.ino` sketch after modified SSID and password of course
* uploaded the compiled `DNS_SD_Arduino_OTA.ino` generated on both using serial port
* Then tried then to upload OTA sample `DNS_SD_Arduino_OTA Sketch.cpp.bin` with
* Windows 8.1 with Node MCU 1.0 board connected to Access point 1
* Linux with Generic ESP board connected to Access point 2

So both same results, does not work, 2 boards, 2 Environements, 2 access point. I really don't know what can I do else for testing ?
I'm open to any investigation to write more info on serial if needed.

hallard, 2015-07-24:
Ok guys, got it working.
You remember my 1st message in this issue, I noticed I needed to add some delay in Python script (old OTA version) to get it working. It's again a delay problem. I digged into the file [updater.h](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/cores/esp8266/Updater.h) where is the `write` template and added some print to see some debug. 
you won't believe it as soon as I added some Serial.print it worked. So I direct thought it was a timing problem (going too fast ?). When I isolated where the delay should be, I put a `delay(1)`, it was working but slowing down a little bit the update process. As I did not found a delay microsecond like (to reduce it) but I thought that `yield()` should do the trick, and it did. 
So in the `template<typename T> size_t write(T &data)`, code looked like 
````
      size_t available = data.available();
      while(available) {
````
becomes
````
      size_t available = data.available();
      yield();
      while(available) {
````

It's now far much better, but I noticed depending on which access point I'm connected to, sometimes I may miss one OTA, so I really think the process would have a basic check, such as when it's okay :
* python send data size (n bytes) to ESP
* ESP answer "Got data n bytes, it was stored ok"
* continue until done
And if answer from ESP does not match, someting like that
* python send data size (n bytes) to ESP
* ESP does not answer good data or number of received bytes, ESP set back it storing pointer to end of last chunk
* python resend same data size (n bytes) to ESP
* .....

av1024, 2015-07-24:
It's a magic (c)!
The latest 39819f0222 commit works like a charm.

As usually all "just works" after a lot of investigations.
I was made changes to espota.py for dry-run with trivial checksum control, test it w/o errors on latest revision.... And got no errors on real OTA update ))

Keep link to gist here, may be it will be helpful for someone. https://gist.github.com/av1024/8d058d31a6dfb74bbd0d

Btw, I think it's will be good idea to add some checksum control as for receiving, as for Flash writing

hallard, 2015-07-24:
@av1024 
Whoo excellent, this works fine. I just changed the sketch because they're wasn't any end to the while loop because cntr was not incremented. I also checked all is correctly transfered after the while loop before sending OK.
I've encapsuled all of this in my sketch to a function to be called from any main loop to do OTA. Here the function :

````arduino
#define DEBUG
//debug added for information, change this according your needs

#ifdef DEBUG
#define Debug(x)    Serial1.print(x)
#define Debugln(x)  Serial1.println(x)
#define Debugf(...) Serial1.printf(__VA_ARGS__)
#define Debugflush  Serial1.flush
#else
#define Debug(x)    {}
#define Debugln(x)  {}
#define Debugf(...) {}
#define Debugflush  {}
#endif

#define OTA_PORT 8266

/* ======================================================================
Function: CheckOTAUpdate 
Purpose : Check if we need to update the firmware over the Air
Input   : -
Output  : - 
Comments: If upgraded, no return, perform update and reboot ESP
====================================================================== */
void CheckOTAUpdate(void)
{
  //OTA detection
  if (OTA.parsePacket()) {
    IPAddress remote = OTA.remoteIP();
    int cmd  = OTA.parseInt();
    int port = OTA.parseInt();
    int size = OTA.parseInt();

    Debug(F("Update Start: ip:"));
    Debug(remote);
    Debugf(", port:%d, size:%d\n", port, size);
    uint32_t startTime = millis();

    WiFiUDP::stopAll();

    int ww = 0;
    WiFiClient client;
    
    if (!client.connect(remote, port)){
      Debugf("Connect Failed: %u\n", millis() - startTime);
      return;
    } 

    client.setTimeout(10000);
    int cntr = 0;
    int z = 0;
    uint16_t crc;

    // do until we do not get transferred full size
    while (cntr<size) { 
      z = 0;
      crc = 0;
      for(int i=client.available();i>0;i--) { 
        int b = client.read();

        // timeout
        if (b == -1) 
          break; 

        crc += (byte)b;
[... 169 lines cut ...]
      f.close()
      sock.close()
      return 0
    except:
      print('Result: No Answer!', file=sys.stderr)
      connection.close()
      f.close()
      sock.close()
      return 1

  finally:
    connection.close()
    f.close()

  sock.close()
  return 1

def main(args):
  return serve(args[1], args[2], args[3])

if __name__ == '__main__':
  sys.exit(main(sys.argv))
````

Seems we have now descent working OTA, thank all for your help, not perfect (checksum control, flash write control, retry sent packet) but that works and do perfectly the job ;-)

hallard, 2015-07-24:
Oh, BTW, I noticed that when you upload sketch by serial then, after you do your 1st OTA upload, the sketch don't start, you need to push reset button to get it starting. Once this manual reset is done once all is working fine.
Nothing serious, just strange, but it's good to know.

av1024, 2015-07-24:
@hallard, the hardware reset after serial upload required at least since May 2015 as far I remember. Looks like ESP does not check 'PROG' pin on software reset

About code: Good job. The sample code in python header is just as reference (not copy from project so may have some typos). The real code in my project slightly more complex (#ifdefs, project integrations etc). The actual code here: [bitbucket](https://bitbucket.org/esp8266/sensor-v2/src/b0cf9f6fb1eb66cf70efa993f4e726ebbeaa9115/ota.h?at=default)

hallard, 2015-07-24:
@av1024 
Nice project, very lose of what I'm doing with ESP, may be I will grab you some ideas ;-)
BTW, I saw highchart, where does it data from to plot them ? ESP or MQTT or other ?
I didn't thought about a "telnet" like configuration, what a great idea, I'm dealing with Web interface for this, but I like the telnet idea ;-)

igrr, 2015-07-24:
This is getting slightly OT, but have you guys seen [Bitlash](https://github.com/billroy/bitlash)? I think it would be easy to run it on ESP, providing a "shell" with some scripting support over Telnet.

Back to topic, would you consider opening a pull request with the changes above? I'm slightly consumed somewhere in between SSL and Filesystem, so some help getting all the OTA pieces in place would be much appreciated.

hallard, 2015-07-24:
@igrr,
I would like for pull request, but I'm really not comfortable with git, last time I forked (so created one on my repo) then pulled from my repo and done some changes, and the mess started !!!!
My main concern is that after, I'm unable to get my repo in sync with the master (and them my computer to my sync'ed repo) to have the latest code.
I'm pretty sure it comes from me but each time I needed to delete all my repo and start from scratch, just too much each time. I never succeed to find any good documentation (at least working) on how to do that, so that is why for just 2 files, I posted them here. I know it's not so clean than a pull request, but if one of you guys who forked can give me the list of command to do this; I for sure will give a try. Basically my needs are very simple :
* Fork repo to my git (I know I'm doing from git site)
* clone my forked repo (I know, git clone from my computer)
Then 2 directions  

1) I do nothing (no time) and just want to sync my repo from arduino-esp8666 and then update files from my sync'ed repo to my computer (to be sure I have the latest). Should be easy, but I never succeed to do this, as I build the IDE, git says I've got new/modified files bla bla bla !!!!

2) I change some files, and also want to sync my repo from arduino-esp8666 and then update file from my sync'ed repo to my computer (assuming the files I'm working on did not changed on master repo)

I know this seems simple for you guys, but trust me, I think git and I having difficulties to collaborate for branching/syncing ;-)

Links2004, 2015-07-24:
you not force to use the command line :)
on windows i use tortoisegit https://tortoisegit.org it has a easy to use gui for git.
or you edit the file online direct on github.
[image]

the basic commands for git are:
git clone       // init the repo 
git fetch        // get changes from server (not change you working files)
git pull          // get changes from server and try auto merge
git commit    // "saves" to local git
git add file.c  // add new files to the local git
git push         // send changes to the server

hallard, 2015-07-24:
@Links2004 
thks for your help, until there I got it and I understand, may be I can go further.
Now when I want to update the files on my computer with the modification done by @igrr or others from original code, it's from there I'm starting with problems ?

Links2004, 2015-07-24:
if you want to get mortification form other branches like form https://github.com/esp8266/Arduino/ for example you have to done some steps:
1. add the src git as remote repository ( .git/config )  my is looking like this:
```
[core]
	repositoryformatversion = 0
	filemode = false
	bare = false
	logallrefupdates = true
	symlinks = false
	ignorecase = true
	hideDotFiles = dotGitOnly
[remote "origin"]
	url = git@github.com:Links2004/Arduino.git
	fetch = +refs/heads/*:refs/remotes/origin/*
[remote "esp8266"]
	url = git@github.com:esp8266/Arduino.git
	fetch = +refs/heads/*:refs/remotes/esp8266/*
```
"origin" is my one github clone, but i can get stuff from the other branches.

2. you need to fetch all the branches. ```git fetch --all```

the next step i never done at console level, but it shut work like this:
```git merge esp8266/esp8266``` with this you get the latest changes form https://github.com/esp8266/Arduino
if you have conflicts (auto merge fail) then you need to solve them and then commit the resolve.
```git commit -a``` (tortoisegit make this step really easy).

av1024, 2015-07-24:
@hallard, You are welcome ))
about highcharts: I use mosquitto 1.4.2 as MQTT server on Cubietruck and try MQTT over websockets. This is my first time websockets and highcharts usage, so sample html is for reference only )). 

BTW, instead of ubuntu/debian, the OpenWRT's 1.5 mosquitto compiled with websockets enabled, so I can use it from TP-LINK WR703 w/o any modifications

@igrr, Bitlash, on ver-very first look is similar for my CLI sub-repo. I will read more about it on vacation... May be - Crimea, vine, climbing... )) I think it's a good idea to add some "default" command-line handler as example or library. As for me, I wrote my own because does not find simple standard solution

EUA, 2015-07-27:
Still waiting of patch for this issue. :(

sticilface, 2015-07-31:
@hallard  I've had a go at using your python script for the progress bar, but it doesn't seem to play ball with the latest OTA stuff.  Does yours still work?

EUA, 2015-08-06:
@hallard could you please put a patch for this issue? Please.

rvbcrs, 2015-08-07:
Hi guys, I have tried about every OTA script I could find, none of them worked for me, including the WebUpdate example!. I'm building a Sprinkler system that is in a somewhat inconvenient place so it would be perfect if I could just load new code via OTA. Any ideas what is going on?

igrr, 2015-08-07:
@rvbcrs any specifics except for the "none of them worked for me"? Serial output, debug output, wireshark dumps, python script output, anything?

rvbcrs, 2015-08-07:
@igrr ofcourse sorry, was under the impression that you guys were aware of these issues already and just wanted to say that I experience issues with OTA too. Ok with the latest staging version of the ESP8266 arduino build if I load the DNS_SD_ARDUINO_OTA script and run the python script I get this as output in my Serial window as soon as it gets to this line: 

`if(written > 0) client.print(written, DEC);`

```
Arduino OTA Test
Sketch size: 306516
Free size: 741376
IP address: 192.168.1.104
Update Start: ip:192.168.1.113, port:48266, size:306944

Exception (0):
epc1=0x40236528 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  402171d9 3ffef8f8 3ffecd24 000000f5  
3fffff80:  3ffebef0 00000080 00000000 0000011c  
3fffff90:  40216bc0 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffebef0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

Next I upload the example 'WebUpdate' sketch and try to upload a .bin file.

In the browser I see at the status bar: Uploading 32% and then it halts there. In the serial monitor I see this:

```
Booting Sketch...
Ready! Open http://esp8266-webupdate.local in your browser
Update: HelloServer.bin
lmac.c 599

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

Let me know if you need more info, Thanks!

Ramon

CptanPanic, 2015-08-07:
That sounds like same issue that was fixed above https://github.com/esp8266/Arduino/commit/6b3ed670328eb39b012c20b65d0aeecaf18bdcf9 found relating to the wdt.  Are you sure  you have latest update, or maybe this didn't quite make it into last update.

rvbcrs, 2015-08-07:
Well I'm sure I have the latest staging version: esp8266-1.6.5-990-gc8a63ce but I think I also tried downloading the latest version from GIT and replaced all the files in esp8266-1.6.5-990-gc8a63ce and still having the same problems, but if you say it should be fixed I can try again.

[16 more replies]

closed 2015-09-23 by hallard