esp8266/Arduino #518: change esptool exe permission on windows build

hallard, 2015-07-08:
Guys,
when I pull repo and build (cygwin) with ant under windows (8.1 x64) the programmer file esptool does not have the execute permission, this makes the IDE unable to launch the tool and fire a error.

[image]

chmod ug+x the esptool.exe file solve the problem. Would it be possible to add this job to the windows build process ?

Thank you very much for your help

Cheers,
Charles

igrr, 2015-07-08:
Charles,

Please try this patch locally and let me know if it works. I don't have the environment to test.
```diff
diff --git a/build/build.xml b/build/build.xml
index 663bc63..fcbbc1c 100644
--- a/build/build.xml
+++ b/build/build.xml
@@ -987,6 +987,7 @@
     </antcall>
     <move file="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool-0.4.5-win32/esptool.exe" tofile="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool.exe" />
     <delete dir="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool-0.4.5-win32/" />
+    <chmod perm="+x" file="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool.exe" />
  
   </target>
```

hallard, 2015-07-08:
Ivan,
did not worked, chmod seems doing nothing in this case !!
But, got the solution, remembering my Windows Server IT Admin years, move keep perms from source so solution is to copy (instead of moving) the file from source, then delete from source
This is done by 

    </antcall>
    <copy file="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool-0.4.5-win32/esptool.exe" tofile="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool.exe" />
    <delete file="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool-0.4.5-win32/esptool.exe" />
    <delete dir="${staging_folder}/work/${staging_hardware_folder}/esp8266com/esp8266/tools/esptool-0.4.5-win32/" />

Then it works ;-)

[image]

hallard, 2015-07-08:
Oh I'm pretty sure the delete line is not needed since the folder is deleted after

closed 2015-07-23 by igrr