esp8266/Arduino #519: Cannot access GPIO15 -- No FUNCTION_3 switch in extern void __pinMode(uint8_t pin, uint8_t mode) {

drmpf, 2015-07-08:
I am trying to use GPIO15 on the OMILEX DEV BOARD (the one with the relay)
But cannot read that pin.
default function appears to be MTDO,  and there is no FUNCTION_3 switch in the 
extern void __pinMode(uint8_t pin, uint8_t mode)  
method to make this pin accessible as GPIO in stable release 1.6.4-673-g8cd3697

I need two digital inputs and have all ready used GPIO4 as one and am using GPIO16 for ADC,
Cannot use GPIO0 or 2 as my be shorted to gnd on power up by external switch
GPIO5 on this board drives the relay and TX/RX use the other two GPIO's

  if(pin < 16){
    if(mode == SPECIAL){
      GPC(pin) = (GPC(pin) & (0xF << GPCI)); //SOURCE(GPIO) | DRIVER(NORMAL) | INT_TYPE(UNCHANGED) | WAKEUP_ENABLE(DISABLED)
      GPEC = (1 << pin); //Disable
      GPF(pin) = GPFFS(GPFFS_BUS(pin));//Set mode to BUS (RX0, TX0, TX1, SPI, HSPI or CLK depending in the pin)
      if(pin == 3) GPF(pin) |= (1 << GPFPU);//enable pullup on RX
    } else if(mode & FUNCTION_0){
      GPC(pin) = (GPC(pin) & (0xF << GPCI)); //SOURCE(GPIO) | DRIVER(NORMAL) | INT_TYPE(UNCHANGED) | WAKEUP_ENABLE(DISABLED)
      GPEC = (1 << pin); //Disable
      GPF(pin) = GPFFS((mode >> 4) & 0x07);
      if(pin == 13 && mode == FUNCTION_4) GPF(pin) |= (1 << GPFPU);//enable pullup on RX
    }  else if(mode == OUTPUT || mode == OUTPUT_OPEN_DRAIN){
      GPF(pin) = GPFFS(GPFFS_GPIO(pin));//Set mode to GPIO
      GPC(pin) = (GPC(pin) & (0xF << GPCI)); //SOURCE(GPIO) | DRIVER(NORMAL) | INT_TYPE(UNCHANGED) | WAKEUP_ENABLE(DISABLED)
      if(mode == OUTPUT_OPEN_DRAIN) GPC(pin) |= (1 << GPCD);
      GPES = (1 << pin); //Enable
    } else if(mode == INPUT || mode == INPUT_PULLUP || mode == INPUT_PULLDOWN){
      GPF(pin) = GPFFS(GPFFS_GPIO(pin));//Set mode to GPIO
      GPEC = (1 << pin); //Disable
      GPC(pin) = (GPC(pin) & (0xF << GPCI)) | (1 << GPCD); //SOURCE(GPIO) | DRIVER(OPEN_DRAIN) | INT_TYPE(UNCHANGED) | WAKEUP_ENABLE(DISABLED)
      if(mode == INPUT_PULLUP) {
          GPF(pin) |= (1 << GPFPU);  // Enable  Pullup
      } else if(mode == INPUT_PULLDOWN) {
          GPF(pin) |= (1 << GPFPD);  // Enable  Pulldown
      }
    }
  } else if(pin == 16){
    GPF16 = GP16FFS(GPFFS_GPIO(pin));//Set mode to GPIO
    GPC16 = 0;
    if(mode == INPUT || mode == INPUT_PULLDOWN){
      if(mode == INPUT_PULLDOWN){
        GPF16 |= (1 << GP16FPD);//Enable Pulldown
      }
      GP16E &= ~1;
    } else if(mode == OUTPUT){
      GP16E |= 1;
    }
  }
}

me-no-dev, 2015-07-08:
when you call pinMode(pin, INPUT/OUTPUT) it's automatically switched to GPIO mode
since the function number is different for some of the pins, I have wrapped it automatically.
```cpp
GPF(pin) = GPFFS(GPFFS_GPIO(pin));
```

drmpf, 2015-07-08:
Thanks for the info but.. 
this sketch reboots

```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  for (int i = 10; i > 0; i--) {
    Serial.print(i);
    Serial.print(' ');
    delay(500);
  }
  Serial.println();
  // set all the pins as inputs with pullup
  pinMode(6,INPUT_PULLUP); // sd_clk
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  printOut("gpio_6",6);
}

void printOut(const char* lable, int pin) {
  Serial.print(lable);
  if (digitalRead(pin) == LOW) {
    Serial.println(" LOW");
  } else {
    Serial.println(" HIGH");
  }
}
```

Get the following output in Serial Monitor  directly after flashing the program
```
 ets Jan  8 2013,rst cause:4, boot mode:(1,3)

wdt reset
```

Get this output after power down and power up
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0xd3
ho 0 tail 12 room 4
load 0x3ffe8000, len 1332, room 12 
tail 8
chksum 0x0f
load 0x3ffe8540, len 1568, room 0 
tail 0
chksum 0x8e
csum 0x8e
rl10 9 8 7 6 5 4 3 2 1 

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
```
repeating for ever

igrr, 2015-07-09:
You said you were trying to use GPIO15, while your sketch refers to GPIO6:
`pinMode(6,INPUT_PULLUP); // sd_clk`
GPIO6 is taken by flash chip interface, so changing it's mode will lead to a crash.

drmpf, 2015-07-09:
OK I am confused.
After me-no-dev's response I thought I could use ALL the GPIO's but just setting the pinMode

Looked at the Pin Functions table on https://github.com/esp8266/Arduino
which seems to indicate GPIO6 to 16 are available if set correctly i.e.  Function_4

Text on that page also says "GPIO0-GPIO15 can be INPUT, OUTPUT, INPUT_PULLUP, and INPUT_PULLDOWN. GPIO16 can be INPUT, OUTPUT or INPUT_PULLDOWN."

But are you now saying that is not the case and some GPIO's can never be used as I/O?
please advise and perhaps clarify the github readme.

igrr, 2015-07-09:
This depends on the board/module you use — most of them have a flash chip connected to GPIO6-GPIO11, so unless you specifically make sure your code only runs from RAM, you can't use these pins for other purposes. `pinMode` function will do it's job, but then the chip will not be able to load code for execution.
NodeMCU 1.0 board, for instance, can run in DIO flash mode instead of QIO, which frees up two of these flash interface pins for general use. The same may be done on ESP-201 boards which also have flash interface pins broken out to headers.

drmpf, 2015-07-09:
Ok thanks for that ,
I now have gpio 12 to 14 working INPUT_PULLUP and gpio 16 working INPUT_PULLDOWN
BUT
gpio 15 is still not behaving.  It seems to latch the last value
I have set it as
 pinMode(15,INPUT_PULLUP);
but initially shows up LOW
if I short it to +3V3, it goes HIGH and stays HIGH when I remove the short.
but if I then short it to GND it goes LOW and stays LOW when I remove the short

igrr, 2015-07-09:
There is an external pull down resistor connected to GPIO15 (check your
board schematics), so the behavior you describe makes sense.

drmpf, 2015-07-09:
Actually the schematic I have shows a pullup resistor on GPIO15
https://github.com/OLIMEX/ESP8266/blob/master/HARDWARE/ESP8266-EVB/ESP8266-EVB_Rev_A.pdf
Although a multimeter only reads 300mVs on gpio15
perhaps you are correct and the schematic is in-correct

drmpf, 2015-07-09:
No just measure R3 and one side is 3.3V.

igrr, 2015-07-09:
This is the schematic of your module:
https://github.com/OLIMEX/ESP8266/blob/master/HARDWARE/MOD-WIFI-ESP8266-DEV/MOD-WIFI-ESP8266-DEV_schematic.pdf
It has 2k R9 connected to MTDO pin, and a jumper TDOJP which is marked with
"default 2-3", i.e. pull to ground.
So the pin is pulled via 2k to ground and via 47k to VDD. Not sure why
Olimex designed it this way.

drmpf, 2015-07-09:
Oh. I see that is the schematic for the plugged in module.  
Yes GPIO15 pulled to GND via a 2K on my plug in module.  
Odd.  
Thanks for clarifying that.  Will skip using GPIO15 (and 16)

BTW here is a trick for using GPIO 0 and 2 as inputs.
Normally you can not use them if the input could be shorted to GND on power up, (as it can be in my case with limit switches)

But you can use them by supplying the GND via another GPIO set as output LOW. say GPIO4.

Since this GPIO4 is only made low  AFTER startup it does not interfere with GPIO 0 and GPIO 2 detection and operation during initialization of the ESP8266 module

So you connect your external switch between GPIO0 (or GPIO2) and GPIO4,  Make GPIO4 output LOW in setup and then check GPIO0  for LOW or HIGH

closed 2015-07-09 by drmpf

igrr, 2015-07-09:
That's a neat trick! Thanks!

atrent, 2016-01-31:
_A latere_ question: do you know why Olimex brought out (on the EVB board header) the GPIO 6 to 11 (i.e. SD_CLK to SD_CMD) if they cannot be used? Or they can be used someway? Thanks

me-no-dev, 2016-01-31:
those pins are brought out on just about any ESP board those days. GPIOs 9 and 10 can be used of those ONLY if your flash chip is DIO (EVB has QIO) else none of them can be used because they are the data lines for the flash (that is holding your app, SPIFFS and so on)

Juppit, 2016-01-31:
Maybe it can be used for additional hardware with DIO or QIO mode?

me-no-dev, 2016-01-31:
It really depends on how the board is put together.
If you have DIO chip (nodemcu) then you are OK to use GPIOs 9 and 10 (they do not get to the flash's pins)
If you have QIO chip, regardless in what mode you put it in, GPIOs 9 amd 10 are still connected to the flash chip, so I am not sure what results you will get using those in that case.

To be safe, do not use them :) but who cares to be safe anyway :D

atrent, 2016-01-31:
so the question remains: why Olimex brought them out? ;)

(I'm referring to this EVB board: https://www.olimex.com/Products/IoT/ESP8266-EVB/open-source-hardware)

it would be more useful to have a mirror (since they are available on the UEXT connector) of the other GPIOs

thank you again

me-no-dev, 2016-01-31:
they are not :)
[image]
You can see the SPI pins(GPIO12-15), RX/TX(GPIO1 and 3) and GPIOs 2 and 4

me-no-dev, 2016-01-31:
I am a it lost to why on this schematic GPIO15 is pulled up when it needs to be pulled down for the ESP to boot in either Bootloader or Sketch mode

atrent, 2016-01-31:
I've been cryptic, sorry.
I meant that it would have been more useful to have, on the header (CON3 connector), a mirror of the UEXT lines (i.e. GPIO1,4,12,14,3,2,13,15) instead of SD_*