esp8266/Arduino #521: 08.07.2015 ESP8266Wifi.cpp error undefined reference to `wifi_wps_disable'

tontito, 2015-07-08:
ESP8266WiFi\ESP8266WiFi.cpp.o: In function `wifi_dns_found_callback(char const*, ip_addr*, void*)':
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:577: undefined reference to `wifi_wps_disable'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:579: undefined reference to `wifi_wps_disable'
ESP8266WiFi\ESP8266WiFi.cpp.o: In function `ESP8266WiFiClass::begin(char const*, char const*, int, unsigned char*)':
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:70: undefined reference to `wifi_wps_enable'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:72: undefined reference to `wifi_set_wps_cb'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:72: undefined reference to `wifi_wps_start'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:83: undefined reference to `wifi_wps_disable'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:89: undefined reference to `wifi_wps_enable'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:92: undefined reference to `wifi_set_wps_cb'
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g2d340c7\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp:94: undefined reference to `wifi_wps_start'
collect2.exe: error: ld returned 1 exit status
Derleme sırasında hata oluştu.

igrr, 2015-07-08:
It looks like you have partially updated your installation.

Your message indicates that you are using 2d340c7, which is quite old and shouldn't have WPS functions.
C:\Users\VOLKI\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-804-g*2d340c7*\libraries\ESP8266WiFi\src/ESP8266WiFi.cpp

If you did an update manually by copying files from latest git, you might need to check if you have updated platform.txt and libraries. Also make sure IDE has been restarted.

tontito, 2015-07-08:
Yes partially updated. Nightly build old release. I have Download ZIP and change old files.

tontito, 2015-07-08:
remove platform.txt 
runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf
runtime.tools.esptool.path={runtime.platform.path}/tools

running normally but old release 1.6.5-804 sketch size %67 297kb new sketch size %78 342kb

igrr, 2015-07-08:
That's correct, platform.txt is normally patched by build/build_board_manager_package.sh.
Size went up significantly because of WPS (which pulled libcrypto along as well).

tontito, 2015-07-08:
Thank you. close issue.

closed 2015-07-09 by tontito