esp8266/Arduino #522: OTA and flash size issues

kdjani, 2015-07-08:
@igrr , we were able to use the OTA example that is in the works but were hit with storage out of space problem. 

The board we are using is a nodemcu v0.9 (ESP-12). I believe this board has 4MBytes of flash size

Our sketch size is 249616 bytes. However when using OTATest.ino with the python script we ran into "out of space" errors. 

Here are the repro steps.

1.) Start python server to serve bin file
2.) Test code in arduino to receive the bin file
3.) Update almost works and then fails
4.) Update is failing because of not enough free disk space.

Apparently sketch binary is first copied in to the free space and there isn't enough.
Example:
Current Sketch size: 249616

Free size: 188416

Even the minimum blank skecth is ~20000, so does not fit.

Do you have any suggestions for what we can do get our ourselves unblocked ?

Thanks,
Kandarp

me-no-dev, 2015-07-08:
are you sure you have selected the correct flash size in the Arduino Menu? Default is 512KB

kdjani, 2015-07-08:
@me-no-dev Will the flash size in the Arduino menu also impact the flash size value calculated for Over the air (OTA) firmware upgrade flash size value ?

me-no-dev, 2015-07-08:
yes. it all matters :)

kdjani, 2015-07-08:
@me-no-dev  Thank you! I will give it a shot and get back to you

me-no-dev, 2015-07-08:
keep in mind that the version you have is a bit faulty :) I just pushed an update that should fix it all
https://github.com/esp8266/Arduino/pull/523
make sure you do those changes to your files

kdjani, 2015-07-08:
@me-no-dev. You are a genius!

So changing default size in arduino IDE from 512K to 4M solved the issue :D 
Tried 3 times in a row with different sketches...

Even uploaded itself to itself (recursive)..

Now its showing

Sketch size: 249624
Free size: 798720

Yea!!!!

Works flawlessly actually..

me-no-dev, 2015-07-08:
:) glad you got it running

closed 2015-07-23 by igrr

trananhtan, 2018-03-30:
i have same issue with esp8266 v01. when i update flash via OTA. It say erro 100. not  enough space