esp8266/Arduino #525: Eclipse compilation does not recognize esp library include (eg ESP8266WiFi.h)

far5893, 2015-07-09:
eclipse give this error
../.ino.cpp:7:25: fatal error: ESP8266WiFi.h: No such file or directory

But all header file are not recognized (only Arduino.h).

I'm using eclipse 4.5 under ubuntu , arduino 1.6.4 compile same sketch correctly.

far5893, 2015-07-09:
Create from zero new sketck compile , is existent import sketch from 1.6.4 ide problem

Links2004, 2015-07-09:
have you load the ESP8266WiFi libary?

[image]

[image]

far5893, 2015-07-14:
Ok it works , not so user friendly but wokrs.

Now remain a  :
Function 'snprintf' could not be resolved	WebServices.ino	/WebServices	line 74	

But i can use String object if  i not found what it ask for

Links2004, 2015-07-14:
you need to include "stdio.h"

far5893, 2015-07-15:
No , i have stdio.h , and semms only interface trouble i can compile and upload binary.
[image]

Links2004, 2015-07-15:
ok, you need do the include like:
```cpp
extern "C" {
#include <stdio.h>
}
```
the ```snprintf``` is implemented in C not in CPP, you need to tell the compiler this or the function is not found.
https://github.com/esp8266/Arduino/blob/60efe5caed7b8d14636ae5bcc98a1052b55832b1/hardware/esp8266com/esp8266/cores/esp8266/libc_replacements.c#L77

igrr, 2015-07-15:
I think stdio.h is smart enough to include
#ifdef __cplusplus
extern "C" {
#endif
guards, so that shouldn't be an issue.

Posting the full compiler output would certainly be helpful here.

far5893, 2015-07-15:
It's not compilation issue , only eclipse interface say that snprintf could not be resolved.

Links2004, 2015-07-15:
now i get you problem, you mean this:
[image]
I simply ignore it at the moment, its a problem of the eclipse source indexing.

far5893, 2015-07-15:
Arduino IDE is a fireworks and eclipse is the zar bomb (nuke bomb) for seriously programming ,
i will investigate about indexing , if you have a hint you are welcome.

Links2004, 2015-07-15:
I like your compere :+1: 
i think the problem comes from the many ifdef inside the stdio.h,
based on here the indexer is started may some defines are not set
and then the indexing ignore the snprintf be course its "ifdef out".

odilonafonso, 2017-02-26:
Has this issue been resolved? I'm having the same problem with the eclipse sloeber plugin. 

Compiling a sample from pubsubclient library - mqtt_esp8266.ino:

>     snprintf (msg, 75, "hello world #%ld", value);

is not a sample problem; in the arduino ide this error does not occur.

Modifying to:

>      sprintf (msg, "hello world #% ld", value);

The error no longer occurs.

devyte, 2017-10-20:
@far5893 @Links2004 this issue is ancient. Is it still valid?

devyte, 2018-02-28:
Closing due to lack of feedback.

closed 2018-02-28 by devyte