esp8266/Arduino #528: Unable to use IEEE NaN (Not a Number) Float Values

owendelong, 2015-07-09:
The following code:

```
#include <math.h>
void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Initialized");
  // put your setup code here, to run once:
  float a = 0/0;
  Serial.println("Div/0 completed.");
  float b = 0.03511390;

  Serial.print ("a1 = ");
  Serial.println (a, 8);
  Serial.print ("b = ");
  Serial.println (b, 8);

  if (isnan(a)) 
  {
    Serial.println("True");
  }
  else
  {
    Serial.println("False");
  }
  a = a + b;

  Serial.print ("a2 = ");
  Serial.println (a, 8);

}

void loop() {
  // put your main code here, to run repeatedly:
  delay(100);

}
```

Demonstrates the problem. Note that the div/0 is just a convenient way to generate a NaN value (or should be).

However, the isnan() function is returning False and a is represented as 0.0000000 in the first case and as the same value as b in the second case by the print statements.

It appears that the esp8266 version of the compiler does not correctly implement IEEE standard floating point numbers.

This using Arduino IDE version 1.6.5 on MacOS X and tested on an Adafruit Huzzah ESP8266 breakout board.

owendelong, 2015-07-09:
Since this was a protracted topic of discussion on the Arduino forum some time back, I will attempt to pre-empt that rat-hole here...

I realize one should not deliberately divide by zero. As stated, that's just a convenient way to generate a NaN result. There are many other ways. (I started with a sqrt(-1) and also tried 5/0, both of which caused the board to watchdog reset repeatedly and not move past the arithmetic computation).

Since this board is commonly used for data acquisition and there are many data acquisition situations in which one wants to be able to clearly distinguish that no valid data was collected vs. a potentially valid 0 or any other conceivable number, the ability of IEEE floating point numbers to represent NaN, Infinity, and -Infinity (particularly NaN) is quite useful.

For example, if logging temperatures and the sensor fails or cannot be read for some reason, it is desirable to have a sample at the correct time marked NaN indicating a sensor read problem vs. a systemic problem (missed sample) or incorrect seemingly valid (though potentially absurd) data (e.g. 0.0).

owendelong, 2015-07-09:
More data -- Changing the code to 0.0/0.0 results in the expected behavior, including printing nan.

It appears that float = const int / const int is the problem and this may be valid given that that probably occurs in the following order int(int 0 / int 0) -> 0 -> conversion to float 0.000000 

Since integers are incapable of representing NaN, there's no good answer to that situation.

I will leave the bug open because it seems there is still the issue of integer math (float) = sqrt(-1);
and (float) = (int 5) / (int 0) causing an infinite loop or some other kind of processor reset that appears to be a WDT expiration.

Links2004, 2015-07-10:
the WTD is normal its like an arm processor the ESP8266 has an exception mamagment.
you get an exception for dividing by zero. the exception the leads to an WTD reset, this is normal. 
AVR simply does not have an exception management. 

see here:
https://github.com/esp8266/Arduino/blob/esp8266/docs/exception_causes.md
EXC-CAUSE Code: 6

igrr, 2015-07-10:
Regarding two issues described: integer division be zero leads to an
exception (and that's fine), and sqrt issue was fixed recently (but
downloadable package hasn't been updated yet). I'll try sqrt(-1) with
latest trunk and report back.

owendelong, 2015-07-13:
Thanks. I think once igrr completes his testing and reports back, this can be closed.

igrr, 2015-09-30:
I can confirm that the following sketch prints "nan":

```c++
void setup() {
  Serial.begin(115200);
  Serial.println(sqrt(-1));
}
void loop() {}
```

closed 2015-09-30 by igrr