esp8266/Arduino #529: ESP starts in AP_STA after Wifi.begin although WiFi.Mode(WIFI_STA) using staging package

mvdbro, 2015-07-10:
I've switched to the staging package, because it supports NodeMCU V1.0 and has some important fixes included. But the project that I'm running now suddenly starts in AP_STA mode instead of STA only. Code used is 100% unchanged.

I've have created a simple test sketch to demonstrate the issue:

#include <ESP8266WiFi.h>
void setup()
{
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Serial.println("Config AP");
  WiFi.softAP("ESP_DEMO", "secure");

  Serial.println("Set to STA mode");
  WiFi.mode(WIFI_STA);

  Serial.println("Wifi begin");
  WiFi.begin("MYAP", "********");
}

output using package 1.6.4-673-g8cd3697 (may 22nd)
=========================================
Config AP
mode : sta(18:fe:34:a6:d6:84) + softAP(1a:fe:34:a6:d6:84)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1
add if1
bcn 100
Set to STA mode
bcn 0
del if1
mode : sta(18:fe:34:a6:d6:84)
Wifi begin

output using package 1.6.5-804-g2d340c7 (june 26th)
=========================================
Config AP
scandone
del if0
mode : softAP(1a:fe:34:a6:d6:84)
Set to STA mode
bcn 0
del if1
usl
sul 0 0
mode : sta(18:fe:34:a6:d6:84)
add if0
Wifi begin
mode : sta(18:fe:34:a6:d6:84) + softAP(1a:fe:34:a6:d6:84)

as you can see (last line) it now starts sta + softAP ! after Wifi.begin command....

noahwilliamsson, 2015-07-13:
The code in [hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp](https://github.com/esp8266/Arduino/blob/32910546e9f4914e1f1b127e402ece3db58751c0/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp) is definitely buggy in its current state.

Once you call `WiFi.softAP()`, the code internally sets the flag `_useApMode = true`.  This flag is then used in multiple places to decide what the actual desired mode should be when turning on station mode (e.g. `WiFi.begin()`) or turning it off (e.g. `WiFi.disconnect()`). 

For instance, if you have called `WiFi.softAP()` once, the flag will be true, and `WiFi.begin()` will consider it and set the mode to `WIFI_AP_STA` (as opposed to `WIFI_AP` if the flag wasn't set, which happens to be the default).

Unfortunately there is no corresponding `_useApMode = false`. Furthermore, the flags `_useApMode` and `_useClientMode` are not updated to reflect the reality after changing modes with `WiFi.mode()`.

Until this is fixed, I suppose you could either avoid calling `WiFi.softAP()` if you're not going to actually use it, or call `WiFi.mode(WIFI_STA)` after `WiFi.begin()` if you've called `WiFi.softAP()`.

martinayotte, 2015-07-21:
Actually, with the current version, _useApMode is always false from the constructor. 
But effectively, those flags are not reflecting the real mode. 
Also, I found a strange thing related to that : If we have an AP serving a web page to show existing network for future STA connection, the scanNetworks() will stop the AP mode and the current client will be disconnected.

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004

gosewski, 2016-03-03:
I think, in this case the main problem with switching WiFi mode in taht the used password "secure" do not fullfils the minimal length requirement which is 8 characters. In such case WiFi.softAP function returns false and the mode remains unchanged.

sarwadenj, 2016-09-09:
on reboot esp8266 starts in which mode.
AP or station mode