esp8266/Arduino #533: Automatic programming

lrmoreno007, 2015-07-10:
I think you must add a note to the Readme.md in the "esp to Serial" hardware section for clarify and improve the stability of the module.

***Oblogatory: GND must be connected between ESP8266 and Serial Interface.
*** Optional: 3.3V between ESP8266 and Serial Interface 
 
I know it's a silly problem but I was having many problem with it and a simple note can help.

Regards and excuse for my english.

me-no-dev, 2015-07-10:
This A and B of connecting electronic equipment regardless of the bus/protocol used.
Without common ground (except when optocouplers are used) you can not expect anything to work, because every signal level is measured relative to ground.
So I am not sure these need to be written. If you do not know these things, it's maybe better to hit the wall a couple of times and really learn it, rather than just follow instructions on how to hook up your ESP to the serial port, then your Arduino to the SPI port.. so on.
I hope you do not get offended by my comment and understand what I'm trying to say :)

me-no-dev, 2015-07-10:
As for the 3.3V rail, this really depends on your implementation as it is used to transfer power between the devices having common ground (mandatory for power and signal) and 3.3V rail.
One example when it's needed is when your ESP is not powered by anything else and your Serial adapter can provide enough power for the ESP (my FTDI does not)
Another example is when you have your ESP powered, but your Serial adapter is not powered from the USB
If both devices are powered, it's not a very good idea to connect the power rails together, because it may lead to smoke on either end.

drmpf, 2015-07-11:
Ground loops can also be a problem,  in general you should have only 1 (one) ground connection point.  Since the ESP module connects signal and power grounds together.  It should be the one common point.

So common cases are
1) USB supplying power and program connection, ESP board has built in 5V to 3V3 supply -- connect  USB GND
2) Separate isolated 3V3 supply (transfromer in separate USB 5V supply feeding 5V to 3V3 supply)  -- connect USB GND and 3V3 supply GND
3) Non-isolated supply, e.g. 5V to 3V3 supply supplied from computer's USB supply -- Only connect supply GND and do not connect USB gnd

lrmoreno007, 2015-07-15:
I'm not offended, I'm grateful. I'm learning very fast with you all.

Now I've a new problem. With the released version I can open Serial Monitor, upload the sketch, etc without unplug RST and DTR from the CP2101, everithing run OK.

Recently I install the staging version and I can upload the sketch with no problem, but when I open Serial Monitor --->  RST and GPIO0 are activated (Ground) and while I don't close Serial Monitor both are activated, forcing me to unplug RST and DTR from CP2101.

I've RST, CH_PD, GPIO0 and GPIO2 pulled up, GPIO15 pulled down.

What can I do?

JeroenBeemster, 2015-07-21:
It is the same issue as #22.
This issue is closed, but not solved. Will this issue ever be solved. Or should we find another way.
BTW: This issue should be a common problem for a lot of users. Is no one using serial monitor or is no one using DTR? [Currently I have moved back to the 2 button solution , but I do not like it at all

Links2004, 2015-07-21:
its a problem of the arduino IDE from arduino.cc
the options 
```
generic.serial.disableDTR=true
generic.serial.disableRTS=true 
```
are not integrated there.

i use realterm for testing there you have the full control.
you can give them so cmdline as well, i use this in my makefile:
```
$(REALTERM_BIN) -BAUD=$(SERIAL_BAUD)  -PORT=$(SERIAL_PORT) -COLS=100 -DISPLAY=1 -SCROLLBACK=550 -ROWS=50 -OPEN=1 -DTR=0 -RTS=0 
```

lrmoreno007, 2015-07-22:
Thanks!

I'm using CoolTerm now and OK. And I have OTA running very good. I modify the Python script and now try a new connection every 5sec and my ESP-12 try to OTA every restart along 10sec.

It's the python script:

#!/usr/bin/python
# -*- coding: cp1252 -*-
#
# this script will push an OTA update to the ESP
#
# use it like: python ota_server.py <ESP_IP_address> <sketch.bin>
#
# on the ESP side you need code like this: https://gist.github.com/igrr/43d5c52328e955bb6b09 to handle the update
#

# Need PYTHON 2.7
 
import socket
import sys
import os

def serve(remoteAddr, filename):
  serverIP = '192.168.1.35'
  serverPort = 48266
  clientPort = 8266

  while True:
    socket.setdefaulttimeout(5)
    sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    #sock.setsockopt(SOL_SOCKET, SO_REUSEADDR, 1)
    server_address = (serverIP, serverPort)
    sock.bind(server_address)
    sock.listen(1)
    sock2 = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    remote_address = (remoteAddr, clientPort)
    content_size = os.path.getsize(filename)
    message = '%d %d %d\n' % (0, serverPort, content_size)

    print >>sys.stderr, 'Esperando para la conexion'
    sent = sock2.sendto(message, remote_address)
    try:
      connection, client_address = sock.accept()
      print >>sys.stderr, 'OK'
      try:
        f = open(filename, "rb")
        while True:
          chunk = f.read(4096)
          if not chunk:
            break
          print >>sys.stderr, 'Enviando %d' % len(chunk)
          connection.sendall(chunk)
        print >>sys.stderr, 'Hecho!'
        connection.close()
        f.close()
      finally:
        return 1
        break
    except socket.timeout:
      print >>sys.stderr, 'Error'
      sock.close()
      sock2.close()
    finally:
      pass

def main(args):
  if len(args) > 2 :
      return serve(args[1], args[2]) 
  elif len(args) > 1:
      print "No ha introducido uno de los argumentos"
      print "Introduzca como argumentos la IP del Cliente y la direccion del fichero"
      print "Por ejemplo: python ota_server 192.168.1.1 c:\user.bin"
  elif len(args) > 0:
      print "Introduzca como argumentos la IP del Cliente y la direccion del fichero"
      print "Por ejemplo: python ota_server 192.168.1.1 c:\user.bin"
          #      return serve('','')
 
 
if __name__ == '__main__':
    sys.exit(main(sys.argv))

igrr, 2015-07-22:
@JeroenBeemster 
You may try another reset circuit mentioned by @probonopd in this comment: https://github.com/esp8266/Arduino/issues/480#issuecomment-122993574
It seems to be tolerant to DTR/RTS manipulation by the serial monitor.

closed 2015-07-22 by lrmoreno007

JeroenBeemster, 2015-07-23:
thanks that did the job. 
I made a breadboard version of electronic schema to make it easer to read.
http://www.arduinesp.com/blink