esp8266/Arduino #536: Noise on I/O lines

skandragon, 2015-07-11:
I suspect this is a hardware issue, but when hardware and software are so intertwined, sometimes it's hard to tell if this is a hardware issue or a hardware setup issue.

I have this sketch running:

```
void setup() {
  pinMode(4, OUTPUT);
  digitalWrite(4, LOW);

  pinMode(5, OUTPUT);
  digitalWrite(5, LOW);

  pinMode(12, OUTPUT);
  digitalWrite(12, LOW);

  pinMode(13, OUTPUT);
  digitalWrite(13, LOW);

  pinMode(14, OUTPUT);
  digitalWrite(14, LOW);

  pinMode(15, OUTPUT);
  digitalWrite(15, LOW);

  pinMode(16, OUTPUT);
  digitalWrite(16, LOW);
}

void loop() {
  digitalWrite(13, HIGH);
  delay(100);
  digitalWrite(13, LOW);
  delay(100);
  digitalWrite(4, HIGH);
  delay(100);
  digitalWrite(4, LOW);
  digitalWrite(5, HIGH);
  delay(100);
  digitalWrite(5, LOW);
}
```

Normally, I see this as rock solid:

[image]

Sometimes I see small, extremely short duration glitches:

[image]

I saw these glitches constantly when I set the various pins to INPUT_PULLUP or INPUT_PULLDOWN, and also saw that the pins voltages did not actually match if it was pulled up or down.

I am using a National Instruments VirtualBench device, and with an Arduino running at 8Mhz also at 3.3v, I se no glitches when running the same sketch.

me-no-dev, 2015-07-11:
only pin 16 has pulldown, the rest have pullups only.
pin 15 is probably pulled down by a resistor and the ESP has weak pullups so depending on the pulldown resistor, you will not see it being pulled up.
Otherwise I have no clue on the glitches you see.

wireddown, 2015-07-11:
*Sorry for [creepin'](http://stream1.gifsoup.com/view1/1079771/sneaky-cat-o.gif) (I found this issue with a search for VirtualBench)*

@skandragon -- If you update your VirtualBench firmware, you will be able to configure more settings in the logic analyzer, like:

* independently set the digital sample rate up to 1 GHz
* set the digital threshold level
* group digital lines together in buses and decode SPI and I2C communication

[image]

While these new capabilities won't help with this particular issue, perhaps you'll use them later. The update is free -- select *Search for Firmware Update* from the File menu.

skandragon, 2015-07-12:
I've done that, thanks!  I only started using this tool last night.

Makuna, 2015-07-13:
What exactly are your trying to do here? 

While the AVR has no interrupt handlers or tasks running by default, the esp8266 does.  Your code could "glitch" on the AVR if you did have an interrupt handler that happened to trigger during one of your delays.

The esp8266 MCU is meant to be the WiFi service CPU, so it has "background" work it already does at the core and will switch from your code when it needs to.  You will often see this in the first 10 seconds of booting.

An esp8266 specific issue also is that code in flash can get delayed as it is loaded/run, so timing sensitive code should be in functions that are marked with ram attribute.

In general, to get accurate timing, you will need to wrap your code in noInterrupts() / interrupts(); but this should be around the smallest/shortest running amount of code as possible.

The example you have using delays() is not a sustainable model of programming, especially for this case.  While it is simple, you would never want to wrap them in noInterrupts() / interrupts().

skandragon, 2015-07-13:
The code I have is just test code.  However, I think I found the root cause of the glitches, and it is traceable back to insufficient power from the FTDI to power the device, even when not using the WiFi.  When I use the bench 3.3 VDC power, the glitches seem to be gone.  Closing this unless it re-appears.

closed 2015-07-13 by skandragon