esp8266/Arduino #538: How i can compile lastest version IDE in Windows 8.1?

Lstt2005, 2015-07-11:
Dear friends! I am notice in "coding"...Can i compile lastest version of ESP 8266 Arduino IDE form git on WIndows 8.1? Git for Windows is install, download repository local, so, what's next? ))

h4rm0n1c, 2015-07-13:
It's a little bit involved, but the instructions at https://code.google.com/p/arduino/wiki/BuildingArduino work.

You may need to look carefully in to what and how to get the 32 Bit Windows JDK working with Cygwin, but that's easy enough to find on Google.

oscaracena, 2015-07-20:
Hi, 

Please, take note that this is an Issue Traker, it is not conceived for asking general questions, but to report problems in the code or common procedures related with this project.

Thanks.

closed 2015-07-30 by igrr