esp8266/Arduino #546: Using getVcc()  =wont compile

tytower, 2015-07-12:
"ESP.getVcc() may be used to measure supply voltage. ESP needs to reconfigure the ADC at startup in order for this feature to be available. Add the following line to the top of your sketch to use getVcc:
ADC_MODE(ADC_VCC);
TOUT pin has to be disconnected in this mode.
Note that by default ADC is configured to read from TOUT pin using analogRead(A0), and ESP.getVCC() is not available."

I tried to use this line "  11  ADC_MODE(ADC_VCC);" but it does not compile
 
"readvdd33.ino:11:9: error: expected constructor, destructor, or type conversion before '(' token
expected constructor, destructor, or type conversion before '(' token"

So can you expand or correct this in the Readme please

Duality4Y, 2015-07-12:
i assume the 11 part isn't part of the code ?

tytower, 2015-07-13:
Line number to make it easy to understand by the reader . Didn't work hey?

.................................................................................

Links2004, 2015-07-13:
the macro need to be outside of any function.
```cpp

#include <Arduino.h>

ADC_MODE(ADC_VCC);

void setup(void) {
}

void loop(void) {
}
```

tytower, 2015-07-13:
You are just guessing now or purposly ignoring it .
#include <Arduino.h>  is standard on all Arduino programs and does not need to be there 

ADC_MODE(ADC_VCC);  Will not compile on the latest Arduino 1.6.5

----------------------------------------------------------------------------On Mon, 13 Jul 2015 01:59:56 -0700
Markus <notifications@github.com> wrote:

> #include <Arduino.h>

-- 
Yahoo <tytower@yahoo.com>

igrr, 2015-07-13:
I suppose the real reason is that documentation we have here on Github
tracks latest development version, not the one which is delivered via
boards manager package. Some of the changes reflected in Readme are simply
not available in the version you are using, ADC_MODE being one of them.

There is a plan to upload all documentation to a different site, where it
will be kept stable between releases. I started working on this yesterday
in
https://github.com/esp8266/Arduino/commit/32910546e9f4914e1f1b127e402ece3db58751c0,
so documentation update should be complete in a couple of days.

tytower, 2015-07-13:
Oh OK I see now . So will they come through on the next boards update? or am I better to install it directly from your site?

canadaduane, 2015-07-22:
@igrr you are amazing. Keep up the great work.

tytower, 2015-08-02:
I got down the latest version (staging) yesterday but still the thing will not compile with the same error so I don't seem to understand what you are doing here . The Readme has almost disappeared since last time so where do I find what works and what doesn't now ?

tytower, 2015-08-02:
Read me is found now at https://github.com/esp8266/Arduino/tree/32910546e9f4914e1f1b127e402ece3db58751c0

ADC_MODE(ADC_VCC);  will still not compile sorry

Links2004, 2015-08-02:
Stable readme:   http://arduino.esp8266.com/stable/doc/reference.html
Staging readme: http://arduino.esp8266.com/staging/doc/reference.html
(see https://github.com/esp8266/Arduino#available-versions)

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004