esp8266/Arduino #548: lmac.c 662 ?

mvdbro, 2015-07-13:
Have been getting this mysterious message on serial debug output before wdt reset:

lmac.c 662

The message "lmac.c 662" is not some debug print from my own sketch. Have seen it more than once now, so it doesn't seem to be a random glitch on serial...

Does anyone have an idea what's it about ?
(and how to prevent would be even better...)

unfoundbug, 2015-07-17:
Not got a solution as of yet, but I get this as well using the SDK, V1.2 so it's not specific to the arduino branch
Sometimes this actually pushes out as an exception rather than the lmac line.

Not quite sure yet whether its caused during the amount of time with interrupts disabled or in transmission, will be working on this later.

sticilface, 2015-07-28:
I'm getting a lot of these...  and with a simplified sketch that is just running neopixelbus and an animation with OTA enabled.  I still get them without wifi and OTA, but just the lmac.  I don't get the stack trace very often.  However, locating  40103bd the function has lots of lmac references in it, so it looks pretty similar. 

here is sketch
http://pastebin.com/Pambzjwn

here is the elf dump
https://dl.dropboxusercontent.com/u/12627633/OTAdump.txt

here is the stack

```
Fatal exception (28):
epc1=0x40103bdb, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000024, depc=0x00000000

Exception (28):
epc1=0x40103bdb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000024 depc=0x00000000

ctx: cont 
sp: 3ffea710 end: 3ffeaa10 offset: 01a0

>>>stack>>>
3ffea8b0:  00040000 3ee85010 53002200 3ffec640  
3ffea8c0:  4000050c 3fffc278 4010550c 3fffc200  
3ffea8d0:  00000022 000000ff 3e9c9c9d 402025d8  
3ffea8e0:  40203881 00000030 0000001e ffffffff  
3ffea8f0:  4020387c 00000020 3fff2ea0 00000004  
3ffea900:  399428ea 00000000 00000000 60000308  
3ffea910:  00000080 00000040 000000c7 3fff2af8  
3ffea920:  3ffe9950 3fffdc20 3ffeaa3c 00000030  
3ffea930:  00000007 00000006 402017fa 3ffe99f0  
3ffea940:  3fff2808 00000008 00000000 0000003b  
3ffea950:  003bc300 00000000 3ffeaa34 40202eb0  
3ffea960:  63000000 3e409c00 3ffe8b62 3f800000  
3ffea970:  3e70dc37 3ef0dc37 3fff4368 40202450  
3ffea980:  3f1b5b00 3fff3134 402022bc 402023dc  
3ffea990:  3fff2818 00000000 402022bc 00000001  
3ffea9a0:  00000023 0000008d 3fff2bf8 402037be  
3ffea9b0:  00000009 00000009 3fff2bf8 4020375c  
3ffea9c0:  00101abc 432f0000 3fff39bc 00000000  
3ffea9d0:  00000000 3fffdc20 3ffe9950 3fff2af8  
3ffea9e0:  3fffdc20 00000000 3ffe9650 40202c5a  
3ffea9f0:  3fffdc20 00000000 3ffeaa34 40201876  
3ffeaa00:  00000000 00000000 3ffe99f0 4010046c  
<<<stack<<<
```

Duality4Y, 2015-07-29:
I notice this too, and it shows up more on my network then anywhere.

sticilface, 2015-07-31:
Using the latest commits, these have decreased significantly for me, almost completely.

sticilface, 2015-08-10:
Thought I'd throw some more info up here in the hope of a solution.  I'm getting this rather frequently...  here is the offending function from the elf dump.... with the offending address in bold
```
40103c8c <lmacProcessAckTimeout>:
40103c8c:	f0c112        	addi	a1, a1, -16
40103c8f:	0109      	s32i.n	a0, a1, 0
40103c91:	feef85        	call0	40102b8c <lmacIsActive>
40103c94:	02dc      	bnez.n	a2, 40103ca8 <lmacProcessAckTimeout+0x1c>
40103c96:	fbc821        	l32r	a2, 40102bb8 <lmacIsIdle+0x18>
40103c99:	fbc831        	l32r	a3, 40102bbc <lmacIsIdle+0x1c>
40103c9c:	e2a442        	movi	a4, 0x4e2
40103c9f:	f7bd01        	l32r	a0, 40101b94 <ets_timer_setfn+0x1c>
40103ca2:	0000c0        	callx0	a0
40103ca5:	ffff06        	j	40103ca5 <lmacProcessAckTimeout+0x19>
40103ca8:	fbb841        	l32r	a4, 40102b88 <call_user_start_local+0x420>
40103cab:	000422        	l8ui	a2, a4, 0
40103cae:	a438      	l32i.n	a3, a4, 40
40103cb0:	30c442        	addi	a4, a4, 48
**40103cb3**:	9308      	l32i.n	a0, a3, 36
40103cb5:	b02220        	addx8	a2, a2, a2
40103cb8:	0008      	l32i.n	a0, a0, 0
40103cba:	a02240        	addx4	a2, a2, a4
40103cbd:	1060e7        	bbci	a0, 14, 40103cd1 <lmacProcessAckTimeout+0x45>
40103cc0:	030c      	movi.n	a3, 0
40103cc2:	070242        	l8ui	a4, a2, 7
40103cc5:	064242        	s8i	a4, a2, 6
40103cc8:	094232        	s8i	a3, a2, 9
40103ccb:	ffe485        	call0	40103b14 <lmacMSDUAged+0x238>
40103cce:	000106        	j	40103cd6 <lmacProcessAckTimeout+0x4a>
40103cd1:	040c      	movi.n	a4, 0
40103cd3:	ffc105        	call0	401038e4 <lmacMSDUAged+0x8>
40103cd6:	0108      	l32i.n	a0, a1, 0
40103cd8:	10c112        	addi	a1, a1, 16
40103cdb:	f00d      	ret.n
40103cdd:	000000        	ill
```

Duality4Y, 2015-08-10:
i'll try and do a pull and recompile

DanielCasner, 2015-08-13:
There's an interesting note in the v1.3.0 SDK documentation now which says that interrupts may not be disabled for more than 10us. Since WiFi interrupts can't preempt other interrupts this also means that no interrupt handler should run longer than 10us or the wifi will screw up with a lmac error.

igrr, 2015-09-30:
OTA related part of this issue is fixed in staging.

closed 2015-09-30 by igrr