esp8266/Arduino #550: Possible to wake up from deepsleep with hardware pin?

BJvDL, 2015-07-13:
Is it possible to wake up from deepsleep with an external signal, pin?
In a battery powered circuit I want to only wake up the ESP when a switch opens or closes with a rising or faling edge and go to sleep again.
Is it also possible to have set a deepsleep timer in parallel?

I want to detect the change of a switch, wake up and send its statús to a MQTT broker and go to sleep again. If the switch doesn't change for a long time then the timer should wake up the ESP and send the status to say it is still alive.

iamthemadz, 2015-07-13:
The deepsleep just resets the device on wakeup, so I would imagine that just resetting the device using its RST pin would work just fine.

Duality4Y, 2015-07-13:
isn't this more for in a forum then an issue on github ?

iamthemadz, 2015-07-13:
Probably, but its here now. May as well answer the question and point in the right direction for his next one.

duncan-a, 2015-07-14:
Bas

To achieve what you want from the switch, I think you would need to create
a low-going pulse to the Reset pin of the ESP8266, as you then want to
check the switch state - that would need to be done in hardware.

As Robert said, the problem would be better posted on a forum, where you
may receive more complete replies.

chaeplin, 2015-07-15:
* RST : active low. https://github.com/esp8266/esp8266-wiki/wiki/Hardware_versions

* ESP.deepSleep(microseconds, mode) will put the chip into deep sleep. mode is one of WAKE_RF_DEFAULT, WAKE_RFCAL, WAKE_NO_RFCAL, WAKE_RF_DISABLED. (GPIO16 needs to be tied to RST to wake from deepSleep.) https://github.com/esp8266/Arduino

* deepsleep with timer example(GPIO 16) : https://github.com/chaeplin/esp8266_and_arduino/blob/master/_05-two-ds18b20/_05-two-ds18b20.ino

* deep sleep with external signal (using pro mini) : https://github.com/chaeplin/esp8266_and_arduino/tree/master/_01-door-alarm

closed 2015-07-30 by igrr