esp8266/Arduino #555: WDT

asetyde, 2015-07-15:
I need control reboot operation with wdt .. but all information are deleted without a reason ... Any news ?

If i can't use wdt , to reboot my nay 1.0 how can i do ? i've tried 

ESP.wdtEnable(WDTO_8S);
 and after put a while(1) , but in serial shows wdt rest but board not reboot

Links2004, 2015-07-15:
espressif removed the wdt access from SDK.
the current SDK only gives access to what thy called a "software wdt" so not very useful.
if you only want to reboot the ESP try ```ESP.restart();```

igrr, 2015-07-15:
If you see `wdt reset` in serial but the board doesn't reboot, it means some of the bootstrapping pins (GPIO0, GPIO2) are not in the correct state. Make sure GPIO0 and GPIO2 are pulled high.

asetyde, 2015-07-15:
tommorow i ll try , probably gpio0, 
software wdt work so bad , more control is appreciated

igrr, 2015-07-16:
There are two WDT timers actually: software one, attached to a sorta-high priority timer interrupt, and a hardware one. If software WDT doesn't fire, hardware WDT will. When you see `wdt reset` line in serial console, it means that hardware WDT has fired.

If you simply need to reboot your module, use `ESP.restart()` function.

asetyde, 2015-07-16:
ok , but before with arduino nano , i 've used avr wdt function , you can set 8ms as time , you enable use of wdt hw and every times your loop time elapse 8ms if you not use wdt reset , system reboot , this i need . But if not possible to preserve system from freeze , i will you use Esp.restart

igrr, 2015-07-16:
WDT is already enabled by default.

Software WDT will fire in about one second, and if it doesn't (say you are in a recursive loop, or an infinite loop inside an interrupt) then the hardware WDT will fire in about 5 seconds.

We don't have a function to change WDT interval at the moment.

asetyde, 2015-07-16:
Ahn ok , now i understand ... Because before i 've see in code this key point

[image]

[image]

asetyde, 2015-07-16:
[image]

asetyde, 2015-07-16:
i thought i can put time and reset every times

asetyde, 2015-07-16:
please last question , what is wdtFeed ??

asetyde, 2015-07-16:
i use ESP.restart(); but system not restart 
[image]

asetyde, 2015-07-16:
System start ... and eprom init / read are my sysoutprint

asetyde, 2015-07-16:
normal system , not rebooted is this  ( before EEPROM INTIT use Esp.restart(); command)
[image]

igrr, 2015-07-16:
Please share your sketch source code. Also please mention which version of
the boards manager package you are using.

asetyde, 2015-07-16:
i've no authorization to do share code , my library manager is 
[image]

asetyde, 2015-07-16:
But i simply change my struct , copy on eprom (virtual eprom in this board) and after call ESP.restart(); to reload with new data

igrr, 2015-07-16:
What is the boards manage package version you use? (not library manager)

If you can't share you code, please create a sketch which reproduces the
issue.

asetyde, 2015-07-16:
[image]
Sorry ok this board , and soon as i can i post sketch

asetyde, 2015-07-16:
Sketch

#include <SPI.h>
#include <SD.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <WiFiUdp.h>
#include <String.h>
#include <EEPROM.h>
#include <esp8266_peri.h>
#include <virtuabotixRTC.h>

/********************************************************/

/*System Variables*/
int value = 0 ;
boolean flag = false ;
boolean flagm = false ;
byte dataarr[30][1];
int dataarray = 0;
const char* ssid = "";
const char* password = "";
int addr = 0 ;

/*DEBUG*/
boolean debug = true;

struct memory {
	char systemversion[7] = "IL1.0B" ;
	boolean isrtc = false ; // v or f
	char ip[16] ;
	char arduinoName[10] = "nessuno" ;
	byte mac_Address[6] = {0xBA, 0xBA, 0xBA, 0xBA, 0xBA, 0xBA};
	int bar = 0 ;
	int manbar = 0 ;
	unsigned int portconfig = 100000000 ;
};

/********************************************************/

/*Network Variables*/
IPAddress ipAddress; // save the ip for check in the code
IPAddress bannedip;  // banned ip to protect communication                  
/********************************************************/

/*Object routine*/

memory mind  ; // be created
//virtuabotixRTC myRTC(5, 6, 4); //Creation of the Real Time Clock Object SCLK -> 6, I/O -> 7, CE -> 8

/**************************************************************************************************************/

void setup() {
	
	
	
	// PIN INITIAL CONFIGURATION
	
	
	pinMode(16, OUTPUT);                         //     GPIO NODEMCU SCHEMATICS
	pinMode(5, OUTPUT);                          //
	pinMode(4, OUTPUT);                          //     Rele8  ln1  ln2  ln3  ln4  ln5  ln6  ln7  ln8 
	pinMode(3, OUTPUT);                          // 
	pinMode(1, OUTPUT);                          //     cable  red  ora  yel  gre  blu  vio  gry  whi
	pinMode(14, OUTPUT);                         // 
	pinMode(12, OUTPUT);                         //     ndmcu  d6   d7   d5   d10  rx/d9  d2   d1   d0
    pinMode(13, OUTPUT);                         //

    testall();  // diagnostic test system
    
    // watchdog
     ESP.wdtEnable(WDTO_8S);
	
    // serial 
    if (debug == true){
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    }
	
	
	// EEPROM capabilities
	EEPROM.begin(4096);

    // SYSTEM START
    Serial.println(F("Start"));

    //READ EPPROM UPLOADING CONFIGURATION INTO TEMPORALLY STRUCT MIND - AUTOMATIC
      readeeprom();
      checkintegrity();

  
	Serial.println("System start...loading wifi connection");
	// Connect to WiFi network
	WiFi.begin(ssid, password);
	//set static ip part
	WiFi.config(IPAddress(192,168,1,118), IPAddress(192,168,1,1), IPAddress(255,255,255,0));
	
	// Wait for connection
	while (WiFi.status() != WL_CONNECTED) {
		delay(200);
	}
	
  

    delay(100);

    if (debug == true) {
	Serial.print(" - SYSn : ");
    Serial.println(mind.arduinoName);
    Serial.print (" - SYSv : ");
    Serial.println(mind.systemversion);
    Serial.print (" - STRUCT SIZE : ");
    Serial.println(sizeof(mind));
	Serial.print (" - FREE HEAP : ");
	Serial.println(ESP.getFreeHeap());
    Serial.print(" - CPU : ");
    Serial.println(ESP.getCpuFreqMHz());
	
  }
}//end setup

// CONTROL CYCLE

void loop() {
  
  //ethOpSc();
 // ethOp();
 // ethStSc();
 // ethSt();

  sensor();
  
 
//  system();

}/*end LOOP*/

void readeeprom() { //- test ok

  
  delay(100);
  eeprom_read_block((uint8_t *)&mind,0, sizeof(mind));
  if (debug == true) {
    Serial.println(F("EEPROM READ"));
  }
  delay(100);

}

void writeconfig() { // system eeprom writing function - test ok

	Serial.println(F("EEPROM  WRITE "));
	eeprom_write_block((uint8_t *)&mind,0, sizeof(mind));
	delay(500);
	if (debug == true) {
		Serial.println(F("ok write it"));
	}
}

void checkintegrity() { //- test ok

  char systemversion[7] = "IL1.0B" ;

  if (  mind.systemversion[2] != systemversion[2] || mind.systemversion[4] != systemversion[4] || mind.systemversion[5] != systemversion[5] ) {
     
	 memory recovermind  ; // be created to recover  
	 
	  
    if (debug == true) {
      Serial.println("EEPROM  INIT ");
    }
    eeprom_write_block((uint8_t *)&recovermind,0, sizeof(recovermind));
	Serial.println("EEPROM  INIT ");
	delay(2000);
	ESP.restart();
  }
  else {
    Serial.println("ok check integrity ");
  }

}

void restartMachine(byte code) {
  Serial.println(F("Attention: Reboot Machine"));
  Serial.println(code);
  while (1) {
  }
}

void eeprom_read_block(uint8_t * data, uint32_t address, size_t len){
	int i;
	for(i=0; i<len; i++){
		data[i] = EEPROM.read(address+i);
	}
}

void eeprom_write_block(uint8_t * data, uint32_t address, size_t len){
	int i;
	for(i=0; i<len; i++){
		EEPROM.write(address+i, data[i]);
	}
	EEPROM.commit();
}
  
  void testall() {
	  
	     	
	digitalWrite(16, HIGH);
	digitalWrite(5, HIGH);
	digitalWrite(4, HIGH);
	digitalWrite(3, HIGH);
	digitalWrite(1, HIGH);
	digitalWrite(14, HIGH);
	digitalWrite(13, HIGH);
	digitalWrite(12, HIGH);
	
	  	delay(500);
	  	digitalWrite(16, LOW);
		  delay(1000);
	  	digitalWrite(5, LOW);
		  delay(1000);
	  	digitalWrite(4, LOW);
		  delay(1000);
	  	digitalWrite(3, LOW);
		  delay(1000);
	  	digitalWrite(1, LOW);
		  delay(1000);
	  	digitalWrite(14, LOW);
		  delay(1000);
	  	digitalWrite(13, LOW);
	  	delay(1000);
        digitalWrite(12, LOW);
        delay(9000);
		
	digitalWrite(16, HIGH);
	digitalWrite(5, HIGH);
	digitalWrite(4, HIGH);
	digitalWrite(3, HIGH);
	digitalWrite(1, HIGH);
	digitalWrite(14, HIGH);
	digitalWrite(13, HIGH);
	digitalWrite(12, HIGH);
	  	
  }

asetyde, 2015-07-16:
[image]

asetyde, 2015-07-16:
on serial this ...
[image]
another quest , but there is a rtc ? how can it be use ? ... board say 2013 but has something

asetyde, 2015-07-16:
my is node mcu 1

asetyde, 2015-07-16:
i put a ESP.wdtDisable(); on setup function
NOW REBOOT !! :?  ... I give external power and put a restart at the end of print .. this is solution ?

asetyde, 2015-07-16:
Opening port
Port open
Start
EEPROM READ
ok check integrity 
System start...loading wifi connection
scandone
reconnect
rm match
scandone
add 0
aid 2
cnt 

connected with Network Office, channel 6
ip:192.168.1.118,mask:255.255.255.0,gw:192.168.1.1
 - SYSn : nessuno
 - SYSv : IL1.0B
 - STRUCT SIZE : 748
 - FREE HEAP : 25936
 - CPU : 80
REBOOT
rm match
del if0
bcn 0
del if1
usl
sul 0 0

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0x48
ho 0 tail 12 room 4
load 0x3ffe8000, len 1548, room 12 
tail 0
chksum 0xb4
load 0x3ffe8610, len 2812, room 8 
tail 4
chksum 0x83
csum 0x83
rlüStart
EEPROM READ
ok check integrity 
System start...loading wifi connection
scandone
reconnect
rm match
scandone
add 0
aid 2
cnt 

connected with Network Office, channel 6
ip:192.168.1.118,mask:255.255.255.0,gw:192.168.1.1
 - SYSn : nessuno
 - SYSv : IL1.0B
 - STRUCT SIZE : 748
 - FREE HEAP : 25936
 - CPU : 80

asetyde, 2015-07-26:
NODEMCU 1 ESP12E
Sorry but something might not work ! It's important that ESP.restart(); work every time also his wdt ... something reboot something not but i not understand diagnostic info data
after load code , i send via udp a reboot command but :
--------------- in this case reboot fail 
Opening port
Port open
Start
scandone
EEPROM READ
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 
ok check integrity 
System start...loading wifi connection

connected with Network Office, channel 11
ip:192.168.1.118,mask:255.255.255.0,gw:192.168.1.1
 - SYSn : nessuno
 - SYSv : IL1.0C
 - STRUCT SIZE : 2764
 - FREE HEAP : 22552
 - CPU : 80
REBOOT
rm 0
pm close 7 0 0/15416313
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
--------------------- nothing work and i must cut power

in this case after load code .. i test and work 
-----------------
Opening port
Debug off
Port failed to open
Port open
EEPROM READ
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 
ok check integrity 
System start...loading wifi connection

connected with Network Office, channel 11
ip:192.168.1.118,mask:255.255.255.0,gw:192.168.1.1
 - SYSn : nessuno
 - SYSv : IL1.0C
 - STRUCT SIZE : 2764
 - FREE HEAP : 22552
 - CPU : 80
REBOOT
rm 0
pm close 7 0 0/28766716
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Start
EEPROM READ
ok check integrity 
System start...loading wifi connection
scandone
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 

connected with Network Office, channel 11
ip:192.168.1.118,mask:255.255.255.0,gw:192.168.1.1
 - SYSn : nessuno
 - SYSv : IL1.0C
 - STRUCT SIZE : 2764
 - FREE HEAP : 22552
 - CPU : 80
-----------------

but after 2-3 times board not put nothing on serial and not reboot .. why ?   
i think that something in WifiUDP function may not work
Please help me to resolve because board si not stable if not reboot correctly also wdt is implicated .. if you want say to me how can i help you ! thx

asetyde, 2015-07-26:
reading old post you say to check gpio0 and gpio2 of esp8266 not nodemcu d0 d2 i think , and if is true i must just pinmode function  ?

asetyde, 2015-07-28:
@igrr please close this issue , wdt now work , 0 ,  2 pin not touch -> good idea

closed 2015-08-17 by asetyde