esp8266/Arduino #556: Ticker library: attach_us

Yoshi20, 2015-07-15:
I need ticker calls less than 1ms to control the frequency of a piezo.

I managed to implement an attach_us-function with the posts in the community forum
(see: http://www.esp8266.com/viewtopic.php?f=33&t=2294&p=13492&hilit=osapi.h#p13492)
It works quite well with the version I downloaded via board manager, but it doesn't work anymore with your current repo-version (Di. 15.07.15).

Could you officially include an attach_us-function, or is there something that speaks against it?
May you know a reason it doesn't work any more with your current repo?

igrr, 2015-07-15:
Ticker is based on system timer, and you will see considerable timing errors if you have a lot of TCP/UDP activity. 
For such short delays I think the better alternative is to use timer interrupts directly.
You can take a look at Servo library using ESP8266 timers for an example:
https://github.com/esp8266/Arduino/blob/ea302aab05480ad36c076b618abb642b1eb1893b/hardware/esp8266com/esp8266/libraries/Servo/src/esp8266/ServoTimers.h

Yoshi20, 2015-07-16:
Thanks for the answer. I didn't know yet that there are timer0 and timer1.

I just tested them now and discovered the following problems:
error: 'timer0_isr_init' was not declared in this scope
-> It seems to can not find the timer0-functions although they're defined in Arduino.h :(

timer1 however works fine until I initialize my ESP8266 as acces point and the wdt resets my program.
-> "WiFi.mode(WIFI_AP);" or "WiFi.softAP(ssid);" seems to be the problem respectively triggers the wdt.
Disableing the interrupt (with "noInterrupts()") during the access point setup didn't solve the problem either.

May you have an explenation and/or a workaround?

devyte, 2017-09-03:
@Yoshi20 did you find a resolution on this? Can the issue be closed?

Yoshi20, 2017-09-03:
Hi @devyte

Sorry I haven't used or tested this ever since.
I'm currently working with the ESP32, so the ESP8266 is not realy interesting anymore :)

It may be good to test this some more, but on my account you can close the issue.

Keep the good work up.

devyte, 2017-09-03:
@Yoshi20 I admit to being curious about the ESP32, but I plan on putting up about 200 ESPs around the house in total. With a current $4 gap in price, that's an $800 difference, so the ESP8266 is still more interesting to me than the ESP32 :)
Besides, how much functionality can the tiny guy _really_ handle?

Anyways, thanks for the answer. I am trying to do some cleanup, but I don't have supapowers here for closing myself, so could you please close?

closed 2017-09-03 by Yoshi20

igrr, 2017-09-04:
@devyte Now you have! Thanks for doing the cleanup!

devyte, 2017-09-04:
@igrr Thank you, I'll do what I can :) could we discuss policy somewhere?
I'd like to avoid closing issues that shouldn't be...

igrr, 2017-09-04:
Maybe a "meta" issue can be opened for such policy discussion? Then based
on the outcome the policy can be written down in contribution guidelines,
for example.

(E.g. which issues can be reported, reporting rules, requirements for MCVE,
the process of handling new issues (labeling, assigning, prioritizing),
closing (wontfix/can't reproduce/no feedback), etc.