esp8266/Arduino #557: light sleep without beeing connected to a WLAN

Yoshi20, 2015-07-15:
I would like to configure my ESP8266 as an access point only in a specific state (of a state machine).
In the meantime I just set some GPIOs and wait (with delays).
The problem now is, that light sleep won't be activated unless the ESP8266 is connected to a WLAN (or was connected once -> it seems to save the ssid and the ssid-pw internaly!)

Just setting the sleept type (with "wifi_set_sleep_type(LIGHT_SLEEP_T)")
and changing the opmode to WIFI_STA doesn't seem to be sufficient.

Is there an other way to go into light sleep without beeing connected to a WLAN?
Are there other ways in the SW to reduce power consumption during the runtime?

Humancell, 2015-12-01:
Bump.

I'm looking for an example or more details on using "light sleep" in a sketch.  I'm not finding much in the docs or when I search ... any pointers that someone can provide?

I'll write things up when I get it figured out, but am looking for a good starting point on using it.

Thoughts?  Links?

Yoshi20, 2016-02-11:
It seems to be still not possible to enable the 'light sleep' when the ESP is not yet connected to a WLAN.
But #1381 seems to approach a solution for the problem.

devyte, 2017-10-20:
Duplicate of #1381. Closing this one.

closed 2017-10-20 by devyte