esp8266/Arduino #558: More recent build for Arduino library manager json URL?

tdicola, 2015-07-15:
Just curious, is it possible to update the build published to the default Arduino library manager JSON URL (http://arduino.esp8266.com/package_esp8266com_index.json) to something more recent?  The current build there is 1.6.4-673-g8cd3697 which is starting to get pretty old now (created May 22nd) for a quickly moving project like this one.  There are some good fixes like one to make the pgmspace.h functions work (https://github.com/Links2004/Arduino/commit/936669e234e997b1d0406af004d1ac56b900d4fb) that are only in later builds that aren't published yet.  Any chance the published build could be updated so folks can get fixes like those?

Alternatively, is the staging build JSON URL (http://arduino.esp8266.com/staging/package_esp8266com_index.json) what people should use for more  recent builds?

hamishcunningham, 2015-07-15:
Staging isn't especially recent AFAIK.

To get the latest (though sometimes unstable) version: https://github.com/esp8266/Arduino/issues/547#issuecomment-121147932

HTH!

toddtreece, 2015-07-16:
@hamishcunningham a manual install will definitely work for more advanced users, but the board manager package makes it a lot easier for people to get started.

maybe the package update be rolled into the travis deploy config so that stuff doesn't fall way out of sync?

igrr, 2015-07-16:
Regarding Travis deploy of packages — it's too easy to deploy a broken
version this way.

I'm fixing documentation and rebuilding the downloads site in preparation
for the package update. So yes, it's on the way.

igrr, 2015-07-22:
I have updated staging release today. Will test a bit more tonight, if there are no new issues, it will be pushed as a stable version.

toddtreece, 2015-07-22:
@igrr re: new staging release. check out #588.

i'm not sure if this would help the ESP8266 arduino project, but i've been using travis ci to test build all of the examples for adafruit's arduino libraries to make sure they compile on different platforms. here's an example config: https://github.com/adafruit/adafruit-io-basics/blob/master/.travis.yml.

thanks for all of your work!

igrr, 2015-07-22:
Todd, thanks for the tip regarding travis tests. I tried the same approach a few weeks back, but it looks like I was missing this part from your script:
```bash
/sbin/start-stop-daemon --start --quiet --pidfile /tmp/custom_xvfb_1.pid --make-pidfile --background --exec /usr/bin/Xvfb -- :1 -ac -screen 0 1280x1024x16
sleep 3
export DISPLAY=:1.0
```
Will certainly add tests to our travis build, thanks for the tip.

Regarding #588, will fix promptly.

igrr, 2015-07-23:
Travis build now includes verification of example sketches under `hardware/esp8266com/esp8266`.

toddtreece, 2015-07-23:
@igrr great! yeah the Arduino CLI expects a display for some reason, so you have to fake it.

igrr, 2015-07-23:
@tdicola @toddtreece please take note that link for the stable package has changed to
http://arduino.esp8266.com/stable/package_esp8266com_index.json

closed 2015-07-23 by igrr

toddtreece, 2015-07-23:
@igrr thanks!

tdicola, 2015-07-23:
Awesome, thanks so much!