esp8266/Arduino #562: ILI9341 white screen

ManiackCZ, 2015-07-16:
Hello,
i have ESP07 module (later version /w ADC pin) on my own breakout board hooked to 2,2" TFT with ILI9341 driver. I have it wired according to pin map on readme page but all I get is white screen. For CS and DC pins I tried to use multiple combinations of pins (15,16,2...). I used latest Arduino + board manager and also tried version i built from souce. None of it worked, all i got was white screen.

tontito, 2015-07-16:
add pinMode(2, OUTPUT); in setup() . Work tft

Links2004, 2015-07-16:
the Reset pin of the display need an pullup if you not connect it.

ManiackCZ, 2015-07-17:
I found the trick with pinMode(2, OUTPUT); in other thread and already tried it. Doesnt work for me. I will try to add pullup on reset pin and see if it helps.

ManiackCZ, 2015-07-17:
I added pullup on RST and added pinMode(2,OUTPUT); into my code. Neither of it helps. Just white screen again.

Links2004, 2015-07-17:
i have here running 2 screens connected like this:
i connect the screen like this:

 TFT       |      ESP 
----------- | ----------------
MISO     |  12
SCK       | 14
MOSI     | 13
DC         |  4
RESET  |  5 or 3V3 (pullup)
CS         |  15

with reset:
```cpp
#define TFT_DC 5
#define TFT_CS 15
#define TFT_RESET 4

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RESET);
```

no reset:
```cpp
#define TFT_DC 5
#define TFT_CS 15

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
```

if you power the TFT with 3V3 volt you need to solder the pad on the Back of the screen (near the LDO).
hope this helps.

ManiackCZ, 2015-07-17:
Oh, I forgo that the board has its own LDO and I was powering it from 3v3. Thanks man, that fixed my problem :)

closed 2015-07-17 by ManiackCZ

pfabri, 2019-05-15:
@Links2004  Holly Cow! Thanks so much!