esp8266/Arduino #566: e_servoIndex 0 when using Servo library with HTTP server

giannoug, 2015-07-17:
Using the latest "staging" version and the Servo library directly downloaded from git, I'm getting the following error during boot and right after flashing.

     ets Jan  8 2013,rst cause:4, boot mode:(3,7)

    wdt reset
    load 0x4010f000, len 1464, room 16 
    tail 8
    chksum 0x7a
    csum 0x7a
    e_servoIndex 0

I have pushed some code to reproduce the problem: https://github.com/giannoug/turtle-feeder

giannoug, 2015-07-17:
In the example provided, I've moved `feeder.attach(2);` after `server.begin();` and the issue magically disappeared. I still get `_servoIndex 0` in the Serial console, but at least the ESP doesn't hang and reboot repeatedly.

Makuna, 2015-07-19:
I can't reproduce your problem.  I have tried it with the order of the two statements as it is listed in the file and swapped them without having a problem.  

What device do you have?  How have you connected it?

Please do the following to capture more information:
1) after ```Serial.begin(115200);``` add ```Serial.setDebugOutput(true);``` The exception should now contain more information.
2) objdump of the elf file and look up the address of the error to find the code causing the problem
```
xtensa-lx106-elf-objdump -S your_elf_file.elf > dum.dobj
```

Makuna, 2015-07-29:
@giannoug do you still reproduce this problem?

giannoug, 2015-07-29:
@Makuna I haven't forgot about this issue, I do have another one too. Give me a couple of days and I'll report back

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004