esp8266/Arduino #567: wdt reset on ESP.getVdd33()

rumbero71, 2015-07-18:
Hi,
sorry for posting a dupe but I dont know if it fit into the other thread I was posting it to.
There seems to be problem with the ESP.getVdd33() function. 
Setup: 
-ESP-01 with XPD_DCDC soldered to RESET
-Code Initializing Serial, Initializing WIFI, connecting to an MQTT broker, Reading out a DHT22, Reading out getVdd33, pushing values to mqtt broker.
Without the getVdd33, everything is fine. Wifi is connected, DHT22 read, Values pushed out, deep sleep invoked, waking up...ad infinitum.
However, when measuring getVdd33, sometimes voltage gets read correctly, but most times the chip wdt resets. Putting in wdt_disable(), program stays indefinitely at getVdd33.
Any hints on that ?
Thanks + Regards
Markus

d-a-v, 2015-07-18:
May be this could help
https://github.com/nekromant/esp8266-frankenstein/blob/master/src/cmd_common.c#L37
(same problem, stable solution)

rumbero71, 2015-07-18:
Hi,
thanks for your response. At least the freeze is gone. However, it does not seem to return the correct Voltage. I put a small LiPo on it for test purposes, and my multimeter shows the voltage dropping slowly. However, the returned voltage always stays at 4094. Does the ADC have to be initialized / reset somehow after deep sleep ?
Here's what I am doing:
In the Arduino IDE I changed the Esp.cpp and Esp.h, adding the function including os_intr_lock(); before and os_intr_unlock(); after the readvdd33 command.
In the setup() section of my sketch, I am doing this:

void setup(void)
{
  // Start Serial
  Serial.begin(115200);

  memset(voltchar, 0, sizeof(voltchar));
  sprintf(voltchar,"%d",ESP.getVdd33());
...

At a later stage, voltchar is printed / sent to mqtt broker and the system is put to sleep.

Thanks+Regards Markus

closed 2015-09-30 by igrr