esp8266/Arduino #568: atan2 wierdness

CupOfTea, 2015-07-19:
Running the HMC5883L_raw example from i2cdevlib (https://github.com/jrowberg/i2cdevlib) the esp8266 freezes and wdt loops it.  I have resolved my problem by replacing the call to atan2() with code I found on the net.
  I cant see any sense in it,  but fwiw,  it would work fine when the atan2() arguments had not been updated by a pass by reference call to getHeading() and/or they were the result of a conversion to int from an array.  I don't think it's specific to i2cdevlib.      String s = String(2);   int a = s.toInt(); atan2( a, a); would freeze but only if the sketch contained a function using pass by reference.  I'm sorry but I got so grumpy yesterday I deleted all the test sketches I made and searched for a atan2 substitute.

igrr, 2015-07-23:
Should be fixed in stable package (1.6.5-947-g39819f0). If it still doesn't work, try deleting all the downloaded tools (xtensa-lx106-elf...) and installing the boards manager package again.

CupOfTea, 2015-07-23:
FIXED: Awesome.... Thank You!  I had reached the point where I thought I was going mad.

closed 2015-07-23 by CupOfTea