esp8266/Arduino #570: How many clients can connect in Access point mode

STASIUK, 2015-07-19:
How many clients can connect to the module in Access point mode?

acoulson2000, 2015-09-15:
anyone?

Links2004, 2015-09-15:
4 clients are possible right now.
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L267

acoulson2000, 2015-09-16:
It seems like it is possible to call wifi_softap_set_config() with a config.max_connections value greater than 4 to override the default? Do you know if this is, in fact the case, and what the ramifications might be? I understand that more clients require more RAM, but from googling around, it's not clear how much. I have a relatively simple host app which I would like to use to send UPD broadcasts to many clients.

I appreciate any sight,
Andy

Links2004, 2015-09-16:
you can try it, if its working we can add an API to change the value.
keep in mind the TCP sockets are limited to 5 and the UDP sockets to 4 form the SDK. 
but multicast / broadcasts shut reaches all with one socket.

acoulson2000, 2015-09-19:
Markus,

Thanks for you're feedback. I'm not very C-savy and certainly the whole ESP8266 toolchain is a bit intimidating. To try it, is it necessary to modify the source code and build the firmware? Or can I simply set up my own config structure and pass it in from my own user code using something like below?  I ask because your response regarding the possibility of providing an API seems to imply this will not work.

struct softap_config cfgESP;
//.... set ssid, auth, etc
cfgESP.max_connection=32;
// ....
wifi_softap_set_config(&cfgESP);

Thanks again,
Andy

Links2004, 2015-09-19:
you can use the SDK directly like you shown. 

to use the SDK you need to do the include like this:
```cpp
extern "C" {
#include "user_interface.h"
}
```

if its working i will add it to the begin function of the AP, so it will be easy to use with out to deal with the SDK.

brianensor, 2015-10-13:
Any luck @acoulson2000? I'd like to increase the number of connections for my project if possible.

acoulson2000, 2015-10-14:
So far, I have only connected 3 simultaneously, but so far, so good. It will probably be a couple of weeks at least before I'm ready to try more, but I'll post back with my results.

bertamf, 2016-01-28:
Hi @acoulson2000! I've also been trying to connect more than 4 devices to the AP, but I haven't succeed so far. What about you?
Thanks!

acoulson2000, 2016-01-28:
@brianensor and @bertamf, sorry, I still have not tried this. I went up working with NodeMCU for my project, and even then, I only got up to exactly 4 clients (this was for the Christmas project I've begun describing here https://austinlightguy.wordpress.com/2016/01/24/lightsuit-v2/).  I've started playing around with some stuff I postponed during the busy holidays, though, so I hope to investigate this again shortly and will post my findings. First, I will probably just try it with my NodeMCU AP and clients, then I'll try an Arduino version. One issue I do have is that I only get a packet throughput rate of something like 8-10Hz and I'm wondering if the Arduino code, which will be more like native, will be faster.

alx-1, 2016-05-10:
I can edit user_interface.h if someone knows how to pass a higher value to max_connection : 
uint8 max_connection;       // Note: default 4, max 4
I don't know C but I have 8 ESP8266's to test it with. @acoulson2000, I like your light suit! We're working on controlling 8 radios from relays connected to carbon monoxyde pollution.[https://github.com/reseauducommun/CarboBlaster](url)

acoulson2000, 2016-05-10:
So, I've started playing around with this a bit again. I was using my NodeMCU code, but was finding that I got a lot of random reboots on my UDP receiver. I think possibly this is because it's also doing a lot of work pumping the WS2812 data for about 300 LEDs across GPIO2 (although this is supposed to be very effiecient with the UART-driven fix (https://github.com/nodemcu/nodemcu-firmware/pull/951) for that driver.) 

I have switched to Arduino, but so far am only up to three clients. I'll rig a couple more up within a few days. So far, I have not added the WS2812 functionality in, so I'm not sure what the stability will look like.

One thing I've definitely determined - pumping packets at high speed, even via UDP multicast, does not work very well when the sender is configured in AP mode. I guess it's too busy handling the WiFi stack and just managing WiFi clients or something, but when I configure it that way, my clients all seem to receive packets in "clumps" - so they'll get something like 4 to 6 packets, then experience a slight delay, then receive another clump. The overall throughput is decent, but for my application I really want a consistent, high-speed stream of packets, since my clients are essentially remote VU meters. When I configure both the sender and receivers to all connect to my home wifi in WIFI_STA mode, I get much more consistent streaming at about 10hz (4 byte packets). With my phone as a tethering AP, it's not as good as my home router, but still better than an ESP8266 in AP mode.

@alx-1, I think the idea is that you may not have to modify existing SDK code, you just include the header file at the top of your sketch wit this:
`
extern "C" {
#include "user_interface.h"
}
`

then something like this in setup() of your sketch:
`
   struct softap_config config;
   wifi_softap_get_config(&config); // Get config first.
   
   os_memset(config.ssid, 0, 32);
   os_memset(config.password, 0, 64);
   os_memcpy(config.ssid, "ESP8266", 7);
   os_memcpy(config.password, "12345678", 8);
   config.authmode = AUTH_WPA_WPA2_PSK;
   config.ssid_len = 0;// or its actual length
   config.beacon_interval = 100;
   config.max_connection = 8; // how many stations can connect to ESP8266 softAP at most.

   wifi_softap_set_config(&config);// Set ESP8266 softap config .
`

Now that I look at this, I wonder if fiddling witht the beacon_interval might help my in AP mode...

NikhilMJ, 2016-07-13:
@acoulson2000 : Any progress on this? I have an application that requires 4-8 ESPs that are sent commands from a single ESP configured in AP mode.

acoulson2000, 2017-07-09:
@NikhilMJ , not sure if you're still interested in this issue, but in case you or anyone else who runs across this thread is, I do have something to report.

While I never got around to trying this with the ESP8266, that is in part because I found that the packet rate from an ESP8266 in AP mode was really quite low. Using the ESP8266 as the AP, I was able to broadcast small (5 byte) packets at around 1 every 150-to-200ms. Connecting to a separate, dedicated AP (just a very cheap WiFi router), I got almost double. Since my project needs to be wearable, I didn't really pursue this option.

HOWEVER, now that the ESP32 is available, I decided to revisit it. It even turns out that, alhtough the default for ESP32 is still 4, there was a recent commit to the ESP32 SDK that supports changing the maximum. Using Sparkfun ESP32 thing with the extended API - which is WiFi.softAP(ssid, passphrase, channel, ssid_hidden, max_connection) - I have been able to set max_connections to 16 and have so far connected 6 clients simultaneously, with a throughput of 40 packets per second (25ms/packet)!  Hooray for the ESP32!

devyte, 2017-10-20:
There is a method in the WiFi class to set the max number of wifi clients that can connect to the SoftAP. I believe that default is 4, the max is 8, but I'm not certain of that.
Closing.

closed 2017-10-20 by devyte

NikhilMJ, 2017-11-16:
@acoulson2000 : Apologies for the delayed response to this. I had about 100+byte packets broadcasted every 15msec or so. I was only looking to increasing the total number of client connections.

devyte, 2018-02-16:
@imdsp it's possible, but this is not the right place to ask such questions. Please refer to a community forum like esp8266.com or stackoverflow.

acoulson2000, 2018-02-16:
@imdsp, If you email me directly, I can point you in the right direction.

chrzimmer, 2018-02-22:
Hello everyone, 
this answer sounds great to me :)

> There is a method in the WiFi class to set the max number of wifi clients that can connect to the SoftAP. I believe that default is 4, the max is 8, but I'm not certain of that.

can one explain to me briefly how i can set this to 8 stations ? I programm my ESP8266 with the Arduino 
IDE and not really what i would call en expert. I desperately need  at least 6  stations to connect to the AP.
Unfortunately i am late in my project:) 
and i appreciate any help.

acoulson2000, 2018-02-22:
Just use:

  WiFi.softAP(const char* ssid, const char* passphrase, int channel, int ssid_hidden, int max_connection);

where max_connection defaults to 4.

chrzimmer, 2018-02-22:
thx thounsand times. u saved my day :+1:

AlbertMN, 2018-03-31:
@acoulson2000 's answer solved it for me. Fast and painless!

AnalysIR, 2018-04-01:
@chrzimmer & @AlbertMN - how many client were you able to configure/verify working correctly?

AlbertMN, 2018-04-01:
@AnalysIR My case is opposite of what most others are trying to achieve. I needed to limit the connections to 1, which is all I have tested. If one device is already connected to the AP other devices will simply be denied access.

chrzimmer, 2018-04-02:
@AnalysIR  my project came to a sudden stop, so i can not answer. Sorry.

harleyfrazee, 2018-10-08:
Where is the gethub link for the 8266 softAP code that allows the setting of max clients?? I tried the following and it wouldn't compile.
WiFi.softAP(const char* ssid, const char* passphrase, int channel, int ssid_hidden, int max_connection);

acoulson2000, 2018-10-08:
@harleyfrazee maybe you to make sure you are on the latest version of Arduino and Arduino Core for ESP8266?

The method is defined in this header file: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiAP.h

harleyfrazee, 2018-10-08:
Thanks, but I guess I messed up things on my laptop. I switched to ada_WiFi because I couldn't get the Arduino version to work. I continue to get "WiFi_AP_w_Blink:65: error: 'class WiFiClass' has no member named 'softAP' WiFi.softAP(ssid, password);"

Do you know of instructions so that I could remove my current Arduino version and get the correct version from gethub and compile it?

acoulson2000, 2018-10-09:
If you have the esp8266 installed under board manager, you're probably good- you probably just need to #include <ESP8266WIFI.h> instead of whatever WiFi library you are trying to use.

pankaj143g, 2019-06-13:
hello everyone!
this thread was really useful to read about number of clients to AP (8 maximum i guess).
But for my project I need to connect around 70 clients to one access point. I am guessing that we can not do with the help of esp8266. So is there any other WiFi chipset which i can use as an AP and being able to connect 70 clients with that.

shahabkarim, 2019-07-23:
It's better to use the zigbee module

acoulson2000, 2019-07-23:
I used to use Zigbees, but one of the main things that got me interested in esp's was the cost of Zigbee's - ~$30/ea, plus a carrier bored with a "non-standard" pin spacing. Then, you still need an processor on reach client to do much.  There are some inexpensive RF comm chips out there - in fact I got a bunch like the ones below, but I haven't tried to do anything with them yet.
 https://www.amazon.com/dp/B07GZ17ZWS

HugoML, 2020-01-12:
ESP-MESH can help to create 70 nodes.