esp8266/Arduino #572: strspn (and other from string.h) are missing

oscaracena, 2015-07-20:
If I compile the following sketch:

```c++
void setup() {}
    
const char* f() {
    static char data[16] = "an example";  
    return data;
}
    
void loop() {
    const char* d = f();  
    Serial.print(strspn(d, "an"));
}
```

I get the following error at link time (either with gcc or g++):

```
/tmp/build3043909424170182954.tmp/sketch_jul20b.cpp.o: In function `f()':
sketch_jul20b.cpp:(.text+0x18): undefined reference to `strspn'
/tmp/build3043909424170182954.tmp/sketch_jul20b.cpp.o: In function `loop':
sketch_jul20b.cpp:(.text+0x28): undefined reference to `strspn'
```

But, if I compile the same piece of code, but using the raw SDK (same version, 1.2.0), which would look like this:

```c++
#include <osapi.h>

const char* f() {
    static char data[16] = "an example";
    return data;
}

void
user_init() {
    const char* d = f();
    os_printf("%d\n", strspn(d, "an"));
}
```

It works as expected, when compiled either with gcc or g++.

bhcuong2008, 2016-08-03:
I'm facing 2 library functions, strsep and strspn (undefined reference to 'xxx'). 

Please help me how to solve it. I've already included string.h.

With original gcc on Eclipse, it works well.

Raphyyy, 2017-09-28:
Any news on this ?

yoursunny, 2017-12-30:
I can no longer produce this issue using @oscaracena 's snippet with a2d16f38d48727bba41befadebc590bdd84588ad .

closed 2018-01-02 by igrr