esp8266/Arduino #574: Ethernet library not compiling

Letreguilly, 2015-07-20:
I trying to use the Ethernet lib (w5100) with the esp8266. 

I know that the esp8266 has built in wifi controler, but wired connection are more stable and can provide power using POE. From my point of view the esp8266 is more than wifi.

When i tried to compile : 

arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp: In member function 'int EthernetClass::begin(uint8_t*)':
arduino-1.6.5/libraries/Ethernet/src/Ethernet.cpp:19:45: error: no matching function for call to 'SPIClass::beginTransaction(int, SPISettings)'
   SPI.beginTransaction(SPI_ETHERNET_SETTINGS);

This error occur many times.

I have look inside the function signature from the esp8266 SPI.h file : 

  void beginTransaction(SPISettings settings);

and from the AVR SPI.h file : 

  inline static void beginTransaction(SPISettings settings) {

The two signature are very close. But each one has is own  SPISettings class. 

According to the projet homepage, SPI is working. I suppose that the SPISettings class from esp8266 is working. And then i have no other idea how debug this issue.

I'm not a C++ or MCU killer, just an enthusiast user and i have limited knowledge

Thank in advance for help

Links2004, 2015-07-20:
the problem comes from the W5100.h 
change:

```cpp
#if defined(ARDUINO_ARCH_AVR)
#define SPI_ETHERNET_SETTINGS SPISettings(4000000, MSBFIRST, SPI_MODE0)
#else
#define SPI_ETHERNET_SETTINGS SPI_CS,SPISettings(4000000, MSBFIRST, SPI_MODE0)
#endif
```

to

```cpp
#if defined(ARDUINO_ARCH_AVR)
#define SPI_ETHERNET_SETTINGS SPISettings(4000000, MSBFIRST, SPI_MODE0)
#elif defined(ESP8266)  
#define SPI_ETHERNET_SETTINGS SPISettings(14000000, MSBFIRST, SPI_MODE0)
#else
#define SPI_ETHERNET_SETTINGS SPI_CS,SPISettings(4000000, MSBFIRST, SPI_MODE0)
#endif
```

Letreguilly, 2015-07-21:
thank for help,

I have made your modification inside my w5100.h file. 

Some other issue appear, i have tried to fix them : 

```c++

#if defined(ARDUINO_ARCH_AVR)
  SPI.begin();
  initSS();
#elif defined(ESP8266)  
  SPI.begin();
#else
  SPI.begin(SPI_CS);
  // Set clock to 4Mhz (W5100 should support up to about 14Mhz)
  SPI.setClockDivider(SPI_CS, 21);
  SPI.setDataMode(SPI_CS, SPI_MODE0);
#endif
```
i have modified 5 codes bloc using the same logic. I hope it's working, but the SPI SS management is completely disabled ...

But this is not the main "new" problem, i have a new compilation error out of my knowledge :  

Ethernet/Ethernet.cpp.o:(.text+0x18): undefined reference to `__cxa_guard_acquire'
Ethernet/Ethernet.cpp.o:(.text+0x1c): undefined reference to `__cxa_guard_release'
Ethernet/Ethernet.cpp.o:(.text+0x3e): undefined reference to `__cxa_guard_acquire'
Ethernet/Ethernet.cpp.o:(.text+0x4e): undefined reference to `__cxa_guard_release'
collect2: error: ld returned 1 exit status

any idea ?

another ways to get ethernet on the ESP8266 is to use the ENC28J60 and this library : https://github.com/ntruchsess/arduino_uip. Maybe this lib is easier to port on ESP8266 ?

Links2004, 2015-07-21:
the __cxa_guard_release problem is handled here #500
but not fixed for the moment.
```cxa_guard_acquire``` and ```cxa_guard_release``` need to be implemented.

Letreguilly, 2015-07-21:
OK, is there a way to workaround this issue or i have just to wait ?

Links2004, 2015-07-21:
yes, replace all places where static is initialized by an function / or class in a function.
like this:
```cpp
int EthernetClass::begin(uint8_t *mac_address)
{
  static DhcpClass s_dhcp;
```
in this specific case this will work:
```cpp
static DhcpClass s_dhcp;
int EthernetClass::begin(uint8_t *mac_address)
{
```

but how to rework it can be different from static case to static case.

Links2004, 2015-07-21:
compiling but no function test.
https://github.com/Links2004/Arduino/commit/10c5dec92a00c544a4b9f800bba590fc309fa016
and has many potential for getting more speed of SPI by using ```writeBytes``` and ```transferBytes``` form SPI instead of transfaring data byte by byte

closed 2015-09-30 by igrr

Chris2011, 2016-01-11:
I use Netbeans with an arduino plugin and I got the same issue, when I build the project. Here is my makefile: http://pastebin.com/Md0R2t4R. I got many errors: http://pastebin.com/qHppisX6

I use windows 7 64bit, GnuWin and Arduino 1.6.4. Do I have really change the w5100.h file? Is there no bug fix for this?

Regards

Chris

Links2004, 2016-01-11:
look like it using the AVR SPI lib sound wrong.
```./hardware/arduino/avr/libraries/spi/SPI.h:```

Chris2011, 2016-01-12:
Do I have to change smth?

Links2004, 2016-01-12:
look like your Netbeans plugin uses the wrong one,
in the Arduino IDE and Eclipse its working.
ask the autor of the plugin to fix the problem.

Chris2011, 2016-01-12:
And which one is the correct one?

Links2004, 2016-01-12:
for ESP8266 the one in the in the esp8266 dir?
https://github.com/esp8266/Arduino/tree/master/libraries/SPI

Chris2011, 2016-01-12:
Thx I will check when I'm home.

Chris2011, 2016-01-12:
As I understand it right, I have to install this plugin first? Because my folder Arduino/libraries has no Folder SPI.