esp8266/Arduino #576: ssd1306 i2c

zenmanenergy, 2015-07-20:
I have been unsuccessful with getting an i2c ssd1306 screen (from ebay, it works fine on an UNO) to work on an esp-12, I have added the library at https://github.com/me-no-dev/Adafruit_SSD1306 have connected the screen to vcc/gnd scl to pin 5 and sda to pin 4. It works fine on an Arduino UNO, so I'm clearly doing something wrong on the esp.

When I upload the example for the 128x64 i2c i don't get any errors, but it also doesn't display anything.

The example i2c code doesn't include Wire.begin(4,5); in the setup(), I've tried adding that but it doesn't display anything. On the back of the screen it says Address Select 0x7A and 0x78.  I have tried changing the address from
display.begin(SSD1306_SWITCHCAPVCC, 0x3D);
to:
display.begin(SSD1306_SWITCHCAPVCC, 0x7A);
and
display.begin(SSD1306_SWITCHCAPVCC, 0x78);

But still nothing. 

This is screen I'm using: http://www.ebay.com/itm/141536719131?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

I'm out of ideas to try. Any suggestions?

Duality4Y, 2015-07-20:
do you have pullups on you i2c pins ?

zenmanenergy, 2015-07-20:
I tried connecting 10K resisters from pin4 to vcc and pin5 to VCC, with no luck.

Is that the correct way to do a pullup?

Duality4Y, 2015-07-21:
Yes that is correct. you could try and see if i2c is working at all with a scope.

far5893, 2015-07-21:
Sometimes pin 4 and 5 are swapped in china modules, check very well that 4 is 4 and 5 is 5!!

zenmanenergy, 2015-07-21:
haha. To be fair to Chinese manufacturers... while they're English is not great, they can speak more languages than I do!

I have tried swapping with no luck. But I haven't tried doing a digitalWrite() to an led to verify it on my esp12. I'll try that.

I did start messing with the .h and .cpp files by putting Serial.println()s in them, but still haven't figured this out. I found that the wire.begin() is actually in the library itself, so we don't have to include it in the main code.

zenmanenergy, 2015-07-21:
Duality, I would connect it to a scope... if I knew how to use one and... if I had one. :-)

TimeTravelingOwls, 2015-07-21:
You might find this useful: http://playground.arduino.cc/Main/I2cScanner

It'll tell you whether your ESP can see any devices, and their addresses.

zenmanenergy, 2015-07-21:
Oh wow. That looks awesome. Thanks TimeTraveling, I'll check that out.

TimeTravelingOwls, 2015-07-21:
My cheap OLED was marked 0x7A, but its I2C address was actually 0x3C.

Duality4Y, 2015-07-21:
the scanner is a good point :)

zenmanenergy, 2015-07-22:
TimeTraveling, have you gotten your oled to work with an esp? are you open to sharing your code with me?

canadaduane, 2015-07-22:
I've been working on improvements to the Adafruit SSD1306 library--I made SPI compatible, and then used I2C successfully with this: https://github.com/canadaduane/Adafruit_SSD1306. You may also be interested in a new Adafruit_mfGFX fork I made which supports multiple fonts (but loads only the ones you actually use into flash memory): https://github.com/canadaduane/Adafruit_mfGFX

duncan-a, 2015-07-22:
Does anyone know if the u8glib OLED library has been converted to work with
the ESP8266?

In Arduino form it seems to have a few advantages over the Adafruit
library, from what I can understand, like smoother large fonts and I
believe it's faster.

I've only seen it on YouTube but it looks ideal for things like room
temperature display, which is what I'm wanting to do with I2C.

I'm nit terribly 'techie' so I'm flying a bit blind on this.

TimeTravelingOwls, 2015-07-22:
I don't remember whether I ever got the adafruit libraries to work, and I
can't get to my computer to check today, but I definitely got the oled
working with Dan Bicks' WifiScan code:
http://www.esp8266.com/viewtopic.php?f=29&t=3256&hilit=Wifi+scan+Oled

The esp8266 community site is a better place for this conversation.

zenmanenergy, 2015-07-23:
Finally! I got it!

Thank you ALL!

The wifi scan worked great, it gave me a value of 0x3C. I then used canadaduane's library which didn't work at first, but then realized I had left out the pullup resistors thanks to Duality4y. connecting the sca and scl to 3.3v with a 10k resistor was the final solution.

closed 2015-07-23 by zenmanenergy

dunk8888, 2017-01-07:
Hi all,if you are having trouble with ssd1306 oled there is a start up problem ive noticed with some ssd1306 i2c screens.
Ive an i2c ssd1306 screen wich doesnt power up properly but if i swith it on then off then back on fast it works,it seems like a timing issue maybe,all my i2c screens are like this as i got them from same seller,my screen has the flames on the back,if i just power up it doesnt work,but if i power up and down and up again fast they work,its strange,ive tryed all arduino boards and get same results,so bare this in mind too,any one with an idea as to why this happens please explain.thanks

gicho, 2017-01-07:
Do you have #RES pin connected to GPIO of the ESP, or to reset supervisor
chip?
Those OLEDs do require proper reset at power on. Connecting #RES to Vdd is
not enough - sometimes it does not reset properly. It depends on the power
supply ramps.
Clean solution is to use the #RES pin connected to an output from ESP and
to issue a reset before attempting to connect.

dunk8888, 2017-01-08:
Hi my i2c oled has no reset!

eudoxos, 2018-04-20:
Same issue here, the OLED needs to be powered-up twice, I tried inserting delays into setup code at verious places, it does not really help. Did you find some solution? This cheap OLED has no RES either.

papconst, 2018-12-04:
Have the same problem with my SSD1306.
I've tried everything written here, and even I2C scanner cant see my display.
BUT!  Display perfectly works on esp-12f with this lib https://github.com/ThingPulse/esp8266-oled-ssd1306
Have no ideas what is wrong.

marcelooggero, 2021-09-01:
Exactly same issue with chinese oled 128x64  ssd1306 marked as 0x7A addres but working in 0x3C.