esp8266/Arduino #578: .o files from .c files not in link

mwolfeu, 2015-07-20:
In the arduino ide if you:
 create an extra .cpp tab
 write a function    (I wrote one that returned the first int arg +1)
 write a prototype for it in the main .ino
 call it from the main .ino

everything works perfect.  

If however you rename that extension to .c  you will get an " undefined reference" error from ld.

igrr, 2015-07-21:
Make sure you add `extern "C"` qualifier to the function prototype,
otherwise declaration will have C++ linkage, while definition will have C
linkage.

mwolfeu, 2015-07-21:
I used the commandline sdk with some C just before this and forgot Arduino compiles .ino files with g++.  My fault entirely.  Thanks

closed 2015-07-22 by igrr