esp8266/Arduino #579: Change CPU frequency

Pablotis, 2015-07-20:
Add ESP library command for system_update_cpu_freq

igrr, 2015-08-05:
CPU frequency is set automatically depending on what is chosen in Tools > CPU Frequency menu in the IDE. If you want to override this behaviour for some reasons, you may provide an alternative implementation for `preloop_update_frequency` function inside your sketch. This function is declared as *weak*, so you can override it without modifying the library.
Also if you need to call `system_update_cpu_freq`, just include `user_interface.h` and call it directly. I think this function doesn't need a wrapper.

closed 2015-08-05 by igrr

dadler, 2017-05-19:
I've noticed that after using the Arduino IDE to set rate to 160 MHZ and then back to 80 MHZ, it doesn't properly return to 80 MHZ until I cycle power.  Recompiling properly rebuilds with F_CPU set to 80 MHZ but preloop_update_frequency doesn't clear the clock frequency bit.  Note that it's a no-op when F_CPU isn't 160 MHZ:

void preloop_update_frequency() {
#if defined(F_CPU) && (F_CPU == 160000000L)
    REG_SET_BIT(0x3ff00014, BIT(0));
    ets_update_cpu_frequency(160);
#endif
}

That caused pulseIn() to return values twice as large as it should (until I power cycled).

commit 6cb16997 (2020-04-15): Use F_CPU if (?) CPU frequency switch is compile-time only (#6833)

* At least the F_CPU define in host mock.h is needed by host Arduino.h - need to include Arduino.h further down in mock.h for this to work.

* Geting the include order right

* Prepare for runtime CPU clock rate selection

* Fix compile for not defined F_CPU

If defined F_CPU, make getCpuFreqMHz() a constexpr

* Use defines for register CPU2X instead of hex value

* Fix build for host - getCpuFreqMHz there was also in conflict with getCycleCount, using F_CPU:

tests/host/common/mock.h:#define F_CPU 80000000 (!)

* Asymmetrical [... cut]