esp8266/Arduino #582: tone()?

matroll, 2015-07-21:
hello everyone. i can't make the tone() function work on my esp-201.
i tried a few different gpios (2,4,5..), but to no avail.
not sure if this is not working because interupts used for tone() are used somewhere else?
and speaking of gpios, do you know the difference between the pins marked GPIOxx and Dxx in the reference attached? (especially GPIO0 and D0).
thanks!
[image]

Links2004, 2015-07-21:
there is no implementation for it
https://github.com/esp8266/Arduino/blob/ea302aab05480ad36c076b618abb642b1eb1893b/hardware/esp8266com/esp8266/cores/esp8266/Tone.cpp#L38

matroll, 2015-07-22:
i see.. work in progress.. hoping someone gets to it :))
thanks for the prompt answer and for the tip about the D pins!

creationix, 2015-07-24:
It's not hard to make a basic bit-banging tone. https://github.com/creationix/uscript/blob/master/uscript.c#L660-L668  It's not perfect sound, but it does make different pitched sounds.

matroll, 2015-07-24:
thanks creationix

adrian0, 2015-08-10:
Hi Creationix,
I wouldn't mind checking out your solution, but the link ends at a 404 for me. Can you repost it?

Thanks,
-Adrian

creationix, 2015-08-10:
https://github.com/creationix/uscript/blob/bbb4f4ef144d992903a7748243a2d6ef8434b855/alternative-vms/unix.c#L153-L166

adrian0, 2015-08-10:
Thanks Creationix,
You're a champ.

mralexgray, 2015-12-27:
any progress on this?

adrian0, 2016-01-01:
None from my side. Unfortunately the project I need this for has stalled due to time.
It should kick off again soon. I'll report back asap.

bjpirt, 2016-02-05:
I've just submitted a pull request with an implementation of Tone:

https://github.com/esp8266/Arduino/pull/1581

Give it a try and let me know if it works for you.
Cheers,
Ben

matroll, 2016-02-09:
hi, thanks for this.
did not work for me though (nodemcu). any pin that needs to be used in particular? (i tried 4 & 5)
thanks.

bjpirt, 2016-02-09:
It should work with any pin. I used it on GPIO5 and it worked for me.

Did you check it with a scope or just use a speaker?

matroll, 2016-02-11:
thanks ben  
no scope unfortunately, straight to a buzzer. i ll give it another try. cheers

matroll, 2016-02-25:
hi ben, sorry to bug again with this, but i had time to try more, still with no luck.
i don't have a scope to check.
i connect a buzzer directly to the output pin and gnd. 
works fine on an arduino nano (with same buzzer to be sure it's not defective).
tried on 2 different nodemcus and different pins, with no luck at all.
(i also have a led blinking to make sure the program runs fine, and it does).
any idea?
thanks

duncan-a, 2016-02-25:
Output current maybe???

matroll, 2016-02-26:
i think it's 12mA as opposed to 25 for arduino so might be the issue indeed. though ben does not seem to have a problem connecting directly?

bjpirt, 2016-02-26:
I'm just connecting one of this type of buzzer:

http://www.aliexpress.com/item/Free-shipping-Passive-buzzer-impedance-16r-16-hx/1527549925.html

from the pin to ground and it's working fine. Though I might well need to put a resistor in there in future :-)

matroll, 2016-02-26:
yep, same i'm using. mystery remains.. you do that on a nodemcu or naked esp?

bjpirt, 2016-02-26:
Naked ESP

matroll, 2016-02-26:
ok, that might be it then. i'll see if i can dig up an esp to try it out.. if you ever get the chance to try it on a nodemcu, i'd be interested to hear what you find. 
thanks again ben.

bjpirt, 2016-02-26:
I'll give it a shot - want to post a gist of a stripped down sketch for me to try?

Also, which version of the IDE did you try? Not sure which one it made it into yet.

matroll, 2016-02-26:
sure, pasted below. just beeps and boobs to keep it fun ;) 
(again, that works just fine on a nano)
using arduino IDE 1.6.5 and the version @ https://github.com/esp8266/Arduino when they merged your work into it (a month ago maybe?) latest version should have it tough.
are you up all night or are you in EU? (i'm in france myself).

int buzz=12;
int led=16;
void setup() {
  pinMode(led, OUTPUT);
}

void loop() {
  int d = random(200);
  tone(buzz,random(50,800),d);
  digitalWrite(led,HIGH);
  delay(d);
  digitalWrite(led,LOW);
  delay(random(100));
}

bjpirt, 2016-02-26:
Thanks, I'll give that a shot.

I'm in the UK :-)

duncan-a, 2016-02-26:
Two thoughts.

The Arduino Nano is 5V logic at 40mA per pin, the ESP (unless it has level
shifters on the GPIOs, and I don't think NodeMCU does) is 3.3V logic at
12mA per pin.

The buzzer webpage states 16R, so 3.3 divided by 16  = 20.6mA - and there's
no mention of it being usable at 3.3V (or anything else, but 5V can
probably be assumed).

tweedius, 2016-03-06:
I just tried this code works on the NodeMCU ESP8266.  It will generate a square wave with the frequency variable set to what you're looking for.  Set noteLength to how long you want the frequency to play.

> analogWriteFreq(frequency);
> analogWrite(BUZZER_PIN, 512);
> delay(noteLength);
> analogWrite(BUZZER_PIN, 0);

duncan-a, 2016-03-06:
I don't think the issue is an inability to generate a tone (as a
squarewave) on an output pin, it seems to be an inability to drive the
particular output device from a NodeMCU - that's the way I'm reading it...

matroll, 2016-03-07:
hey guys, tried this quickly. thanks everyone for the contribution.
i was not aware of the writefreq() function which solves the pb.
repeating @tweedius' code in loop() DOES work:
void loop() {
  analogWriteFreq(random(200,2000));
  analogWrite(buzz, 512);
  delay(200);
  analogWrite(buzz, 0);
  delay(200);
}

(the buzzer was actually loud enough -even at 512- that i had to add a resistor to bring it down)

replacing the code with tone() still does NOT work:
void loop() {
  tone(buzz,random(200,2000),200);
  delay(400);
}
i also tried tone() without duration and notone() after a delay to see if the pb was coming from that. no luck there.

now i tried to push the volume by doing analogWrite(buzz, 1023); and that broke down, no sound any more (some clicking). so i'm guessing you're using 1023 or PWM_RANGE as the output in the lib @bjpirt? that might be the issue?
why that does not work, i'm not sure. i tried 1000 and that worked. 
also, hard to tell without a scope but i also "felt" like 512 was actually louder than 1000 - though that clearly does not make sense.

pieman64, 2016-04-09:
Tone was included in 2.1.0 release from February so WeMos and other ESP's can now use the feature.

tone(13, 2000, 1000) will send a tone to GPIO 13 (D7) at 2000Hz for 1 second.

Probably due to the way tone uses interrupts (just one at present I believe) I can't call tone for a second time or call noTone(13) followed by tone(13, 2000, 1000) for a second time.

Is there a work around for this so that the buzzer doesn't just play one fairly short tone but rather tone on, tone off, tone on etc?

closed 2016-04-18 by igrr

guidogam, 2017-07-01:
Hello, everybody! If you want to check if there is an ac signal and you don't have a scope, use your multimeter set to... Volt AC.
A piezoelectic buzzer is a capacitor, it only draws current when it's charged or discharged. Any pin should be able to drive it directly.