esp8266/Arduino #584: nodemcu v2 pinmapping are outputting to the wrong pins

therobotworkshop, 2015-07-22:
Hi,
First off I want to thankyou for incorporating the ESP into the Arduino IDE!

I have a www.doit.am esp12e devkit module with the motor shield board.
it programs fine with the Arduino IDE. Board esp12e, 80mhz etc
So I tested it with the blink sketch and if I said blink pin 0 it would blink pin 3 on the board/chip.
So here is the other pins.
pin0 = pin3
pin1 = flashing led on the nodemcu
pin2 = pin4
pin3 = nothing
pin4 = pin2
pin5 = pin1
pin6 = nothing
pin7 = nothing
pin8 = nothing

Any ideas?
Do I need a different firmware. I have version 0.95_20150318

Cheers

igrr, 2015-07-22:
NodeMCU has weird pin mapping.
Pin numbers written on the board itself do not correspond to ESP8266 GPIO pin numbers. We have constants defined to make using this board easier:
```C++
static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;
```
These are defined [here](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/variants/nodemcu/pins_arduino.h#L49).

If you want to use NodeMCU pin 5, use `D5` for pin number, and it will be translated to 'real' GPIO pin 14.

therobotworkshop, 2015-07-22:
Awesome. Thankyou so much.
Worked perfectly!

Cheers

closed 2015-07-23 by igrr

JerrySievert, 2015-12-13:
for others with the same issue that made it here, the pinouts can be found here:

https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h#L37-L59

manuelseromenho, 2016-03-15:
Hello,

First of all thanks for all your work, without that I could not do this small project on university.
I have a newbie question, but first I show you my small code:

```
/*
   YF-S201 Hall Effect Water Flow Meter / Sensor
   http://www.hobbytronics.co.uk/yf-s201-water-flow-meter
   Read Water Flow Meter and output reading in litres/hour
*/

volatile int  flow_frequency;  // Measures flow meter pulses
unsigned int  l_hour;          // Calculated litres/hour                      
unsigned char flowmeter = D2;  // Flow Meter Pin number
unsigned long currentTime;
unsigned long cloopTime;

void flow ()                  // Interruot function
{ 
   flow_frequency++;
} 

void setup()
{ 
   pinMode(flowmeter, INPUT);
   Serial.begin(9600); 
   attachInterrupt(0, flow, RISING); // Setup Interrupt 
                                     // see http://arduino.cc/en/Reference/attachInterrupt
   sei();                            // Enable interrupts  
   currentTime = millis();
   cloopTime = currentTime;
} 

void loop ()    
{
      Serial.println(digitalRead(flowmeter));            // Print litres/hour
      delay(100);//Serial.println(" L/hour");

}
```

Based on a code made available on hobbytronics.co.uk.

The code it's very incomplete at the moment, but the idea it's to create a flow meter, where I can get information from the flowmeter into a pin of the nodemcu, I'm using the pin "D2", but I'm not sure if this is correct. **Do I have to initialize this variables in my code?** 

```
static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;
```

At the moment I only get 1's an 0's on Arduino Ide serial monitor. So when I send some air or water throught the flowmeter, the info on the monitor changes from zero to one, repeatedly.

TheAustrian, 2016-03-15:
The flowmeter sends pulses - "Flow rate pulse characteristics: Frequency (Hz) = 7.5 * Flow rate (L/min)" - you need to count them (a variable that gets increased by one every pulse), divide by some time interval and output the result. At the moment you just display each pulse separately...

randmor, 2016-04-13:
Continuing with igrr's comment...

Another way to define Arduino friendly names to the ESP8266 GPIO pins is to use
the C/C++ pre-processor directive #define as shown here:

	#define        D0	16	// No PWM on GPIO16
	#define	    D1	 5	// I2C Bus SCL (clock)
	#define	    D2	 4	// I2C Bus SDA (data)
	#define	    D3	 0
	#define	    D4	 2	// Same as "LED_BUILTIN", but inverted logic
	#define	    D5	 14	// SPI Bus SCK (clock)
	#define	    D6	 12	// SPI Bus MISO 
	#define	    D7	 13	// SPI Bus MOSI
	#define	    D8	 15	// SPI Bus SS (CS)
	#define	    D9	 3	// RX0 (Serial console)
	#define	    D10	 1	// TX0 (Serial console)

The advantage of using the #define commands is that they use up no memory in the compiled sketch. Feel free to strip out the comments as well.

I have a  WeMos/LoLin NodeMCU that I bought a week ago and have been playing around (having played with the WeMos D1 board for a few weeks earlier). In testing the NodeMCU, I found out that the RX0 and TX0 pins are on D9 (GPIO3) and D10 (GPIO1) and they were not give names like D0 and D1 because these 2 pins are essentially reserved for Serial console communication. So, of the 11 GPIO pins (and ignoring the 5 GPIO pins used internally by the ESP8266 SoC (i.e. GPIO6, 7, 8, 9, 10 & 11)) that leaves only 9 GPIO pins which conveniently falls within the range of D0 and D8, and so were labeled this way.

Of the nine GPIO pins assigned the Arduino friendly names of D0 thru D8, D0 (GPIO16) is the only one that does not support PWM. The other 8 GPIO pins do support PWM.  The I2C Bus signals SCL and SDA have been assigned to D1 and D2 (GPIO5 & GPIO4) while the four SPI Bus signals (SCK, MISO, MOSI & SS) have been assigned to GPIO pins 14, 12, 13 and 15, respectively). That leaves just 2 pins unassigned. Not much to play with. I guess you need to focus on I2C and SPI compatible devices when interfacing modules to this NodeMCU board.
 
However, the speed of the ESP8266 and convenience of WiFi should make this a popular alternative to the Arduino Uno and Arduino Nano. Now on to reliability testing...

igrr, 2016-04-13:
GPIO16 does support PWM.

randmor, 2016-04-13:
Continuing with igrr's comment...

Another way to define Arduino friendly names to the ESP8266 GPIO pins is to use
the C/C++ pre-processor directive #define as shown here:

#define        D0   16  // No PWM on GPIO16
#define     D1   5  // I2C Bus SCL (clock)
#define     D2   4  // I2C Bus SDA (data)
#define     D3   0
#define     D4   2  // Same as "LED_BUILTIN", but inverted logic
#define     D5   14 // SPI Bus SCK (clock)
#define     D6   12 // SPI Bus MISO 
#define     D7   13 // SPI Bus MOSI
#define     D8   15 // SPI Bus SS (CS)
#define     D9   3  // RX0 (Serial console)
#define     D10  1  // TX0 (Serial console)
The advantage of using the #define commands is that they use up no memory in the compiled sketch. Feel free to strip out the comments as well.

I have a WeMos/LoLin NodeMCU that I bought a week ago and have been playing around (having played with the WeMos D1 board for a few weeks earlier). In testing the NodeMCU, I found out that the RX0 and TX0 pins are on D9 (GPIO3) and D10 (GPIO1) and they were not give names like D0 and D1 because these 2 pins are essentially reserved for Serial console communication. So, of the 11 GPIO pins (and ignoring the 5 GPIO pins used internally by the ESP8266 SoC (i.e. GPIO6, 7, 8, 9, 10 & 11)) that leaves only 9 GPIO pins which conveniently falls within the range of D0 and D8, and so were labeled this way.

Of the nine GPIO pins assigned the Arduino friendly names of D0 thru D8, D0 (GPIO16) is the only one that does not support PWM. The other 8 GPIO pins do support PWM. The I2C Bus signals SCL and SDA have been assigned to D1 and D2 (GPIO5 & GPIO4) while the four SPI Bus signals (SCK, MISO, MOSI & SS) have been assigned to GPIO pins 14, 12, 13 and 15, respectively). That leaves just 3 pins unassigned D0, D1 and D4. Not much to play with. I guess we need to focus on I2C and SPI compatible devices when interfacing modules to this NodeMCU board.

However, the speed of the ESP8266 and convenience of WiFi should make this a popular alternative to the Arduino Uno and Arduino Nano. Now on to reliability testing...

randmor, 2016-04-13:
I stand corrected, igrr! That was a fast response!

===
Continuing with igrr's comment...

Another way to define Arduino friendly names to the ESP8266 GPIO pins is to use
the C/C++ pre-processor directive #define as shown here:

```
#define D0 16    
#define D1   5         // I2C Bus SCL (clock)
#define D2   4         // I2C Bus SDA (data)
#define D3   0
#define D4   2         // Same as "LED_BUILTIN", but inverted logic
#define D5 14         // SPI Bus SCK (clock)
#define D6 12         // SPI Bus MISO 
#define D7 13         // SPI Bus MOSI
#define D8 15         // SPI Bus SS (CS)
#define D9   3         // RX0 (Serial console)
#define D10 1         // TX0 (Serial console)
```

The advantage of using the #define commands is that they use up no memory in the compiled sketch. Feel free to strip out the comments as well.

I have a WeMos/LoLin NodeMCU that I bought a week ago and have been playing around (having played with the WeMos D1 board for a few weeks earlier). In testing the NodeMCU, I found out that the RX0 and TX0 pins are on D9 (GPIO3) and D10 (GPIO1) and they were not give names like D0 and D1 because these 2 pins are essentially reserved for Serial console communication. So, of the 11 GPIO pins (and ignoring the 5 GPIO pins used internally by the ESP8266 SoC (i.e. GPIO6, 7, 8, 9, 10 & 11)) that leaves only 9 GPIO pins which conveniently falls within the range of D0 and D8, and so were labeled this way.

Of the nine GPIO pins assigned the Arduino friendly names of D0 thru D8, D0 (GPIO16) is the only one that does not support PWM. The other 8 GPIO pins do support PWM. The I2C Bus signals SCL and SDA have been assigned to D1 and D2 (GPIO5 & GPIO4) while the four SPI Bus signals (SCK, MISO, MOSI & SS) have been assigned to GPIO pins 14, 12, 13 and 15, respectively). That leaves just 3 pins unassigned D0, D1 and D4. Not much to play with. I guess we need to focus on I2C and SPI compatible devices when interfacing modules to this NodeMCU board.

However, the speed of the ESP8266 and convenience of WiFi should make this a popular alternative to the Arduino Uno and Arduino Nano. Now on to reliability testing...

randmor, 2016-04-13:
I stand corrected, igrr! That was a fast response!

Continuing with igrr's comment...

Another way to define Arduino friendly names to the ESP8266 GPIO pins is to use
the C/C++ pre-processor directive #define as shown here:

#define D0 16    
#define D1   5         // I2C Bus SCL (clock)
#define D2   4         // I2C Bus SDA (data)
#define D3   0
#define D4   2         // Same as "LED_BUILTIN", but inverted logic
#define D5 14         // SPI Bus SCK (clock)
#define D6 12         // SPI Bus MISO 
#define D7 13         // SPI Bus MOSI
#define D8 15         // SPI Bus SS (CS)
#define D9   3         // RX0 (Serial console)
#define D10 1         // TX0 (Serial console)
The advantage of using the #define commands is that they use up no memory in the compiled sketch. Feel free to strip out the comments as well.

I have a WeMos/LoLin NodeMCU that I bought a week ago and have been playing around (having played with the WeMos D1 board for a few weeks earlier). In testing the NodeMCU, I found out that the RX0 and TX0 pins are on D9 (GPIO3) and D10 (GPIO1) and they were not give names like D0 and D1 because these 2 pins are essentially reserved for Serial console communication. So, of the 11 GPIO pins (and ignoring the 5 GPIO pins used internally by the ESP8266 SoC (i.e. GPIO6, 7, 8, 9, 10 & 11)) that leaves only 9 GPIO pins which conveniently falls within the range of D0 and D8, and so were labeled this way.

The I2C Bus signals SCL and SDA have been assigned to D1 and D2 (GPIO5 & GPIO4) while the four SPI Bus signals (SCK, MISO, MOSI & SS) have been assigned to GPIO pins 14, 12, 13 and 15, respectively). That leaves just 3 pins unassigned D0, D1 and D4. Not much to play with. I guess we need to focus on I2C and SPI compatible devices when interfacing modules to this NodeMCU board.

However, the speed of the ESP8266 and convenience of WiFi should make this a popular alternative to the Arduino Uno and Arduino Nano. Now on to reliability testing...

randmor, 2016-04-13:
I stand corrected, igrr! That was a fast response!

Continuing with igrr's comment...

Another way to define Arduino friendly names to the ESP8266 GPIO pins is to use
the C/C++ pre-processor directive #define as shown here:

```
#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
#define D4 2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO 
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9 3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)
```

The advantage of using the #define commands is that they use up no memory in the compiled sketch. Feel free to strip out the comments as well.

I have a WeMos/LoLin NodeMCU that I bought a week ago and have been playing around (having played with the WeMos D1 board for a few weeks earlier). In testing the NodeMCU, I found out that the RX0 and TX0 pins are on D9 (GPIO3) and D10 (GPIO1) and they were not give names like D0 and D1 because these 2 pins are essentially reserved for Serial console communication. So, of the 11 GPIO pins (and ignoring the 5 GPIO pins used internally by the ESP8266 SoC (i.e. GPIO6, 7, 8, 9, 10 & 11)) that leaves only 9 GPIO pins which conveniently falls within the range of D0 and D8, and so were labeled this way.

The I2C Bus signals SCL and SDA have been assigned to D1 and D2 (GPIO5 & GPIO4) while the four SPI Bus signals (SCK, MISO, MOSI & SS) have been assigned to GPIO pins 14, 12, 13 and 15, respectively). That leaves just 3 pins unassigned D0, D1 and D4. Not much to play with. I guess we need to focus on I2C and SPI compatible devices when interfacing modules to this NodeMCU board.

However, the speed of the ESP8266 and convenience of WiFi should make this a popular alternative to the Arduino Uno and Arduino Nano. Now on to reliability testing...

arlaor, 2016-04-22:
How I can use the GPIO module with Arduino IDE?
example
gpio.mode (pin1, gpio.INPUT); // Set pin1 to input mode
gpio.mode (pin2, gpio.OUTPUT); // Set to output mode pin2
gpio.mode (pin3, gpio.INT); // Set to interrupt mode pin3

and not have the problem;
'Gpio' does not name a type
'Gpio' was not Declared In This scope

randmor, 2016-04-22:
==> arlaor
When programming the ESP8266 using the Arduino IDE and development system, you initialize the GPIO pins using the pinMode() function as shown below:

    pinMode(16, OUTPUT);
    pinMode(4, INPUT);
    pinMode(5, INPUT_PULLUP);

The numbers (16, 4, 5) are GPIO pin numbers, and if you have a NodeMCU, these pins are most likely labeled D0, D1, D2... D8. But the Arduino development system needs to be told how to relate the "Arduino Friendly" pin names to the GPIO pin numbers. There are a couple ways to do this: either define D0 thru D8 as integer variables and initialize them to the GPIO pin numbers, or (my favorite way) use the C pre-processor command #define as shown below:

  #define D0 16         // Defines "D0" as "16"

When you compile your code the pre-processor will go thru and substitute in "16" wherever there is a "D0". So, Arduino users don't have to worry about looking up GPIO pin numbers and such, just define it one time and include it in their source file.

The examples you gave (shown again below) **do not look like Arduino "C" code** unless there is some kind of library where a gpio object has been defined and it includes a "mode" member function. 

    gpio.mode (pin1, gpio.INPUT); // Set pin1 to input mode
    gpio.mode (pin2, gpio.OUTPUT); // Set to output mode pin2
    gpio.mode (pin3, gpio.INT); // Set to interrupt mode pin3

So, if I were to rewrite your code, I'd write something like:

````
    // Assign Arduino Friendly Names to GPIO pins
    #define D0 16
    #define D1 5
    #define D2 4
    #define D3 0
    #define D4 2
    #define D5 14
    #define D6 12
    #define D7 13

    void setup() {
      pinMode(D1, INPUT);        // Set pin D1 to input mode (Put Active High PushButton on D1)
      pinMode(D2, OUTPUT);    // Set pin D2 to input mode (Put LED & serial resistor 330 ohms on D2)
    }

   void loop() {
      int value = digitalRead(D1);    // Read state of PushButton
      if (value == HIGH) {
         digitalWrite(D2, HIGH);        // If HIGH, then turn LED on.
      } else {
         digitalWrite(D2, LOW);         // If LOW, then turn LED off.
      }
      delay(250);
    }
````
Try to compile this version. I believe it will work. Also, look under Arduino IDE's File/Examples/Basics for some examples sketches to clue you into how to write "sketches" (C programs) using the Arduino IDE. Also, they have some ESP8266 example sketches under File/Examples/ESP8266. Hopefully this will put you on the right track.

-WRM

arlaor, 2016-04-22:
==> randmor
I have included in my NODEMCU 12E devkit a build custom firmware including GPIO and PWM modules and hoped to use this information (http://nodemcu.readthedocs.org/en/dev/en/modules/gpio/)
but Arduino IDE does not recognize the commands described in the attached link.

My problem is this: I have occupied all the pins of my board and when I use the pins GPIO3 (RX) and GPIO1 (TX) to control the switching on and off of leds by IR receiver sensor connected to GPIO14 (D5), my nodemcu not respond to commands and LEDs remain always on, as if they lost the instruction and remain always LOW HIGH.

When I use only the RX and TX pins as OUTPUT without connecting anything else on the board, they receive the HIGH and LOW commands, but when used in conjunction with other pins, these do not work and remain ON

I expected by instruction gpio.mode () would obtain solution to this problem, but from what I understand, they are equivalent to pinMode () commands.

questions: Is there any way to conver most of the pins (CLK, SD0, CMD, SD1, SD2, SD3, RX and TX) in pins OUTOUT/INPUT? by firmware or any module

randmor, 2016-04-22:
I have also noticed some odd behavior when trying to use GPIO3 (RX), GPIO1 (TX) and GPIO15 (D8) when I was using all the Digital I/O pins (a test I was doing to check out my LoLin NodeMCU board. I found that I had to lift (take out) the jumper to GPIO3 (RX) and GPIO15 (D8) in order to be able to download compiled programs from my PC to the NodeMCU. Once downloaded, I could plug GPIO15 (D8) and GPIO3 (RX) and the sketch would be working. I did notice that if I reset the board, I would also have to lift these 2 pins in order to get it to boot up. Now, I can understand why using RX might mess up downloads or reboots as it is being used to receive data from the PC, but I'm clueless why D8 would also cause this problem. My circuit (as I recall) used an I2C_1602_LCD display on D1 and D2, an 4x4 matrix keypad (Columns on D0, D3, D8, and TX, Rows on D4, D5, D6 and D7). The best I could do was to use TX in place of RX, and then just lift D8 (GPIO15) whenever I downloaded new code or reset the NodMCU. Again, I'm a newbie to the ESP8266, so I don't know all the answers.

What I do know is the ESP8266 is a system on a chip (SoC) design with components like the processor chip bought from another vendor/manufacturer. The processor they use (I don't know the name/number or manufacturer) has 16 GPIO lines, some of which are used internally to interface with other components of the SoC, like flash memory. Because these several lines are used internally within the ESP8266 SoC, in the ESP8266 literature they talk only about 11 GPIO pins, 2 of which are generally reserved for RX and TX in order to communicate with a host PC from which compiled object code is downloaded. This leaves just nine real general purpose I/O pins, but a few of these have secondary functions assigned to them and are supported in software by Arduino libraries like the SPI and WIRE libraries. SPI uses 4 GPIO lines (called SCLK, MISO, MOSI and SS (aka CS)) and I2C uses two GPIO lines (called SDA and SCL). So, that just leaves 3 unassigned GPIO lines (pins). Again, the ESP8266 was designed more of as a WiFi server or WiFi Client, not as a Arduino Mega2560, so the number of GPIO pins was not among their primary concerns. If you need a WiFi board with more I/O capability, you might check out the Arduino MKR1000 board which as I recall has a few more pins available.

To live with what we got, I can offer a few suggestions.  You can use "port expander" chips which are essentially serial-to-parallel shift registers with latched outputs (classic example: 74HC595). The use of shift registers to expand your number of I/O ports can be problematic, especially if you are relying on someone else's library of code to run it. Another option is making some or all of the devices you want to interface SPI or I2C bus compatible. I2C seems to be the most efficient option in terms of pins used. There are just 2, SDA (data) & SCL (clock). Many chip manufacturers use I2C (or SPI) to make their components interface to MCUs more easily, but they also require software to support them. So, having good software development skills is essential to get at the full potential of these devices.

Now if you look around the edges of the LoLin NodeMCU, you will see on the others side (opposite the side with the pins marked D0-D8, RX and TX, you will see a bunch of other pins. Some of these are power, ground and system related pins (EN, RST). We know A0 is the single on board ADC (Analog Input). "VU" is the +5V output from the board that comes from the PC via the USB cable (and Vin is where you can plug in a regulated +5V power supply to externally power the NodeMCU without having to use the USB cable.) Then we have S3, S2, S1, SC, S0 and SK. What are these pins used for? I don't really know. In previous versions of the NodeMCU (made by other manufacturers), most if not all these pins were labeled "Unassigned" as I recall from my random readings about the ESP8266 on the Internet. On more recent pinout diagrams of the NodeMCU and on the silkscreen artwork on the LoLin NodeMCU r3, these pins have been assigned some functionality. So far I have not found an article that describes their use (Can anyone else pipe-in on this?). If you accept the latest pinout diagrams (on GitHub as I recall), then the assigned functions seem to be:

     1.)  A second SPI Bus? Or do these pins just duplicate the SPI bus on this side of the board?
                 S1 = MOSI                 // Is this the same as D7?
                 SC = SS (aka CS)     // Is this the same as D8?
                 S0 = MISO                 // Is this the same as D6?
                 SK = SCLK                 // Is this the same as D5?

And what's the difference between SCLK, MOSI, MISO, CS and HSCLK, HMOSI, HMISO, HCS?
Other people, please pipe in if you know.

     2.)  Two more GPIO pins?
                 S2 = GPIO10      // What? Isn't this GPIO pin being used internally by the ESP8266 SoC?
                 S3 = GPIO9      // What? Isn't this GPIO pin being used internally by the ESP8266 SoC?

It appears that maybe of the latest releases of the ESP8266 maybe they have stream-lined the internal connections and freed up 2 GPIO pins (my best guess). Have yet to test these.

     3.) What are the orange SDIO pins all about? Seems to be another set of secondary functions assigned to the six mysterious pins. Are they supported by an ESP8266 Library or an Arduino Library?  How to use them?  Other folks, please pipe in if you kno [... cut ...]
My testing of the ESP8266 has been sidelined until I figure out several means by which I can interface 3.3V components (MCUs) to 5V components (TTL). I am in the process of testing and documenting three methods by which 3.3V and 5.0V can be interfaced: "Two Resistor Voltage Divider", "The 3 Diodes in Series Method", and my favorite, the "FET Bi-Directional Level Shifter" (using BSS138 FETs). I am also experimenting with lowering the source voltage on 7400 Series HC Family chips like the 74HC04 hex inverter and maybe the 74HC595 shift resister.

I'll see if I can spend more time testing these 6 mysterious pins on the NodeMCU.

-WRM

arlaor, 2016-04-24:
==> randmor
I appreciate the complete information you have given me. Now hopefully the ESP32 solve the limitation of GPIO ports

DeckerDaniel, 2016-06-17:
Dear all, 

I'm a beginner on nodemcu. I buyed a Wemos/Lolin v3 version and I'm testing with the arduino IDE examples. Can be a dummy question, but I tryied with all configurations that I found on the google to turn on the onboard led!
All information says that is connected on GPIO16 (D0) with inverted logic. How can I do it?

On the Schematic diagram I found "S1 Key". I don't see nothing more about this key. 

I wil thank a lot any help

arlaor, 2016-06-17:
you tried this example?

int ledState = LOW;     

unsigned long previousMillis = 0;
const long interval = 1000;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;   
    if (ledState == LOW)
      ledState = HIGH;  // Note that this switches the LED *off*
    else
      ledState = LOW;   // Note that this switches the LED *on*
    digitalWrite(LED_BUILTIN, ledState);
  }
}

randmor, 2016-06-17:
If my recollection is right, the on-board LED is not on the D0 pin, but rather the Tx (aka TxD0) pin. If you orient the LoLin NodeMCU board such that the WiFi antenna is up, you'll be able to read the pin labels. On the right you will see the data pins labled D0, D1, D2, etc. Follow that down till you see TX (maybe Tx or TxD0, I forget) and that should be the pin that is tied to the onboard LED. Since this is active anytime you send data to the Serial console, I don't use it the same as D13 (digital I/O pin 13) on the Arduino Uno. It's nothing to add a separate LED and current limiting resistor between another digital i/o pin and GND and program it to blink in the manor of the Arduino Uno BLINK demo sketch.I haven't use the LoLin NodeMCU for aboout 6 weeks as I have just moved from China to the States and have not unpacked them, my focus on other things more important in settling into my new digs.
-William Moore

randmor, 2016-06-17:
Being not 100% sure of  my previous response, I unpacked my Arduino & ESP8266 stuff and tried out the Blink Sketch on the LoLin NodeMCU board. It does appear from the demo Blink sketch that "LED_BUILTIN" refers to D0 because when I run the Blink sketch it will blink a LED connected to D0. However, the little blue LED on the NodeMCU board does not blink. However, after testings a few of the other  digital I/O lines, I found that D4 (GPIO 2) does correspond to the the little blue on-board LED, and the relations ship is inverted compared to the external LED I had attached to D4 via a current limiting resistor. So, if you want to modify the demo Blink sketch to work on the LoLin NodeMCU board so that the blue on-board LED will blink, change the 3 occurrences of "LED_BUILTIN" to "2" (the GPIO number for the D4 port), save and recompile. When you download it, the blue LED should blink. In my sketch, as I usually do, I use the pre-processor #define command to define "Arduino friendly" names to the GPIO pins using the following block of code, and when I use this code, instead of "2" I substitute in "D4" in place of the the 3 occurrences of "LED_BUILTIN".
#define D0    16
#define D1      5   // I2C Bus SCL (clock)#define D2      4   // I2C Bus SDA (data)#define D3      0#define D4      2   // Also blinks on-board blue LED, but with inverted logic#define D5    14   // SPI Bus SCK (clock)#define D6    12   // SPI Bus MISO #define D7    13   // SPI Bus MOSI#define D8    15   // SPI Bus SS (CS)#define D9      3   // RX0 (Serial console)#define D10    1   // TX0 (Serial console)

Here's the full "corrected" sketch...
=====
/* ESP8266 Blink by Simon Peter Blink the blue LED on the ESP-01 module This example code is in the public domain  The blue LED on the ESP-01 module is connected to GPIO1  (which is also the TXD pin; so we cannot use Serial.print() at the same time)  Note that this sketch uses LED_BUILTIN to find the pin with the internal LED*/#define D0    16#define D1      5   // I2C Bus SCL (clock)#define D2      4   // I2C Bus SDA (data)#define D3      0#define D4      2   // Also blinks on-board blue LED, but with inverted logic#define D5    14   // SPI Bus SCK (clock)#define D6    12   // SPI Bus MISO #define D7    13   // SPI Bus MOSI#define D8    15   // SPI Bus SS (CS)#define D9      3   // RX0 (Serial console)#define D10    1   // TX0 (Serial console)  void setup() {  pinMode(D4, OUTPUT);     // Initialize the LED_BUILTIN pin as an output}
// the loop function runs over and over again forevervoid loop() {  digitalWrite(D4, LOW);      // Turn the LED on (Note that LOW is the voltage level                                            //     but actually the LED is on; this is because                                             //     it is active low on the ESP-01)  delay(1000);                      // Wait for a second  digitalWrite(D4, HIGH);     // Turn the LED off by making the voltage HIGH  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)}
=====
I suppose an easier way to fix the problem would be to re-define "LED_BUILTIN" as "2" by addingthis following line at the top end of the sketch:
#define LED_BUILTIN 2   // Correction for LoLin NodeMCU V3 board where GPIO2 (D4) activates
                                         //   the on-board blue LED.

=====

I also tested my (erroneous) idea that the blue LED is set to the Tx (TX or TxD0) line which should cause the blue LED when sending data to the Serial console. I modified the Blink sketch my adding 'Serial.begin(9600);' to the setup() function and I replaced all the code in the loop() function with 'Serial.println("This is a test... ");'. I was expecting it to also make the blue LED light up, but it does not, so I was was wrong with my initial response to this email thread. Sorry about that. Perhaps it's the RX line it's attached to (otherwise why does the blue LED blink when you upload sketches to the NodeMCU board... that must be it).
Well, hopefully you have figured out the problem by now. My opinion is the documentation for the LoLin NodeMCU basically sucks and YOU ALWAYS NEED TO DOUBLE CHECK what is documented on the internet because most of us work on multiple versions of the ESP8266 and ESP8266 based boards which tend to vary somewhat from one version to the next and most of the documentation is (machine?) translated from sketchy (incomplete) Chinese documentation in the first place. By "double check", I mean to set up an experiment as I did above to verify that the board operates as expected, and if not, keep playing around with it till you find the right answer, or at least a good work around. 
-WRM

  

    On Friday, June 17, 2016 7:23 AM, Randy Moore <randmor@yahoo.com> wrote:
 

 If my recollection is right, the on-board LED is not on the D0 pin, but rather the Tx (aka TxD0) pin. If you orient the LoLin NodeMCU board such that the WiFi antenna is up, you'll be able to read the pin labels. On the right you will see the data pins labled D0, D1, D2, etc. Follow that down till you see TX (maybe Tx or TxD0, I forget) and that should be the pin that is tied to the onboard LED. Since this is active anytime you send data to the Serial console, I don't use it the same as D13 (digital I/O pin 13) on the Arduino Uno. It's nothing to add a separate LED and current limiting resistor between another digital i/o pin and GND and program it to blink in the manor of the Arduino Uno BLINK demo sketch.I haven't use the LoLin NodeMCU for aboout 6 weeks as I have just moved from China to the States and have not unpacked them, my focus on other things more important in settling into my new digs.
-William Moore

DeckerDaniel, 2016-06-20:
randmor, you are completly right.

The output is gpio 2 (D4) inverted logic.

Thanks for the explanation!

kiralikbeyin, 2016-07-11:
#define LED_BUILTIN D4

@randmor thanks 
@igrr will make an update for v3?

elixirml, 2016-10-19:
Hello friends I'm new to developing with esp8266, I'm trying to connect both an RFID reader rc522 and screen nextion 2.4 "in a NodeMCU 1.0, using arduino IDE v 1.6.11 to read an RFID tag and display the UID in Nextion 2.4 screen ".

I managed to connect and work with each separately, but I can not make them work together.

when the connections operate separately in NodeMCU v 1.0 are the following:

Nextion 2.4 "||| MCU Node 1.0

5V-------------- 5V
RX ------------ (D7) - GPIO13 - RXD2 - HMOSI
TX ------------ (D8) - GPIO15 - TXD2 - HCS
GND ----------GND

RC522 ||| MCU Node 1.0

3.3V------ 3.3V
RST ------ (D2) - GPIO4
GND------- GND 
IRQ -----
MISO -------(D6) - GPIO12 - HMISO
MOSI -------(D7) - GPIO13 - RXD2 - HMOSI
SCK --------(D5) - GPIO14 - HSCLK
SS ----------(D4) - GPIO2 - TXD1

Both devices need to use the pin D7 (RXD2), I would like to use both at the same time, I read that I use UART but do not understand how to apply.

any help is welcome
thank you very much

wjbite, 2017-01-26:
Can anyone tell me where to find the library and files that DEFINE
all of the labels that are used in Arduino ISP?
Or maybe just some file names that I may search for.
I am a newby
I am using W10 right now but would rather use Ubuntu.
Arduino ISP on Ubuntu was driving me nuts with file permission problems.
Thanks

guarrod, 2017-06-07:
Hi!, thanks for the support to this issue

I try this code and not works for me, ¿any idea?

`#define LED_BUILTIN D4
#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
#define D4 2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO 
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9 3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)

void setup() {
  pinMode(D4, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop(){  
  digitalWrite(D4, LOW);      // Turn the LED on (Note that LOW is the voltage level                                            
                              //     but actually the LED is on; this is because 
                              //     it is active low on the ESP-01)  
  delay(1000);                // Wait for a second  
  digitalWrite(D4, HIGH);     // Turn the LED off by making the voltage HIGH  
  delay(2000);                // Wait for two seconds (to demonstrate the active low LED)
}`

randmor, 2017-06-07:
Hi Carlos,
I reviewed the code and it looks correct if you are using a NodeMCU board likethose manufactured or distributed by Amica, DoIt or Lolin. There are likely a few other manufacturers who follow this pin numbering scheme like WeMos on the D1/R2 board. My suspicion is that the board you are using map the "logical" pin names/numbers (as labeled on the board) to different GPIO pin numbers. You should consult any documentation you have on your particular board (perhaps you can "google-it" on the Internet). Or, you can sit down andset up 9 LEDs with current limiting resistors (anything from 220 ohm to 1K ohm),and by testing the pins one by one (and taking notes) you can find out which GPIO pin corresponds to which board pin. In the script below, you can changethe value of "testPin" to one of the values included in the comment below.Hope this helps.
/*#define LED_BUILTIN D4
#define D0 16#define D1 5      // I2C Bus SCL (clock)#define D2 4      // I2C Bus SDA (data)#define D3 0#define D4 2      // Same as "LED_BUILTIN", uses inverted logic#define D5 14     // SPI Bus SCK (clock)#define D6 12     // SPI Bus MISO#define D7 13     // SPI Bus MOSI#define D8 15     // SPI Bus SS (CS)#define D9 3      // RX0 (Serial console)#define D10 1     // TX0 (Serial console)
// Above pin mappings suggest you are using a NodeMCU Board? If not// then it's likely that the board you are using uses a different pin numbering// scheme. */
testPin = 16;    // Usable GPIO pins: 0, 2, 4, 5, 12, 13, 14, 15 & 16                      // Do NOT use GPIO pins 6, 7, 8, 9, 10, or 11 as these are used by the
                      // ESP8266 processor to address on-chip flash memory. Use of these
                      // six forbidden pins will likely crash the ESP8266.
void setup() {    pinMode(testPin, OUTPUT); // Initialize the LED_BUILTIN pin as an output} void loop(){    digitalWrite(testPin, LOW);       // Turn the LED on (Note that LOW is the voltage level                                            // but actually the LED is on; this is because                                            // it is active low on the ESP-01)    delay(1000);                      // Wait for a second    digitalWrite(testPin, HIGH);       // Turn the LED off by making the voltage HIGH    delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)}

guarrod, 2017-06-07:
Hi @randmor thanks for the answer.

I Only have the page  the documentation, is this [link](http://www.wch.cn/product/CH340.html), in the back of mi Nodemcu says CH340G.

Really I dont know what is the next step :(

drmpf, 2017-06-07:
I think your code above is on the right track.
Change the delay to say 10 and check each board pin with a multimeter on HZ (or an oscilliscope) to see which one is toggling.
Then try each of the other D's to find out what you mapping actually is.

guarrod, 2017-06-07:
@drmpf I don't have a multimeter =/
I think that I try to find other ways for this problem

duncan-a, 2017-06-07:
Try using a LED and Current limiting resistor. Just make sure you have the
polarity of the LED correct...

randmor, 2017-06-10:
It sounds like it could be one of two NodeMCU boards: a DoIT brand NodeMCU v1.0 or a LoLin Brand. The LoLin NodeMUC board is wider than the "standard" Amica or DoIT brand NodeMCU boards. The LoLin NodeMCU board is the same width as the early v0.9 Amica brand NodeNCU boards which were considered "breadboard" unfriendly because when you plugged one into a standard MB-102 breadboard, all the pinholes were covered, Of course, the solution to that problem is to use a "mini" breadboard along with the MB-102 breadboard and have the LoLin node MCU board straddle the power rails, If you need further details on using a second breadboard, let me know.
The good thing about the NodeMCU boards are the pinouts are mostly compatible, especially with regard to the digital I/O pins.The mappings in the program should work.
#define D0 16#define D1   5      // I2C Bus SCL (clock)#define D2   4      // I2C Bus SDA (data)#define D3   0#define D4   2      // Same as "LED_BUILTIN", uses inverted logic#define D5 14      // SPI Bus SCK (clock)#define D6 12      // SPI Bus MISO#define D7 13      // SPI Bus MOSI#define D8 15      // SPI Bus SS (CS)#define D9   3      // RX0 (Serial console)#define D10 1      // TX0 (Serial console) 
In reading other peopl's responses, there may still be a open question as to how your LEDs are being oriented. LEDslike all diodes will let current flow in one direction, so look at the two wire leads on the LED. The longer one should go to the high side of the circuit and the short wire lead should go to the low side of the circuit. A current limiting resistor (220 ohms to 1K ohms) should be wired in series with the LED. It can be placed on either side of the LEDand it has no polarity, so it will work either way you plug it in. By habit, I generally place the LED between the shortwire lead of the LED and ground. The high side of the LED then is typically wired to a digital I/O pin of the NodeMCU board, like D2. You can test to verify the circuit is working by taking the jumper wire from D2 and plugging it intoeither +5V or +3.3V, either should cause the LED to light up. If not, try flipping the LED around in the circuit. If that does not work, try another LED.
If the NodeMCU board you are using still is not working for you, and you have another one on hand, try swapping NodeMCU boards.I can't think what else it could be,

jimsy3, 2017-07-23:
hi.
this is really confusing somehow....
1) Wemos D1 has like two D5 pins, and when the SCK pin is called GPIO14(on the IOH jumper), then what is the other D5 pin called that is on the IOL jumper(together with Tx and Rx etc)?
https://wiki.wemos.cc/_media/products:d1:d1_v2.0.0.pdf
2) and if they are called the same(GPIO14), then how does the software know where to read the input data?

3) i guess the IOL14 is the same as Pin7 on Arduino Uno and IOL16 aka D0 is the same as Uno's Pin2? so, would i be better of to use D0 instead of D5 then? (as i need to declare the pin number in IDE software)

devyte, 2017-07-23:
I don't have a D1, but I do have a a bunch of D1 minis. I think they're the
same pin, they're just duplicated on the headers for convenience and
compatibility, so that you can use different shields.

Ashish05178, 2017-08-11:
Hello,

First of all thanks for all your work, without that I could not do this small project on university.
I have a newbie question, but first I show you my small code:
 
#include <Stepper.h>
 
int in1Pin = 12;
int in2Pin = 11;
int in3Pin = 10;
int in4Pin = 9;
 
// change this to the number of steps on your motor
#define STEPS 512
 
Stepper motor(STEPS, in1Pin, in2Pin, in3Pin, in4Pin); 
 
void setup()
{
  pinMode(in1Pin, OUTPUT);
  pinMode(in2Pin, OUTPUT);
  pinMode(in3Pin, OUTPUT);
  pinMode(in4Pin, OUTPUT);
 
  // this line is for Leonardo's, it delays the serial interface
  // until the terminal window is opened
  while (!Serial);
   
  Serial.begin(9600);
  motor.setSpeed(100);
}
 
void loop()
{
  if (Serial.available())
  {
    int steps = Serial.parseInt();
    motor.step(steps);
  }
}

This code is for arduino boar
Based on a code made available on http://www.circuitmagic.com
can anyone please tell what changes I have to do in this code for nodemcu---For pin D0,D1,D2,D3

devyte, 2017-08-11:
There are D0...Di #defines available for most boards. Just assign directly
to your pins:
int pin1 = D0;
...

Ashish05178, 2017-08-11:
These types of error coming in COM5 after uploading program
"⸮⸮⸮⸮⸮|⸮⸮⸮⸮z9⸮⸮B⸮⸮⸮J~⸮⸮⸮⸮⸮z9⸮⸮⸮⸮z9⸮⸮⸮"⸮⸮|⸮3⸮⸮⸮|⸮⸮h⸮P⸮z9⸮⸮⸮⸮⸮⸮|⸮⸮⸮⸮z9⸮⸮⸮⸮⸮⸮z9⸮⸮B⸮⸮⸮J~⸮⸮⸮P⸮z9⸮⸮⸮9⸮J⸮⸮"
Duplicate of #devyte  There are D0...Di #defines available for most boards. Just assign directly
to your pins:
int pin1 = D0;

treii28, 2017-08-24:
I see on some pinouts, they show the SDD2 and SDD3 also labelled as gpio9 and gpio10. If you weren't using anything that required the additional SDD pins, can you map these as additional digital pins? (i.e. as D11 and D12?)

I also see no standard per se for I2C although there does appear to be for SPI. The closest I see is a couple examples of people using D1 (GPIO5) for SCL and D2 (GPIO4) for SDA pins. I'm assuming they grab these two because they aren't really tasked for anything else as many of the others are. Are these the best two to use for I2C?

Simon10901, 2018-02-14:
i don't know what i am doing wrong but somehow i cant get the voltage of D0 (pin16) LOW while LED_BUILTIN ist HIGH. My Problem is, that most of the time the board does nothing. Its just connected to WiFi and waits for commands. I do not want the blue LED to glow all the time. But if i turn it off with digitalWrite(LED_BUILTIN, HIGH) D0 also turns HIGH.
anyone can explain why this is happening and what to do?

lrmoreno007, 2018-02-15:
In NodeMcu, the Led is placed connecting the cathode with GPIO16. This causes the led turn on when GPIO16 is LOW.

[image]

Therefore the led has inverted logic, but not the GPIO.

This is a good practice that prevents the LED from acting as a pulldown.

jimboD, 2018-04-02:
I am trying to turn on and off an LED connected to digital pin 13, when pressing a pushbutton attached to pin 2, on a breadboard and using the ESP8266MOD ai-thinker.

Here is the code I'm using:

// constants won't change. They're used here to
// set pin numbers:
const int buttonPin = 2;     // the number of the pushbutton pin
const int ledPin =  13;      // the number of the LED pin

// variables will change:
int buttonState = 0;         // variable for reading the pushbutton status

void setup() {
  // initialize the LED pin as an output:
  pinMode(ledPin, OUTPUT);
  // initialize the pushbutton pin as an input:
  pinMode(buttonPin, INPUT);
}

void loop() {
  // read the state of the pushbutton value:
  buttonState = digitalRead(buttonPin);

  // check if the pushbutton is pressed.
  // if it is, the buttonState is HIGH:
  if (buttonState == HIGH) {
    // turn LED on:
    digitalWrite(ledPin, HIGH);
  } else { 
    // turn LED off:
    digitalWrite(ledPin, LOW);
  }
}

Despite the fact that the code calls for the push button to be attached at pin 2, I have it attached at D4.  Also, the LED is attached at D7, which corresponds to pin 13. 
Here is a photo:
[image]

My problem - As soon as I plug the USB into the computer, the red LED at D7 comes on and stays on.  When I push the button, the blue onboard LED comes on and when I let up on the button, it goes off.  In the meantime, the red LED just stays on.

Any ideas?

devyte, 2018-04-02:
@jimboD not the right place to seek help.
D4 is gpio2, and D7 is gpio13. Gpio2 is also where the onboard blue led is attached. That's why the button seems to manipulate it.
The red led should most likely not be attached directly to the pin, but rather over a resistor. A led doesn't really have internal resistance, and only has a semiconductor voltage drop of the order of 0.7V (less for newer leds), so there is nothing to limit current (pin outputs 3.3V). You may have burned out that pin.

jimboD, 2018-04-02:
Ahh, I see.  Thank you.  But what do you mean "not the right place to seek help"?  You helped me.

vdeconinck, 2018-04-03:
@jimboD The fact that you are able to park on a crosswalk does not make a crosswalk the right place to park :-)
Indeed devbyte was kind enough to give you insightful hints, but it's not the right place because (a) this is an issue tracker for bugs found in the Arduino core dor ESP8266 (b) you are hijacking a 3-year old discussion and (c) this issue is closed.
Questions and help requests about esp8266 cores or esp8266 in general should be directed to forums such as esp8266.com or stackoverflow.

jimboD, 2018-04-03:
Thank you very much.

Elias323232, 2019-01-11:
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

anitakenchannavar, 2019-08-22:
#define trigpin D0
#define echopin D1
long duration, dist;
float distance; 
int action;

void setup() 
{ 
 pinMode(trigpin, OUTPUT); // D0 - TrigPin
 pinMode(echopin, INPUT);  // D1 - EchoPin
 Serial.begin(9600);

} 

void loop() 
{ 
 digitalWrite(trigpin, LOW); 
 delayMicroseconds(2); 
 digitalWrite(trigpin, HIGH); 
 delayMicroseconds(10); 
 digitalWrite(trigpin, LOW); 

 duration = pulseIn(echopin, HIGH); 

 dist = (duration*0.343)/2; 

 distance = dist / 100.0;
Serial.println(distance);

delay(2000);

} 

Hello everyone,
I am a newbie working on IOT projects.  This is my code which I am interfacing with NodeMcu Amica Board. I am not sure of the pin numbers. In the above code I am getting an error saying D0 not defined in this scope. How to understand the pin numbers.

desireesantos, 2019-08-22:
What board (ESP) have you selected in the IDE ?

desireesantos, 2019-08-22:
Another option is to use the pin numbers instead of constant. (Reference
here <https://github.com/esp8266/Arduino/issues/584#issuecomment-123715951>)

static const uint8_t D0 = 16; static const uint8_t D1 = 5;
static const uint8_t D2 = 4;
static const uint8_t D3 = 0;
static const uint8_t D4 = 2;
static const uint8_t D5 = 14;
static const uint8_t D6 = 12;
static const uint8_t D7 = 13;
static const uint8_t D8 = 15;
static const uint8_t RX = 3;
static const uint8_t TX = 1;

timpulver, 2019-08-23:
For **NodeMCU** users:

Select **NodeMCU 1.0 (ESP-12E Module)**, _not_ Generic ESP8622 Module. Even if you are using a NodeMCU V2. After doing so the pin mappings (e.g. for the digital pins) printed on the board can be used without defining them manually.

The following will work:

```
digitalWrite(D0, HIGH);
```

desireesantos, 2019-08-23:
Agree completely @Tim Pulver !!
It's a typical error message around setup board

Sanjayprakash-p, 2024-03-01:
thanks alot bro..worked