esp8266/Arduino #592: Eclipse and 1.6.5-939

czopas, 2015-07-23:
After using 1.6.5-939 version from staging I have a problem with compiling code in Eclipse. Below info from eclipse console:

```
Starting combiner
"C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/../../../tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-gcc" -g -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/tools/sdk//lib" "-LC:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/tools/sdk//ld" "-Teagle.flash.4m.ld" -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:/ArduinoEclipse/workspace/iot4home_v7/Release/iot4home_v7.elf" -Wl,--start-group    ./iot4home_v7.cpp.o  ./Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o ./Libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp.o ./Libraries/ESP8266WiFi/src/WiFiClient.cpp.o ./Libraries/ESP8266WiFi/src/WiFiServer.cpp.o ./Libraries/ESP8266WiFi/src/WiFiUdp.cpp.o  ./Libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o ./Libraries/ESP8266WebServer/src/Parsing.cpp.o  ./Libraries/EEPROM/EEPROM.cpp.o  ./Libraries/DHT/dht.cpp.o   arduino.ar   "C:/ArduinoEclipse/workspace/iot4home_v7/Release/arduino.ar" -lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -Wl,--end-group  "-LC:/ArduinoEclipse/workspace/iot4home_v7/Release"
./Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o: In function `wifi_dns_found_callback(char const*, ip_addr*, void*)':
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:577: undefined reference to `wifi_wps_disable'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:579: undefined reference to `wifi_wps_disable'
./Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o: In function `ESP8266WiFiClass::begin(char const*, char const*, int, unsigned char*)':
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:70: undefined reference to `wifi_wps_enable'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:72: undefined reference to `wifi_set_wps_cb'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:72: undefined reference to `wifi_wps_start'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:83: undefined reference to `wifi_wps_disable'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:89: undefined reference to `wifi_wps_enable'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:92: undefined reference to `wifi_set_wps_cb'
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp:94: undefined reference to `wifi_wps_start'
collect2.exe: error: ld returned 1 exit status
```

Eclipse say that problem is in undefinied reference - anyone can help ?

CptanPanic, 2015-07-23:
I am not sure of your problem, as I do not know how compiling from eclipse works, but wifi_wps_disable is part of the espressif sdk 1.2, so I would make sure you are linking all the libraries in https://github.com/esp8266/Arduino/tree/0cc192525fe4d9f784361408dbb82fff2fb96b39/hardware/esp8266com/esp8266/tools/sdk/lib

czopas, 2015-07-23:
I downloded all files from https://github.com/esp8266/Arduino/tree/0cc192525fe4d9f784361408dbb82fff2fb96b39/hardware/esp8266com/esp8266/tools/sdk/lib and put them to:
```
(...)\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-939-g033463a\tools\sdk\lib\ 
```
but it still didn't work. For now I have other error message:
```
Starting combiner
"C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/../../../tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/xtensa-lx106-elf-gcc" -g -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/tools/sdk//lib" "-LC:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/tools/sdk//ld" "-Teagle.flash.4m.ld" -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:/ArduinoEclipse/workspace/iot4home_v7/Release/iot4home_v7.elf" -Wl,--start-group    ./iot4home_v7.cpp.o  ./Libraries/ESP8266WiFi/src/ESP8266WiFi.cpp.o ./Libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp.o ./Libraries/ESP8266WiFi/src/WiFiClient.cpp.o ./Libraries/ESP8266WiFi/src/WiFiServer.cpp.o ./Libraries/ESP8266WiFi/src/WiFiUdp.cpp.o  ./Libraries/ESP8266WebServer/src/ESP8266WebServer.cpp.o ./Libraries/ESP8266WebServer/src/Parsing.cpp.o  ./Libraries/EEPROM/EEPROM.cpp.o  ./Libraries/DHT/dht.cpp.o   arduino.ar   "C:/ArduinoEclipse/workspace/iot4home_v7/Release/arduino.ar" -lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -Wl,--end-group  "-LC:/ArduinoEclipse/workspace/iot4home_v7/Release"
C:/Users/dopalo/AppData/Roaming/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-939-g033463a/tools/sdk//lib\libhal.a: file not recognized: File format not recognized
collect2.exe: error: ld returned 1 exit status
make: *** [iot4home_v7.elf] Error 1
```

:(

czopas, 2015-07-23:
I found a solution. I coppied old platform.txt from 1.6.5-804-g2d340c7 because eclipse need additional lines for compilation:
```
runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/../../../tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9
runtime.tools.esptool.path={runtime.platform.path}/../../../tools/esptool/0.4.5
```

In the new platform.txt this line is modifed:
```
compiler.c.elf.libs=-lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -lwps -lcrypto
```
by adding -lwps and this was a problem :)

closed 2015-07-23 by czopas

Links2004, 2015-07-23:
for wps take a look here: 
https://github.com/Links2004/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L691