esp8266/Arduino #596: Crash when sending content from flash to client using the webserver

rnplus, 2015-07-23:
I get this error using the stable, staging and latest (compiled from source) version of the arduino package.

```
Exception (3):
epc1=0x4000bef4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40243566 depc=0x00000000
 
ctx: cont
sp: 3ffec940 end: 3ffecbd0 offset: 01a0
 
>>>stack>>>
3ffecae0:  40243566 00000000 00000000 00000030  
3ffecaf0:  00000000 00000000 3ffeaec8 4020244a  
3ffecb00:  3fff7120 00004526 00004526 00000030  
3ffecb10:  3fff65b8 00000000 00000000 4020742e  
3ffecb20:  00000000 00000000 3ffeaec8 40207a7d  
3ffecb30:  00000011 3fff6c08 00000003 00000003  
3ffecb40:  3fff7060 0000001e 0000001e 402052bd  
3ffecb50:  3ffea360 00000000 000003e8 3ffebbb0  
3ffecb60:  00000000 3fff70e8 0101a8c0 3ffecbfc  
3ffecb70:  3fffdc20 000003e9 3ffeaec8 40207b6d  
3ffecb80:  3ffea360 00000000 000003e8 000147a4  
3ffecb90:  3ffeaee0 3fff70e8 00000016 40102185  
3ffecba0:  40201cdd 00000000 00000000 402044f6  
3ffecbb0:  3fffdc20 00000000 3ffecbf4 40201d3a  
3ffecbc0:  00000000 00000000 3ffebbb0 40100378  
<<<stack<<<
 
 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
 
 
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)
 
wdt reset
 
```

``` 
server.on ("/style.css", []() { server.send ( 200, "text/css", DATA_css ); } );
 ```

Data:
```
const char DATA_css[] PROGMEM = R"=====(
body {margin: 0px; padding: 0px; font-family: "Open Sans", sans-serif; font-color: #000; font-size: 14px; background-color: #F2F2F2;}
div.header {position: absolute; left: 0px; right: 0px; top: 0px; background-color: #363636; color: #FFF; margin: 0px; padding: 0px; padding-bottom: 0px; height: 55px;}
div.header h1 {padding: 0px; margin: 0px; padding-top: 5px;}
div.header h1 a {color: #FFFFFF; text-decoration: none;}
div.header .float_left {float: left; padding-top: 5px;padding-left: 10px;}
div.header .float_left h1 {margin: 0px; float: left; padding: 0px; padding-left: 5px; padding-right: 10px; font-size: 35px; font-weight: bold;}
div.header .float_left h1 .logo1 {color: #00AEEF;}
div.header .float_left h2 {margin: 0px; float: left; padding: 0px; padding-left: 5px; padding-right: 10px; font-size: 20px; padding-top: 10px;}
div.header .float_left form {float: left; margin-top: 3px;}
div.header .float_right {float: right; margin: 0px; margin-right: 20px; padding: 0px; height: 100%;}
.mobile_menu {width: 100%; margin-top: 50px; font-size: 30px; text-align: center; }
.mobile_menu a {display: block; width: 100%; color: #FFFFFF; height: 40px; background-color: #363636; text-decoration: none; }
.mobile_menu a:hover {background-color: #00AEEF;}
div.header .float_right a {display: inline-block; color: #FFF; text-decoration: none; padding-bottom: 0px; padding-top: 20px; padding-left: 10px; padding-right: 10px; margin: 0px; height: 35px;}
div.header .float_right a:hover {background-color: #00AEEF;}
.main{width: 600px; margin: 0 auto; margin-top: 100px; padding: 15px; border-color:#000000; border-width: 1px; border-style: solid; background-color: #FFFFFF;}
.main table { border-collapse: true; border-width: 0px; }
.main td { border-collapse: true; border-width: 1px; border-style: solid; border-color: #CCCCCC;}
.home a { display: block; height: 20px; width: auto; text-decoration: none; background-color: #363636; color: #FFFFFF; padding: 5px; }
.home a:hover { background-color: #00AEEF; }
.row {width: 100%; height: 40px; font-size: 15px; margin: 0px; }
.row .left { float: left; padding-top: 7px; }
.row .right {float: right; }
hr{ color: #000; background-color: #000; height: 1px; border: none; width: 100%;}
.center { float: center; width: 100%; text-align: center;}
.center table {text-align: left; width: 150px; margin-left: 75px;}
@media (min-width: 500px) {.home{ width: 300px; margin: 0 auto; padding: 15px; border-color:#DADADA; border-width: 1px; border-style: solid; background-color: #FFFFFF;} .wide{ width: 500px;}}
@media (max-width: 499px) {.home{ left: 3px; right: 3px; margin: 0px; padding: 0px; border-color:#DADADA; border-width: 1px; border-style: solid; background-color: #FFFFFF;}}
@media (max-width: 699px) {div.header .float_right { visibility: hidden;}}
@media (min-width: 700px) {.mobile_menu {visibility: hidden; height: 0px;}}
)=====";
```

igrr, 2015-07-23:
```server.send``` doesn't seem to be compatible with PROGMEM variables. It expects the data to be in RAM, and doesn't take possible alignment issues into account.
Adding PROGMEM support to WebServer would be an improvement.

rnplus, 2015-07-23:
The weird thing is that this function works, even though the data is in flash.
This same function fails with the same exception when using it on the css stylesheet that I posted earlier.

```
void handleLogo() {
  WiFiClient client = server.client();
  handleStaticHeader(client, sizeof(DATA_logo), "image/png");
  client.write(DATA_logo, sizeof(DATA_logo));
  client.stop();
}
```

----

Because flash data apparently isn't supported that well, how can I copy the flash data to a buffer in ram before sending it?

igrr, 2015-07-23:
Yes you can copy PROGMEM data into RAM via String:
```C++
String content(DATA_css);
server.send(200, "text/css", content);
```

rnplus, 2015-07-23:
server.on ("/style.css", []() { server.send(server.client(), "text/css", String(DATA_css) ); } );

Still crashes the ESP

```
Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40248089 depc=0x00000000

ctx: cont 
sp: 3ffec930 end: 3ffecbd0 offset: 01a0

>>>stack>>>
3ffecad0:  3ffeaec8 3ffe8c54 00000001 402039b5  
3ffecae0:  3ffea360 00000000 000003e8 00000030  
3ffecaf0:  3ffecb80 3fff7160 00000000 00000000  
3ffecb00:  00000000 3fffdc20 3ffecbfc 00000030  
3ffecb10:  3fff65b8 00000000 3ffeaec8 4020743e  
3ffecb20:  00000000 00000000 3ffeaec8 40207a8d  
3ffecb30:  0000000a 3fff70b0 00000003 00000003  
3ffecb40:  3fff70e8 00000017 00000017 3ffebaa0  
3ffecb50:  3fffdc20 00000000 40201cbe 3ffebbb0  
3ffecb60:  402015a2 00000001 3ffeaec8 3ffecbfc  
3ffecb70:  3fffdc20 00000396 3ffeaec8 40207b7d  
3ffecb80:  3ffea360 00000000 000003e8 00009ef9  
3ffecb90:  3ffeaee0 3fff7160 00000016 40102185  
3ffecba0:  40201cdd 00000000 00000000 40204506  
3ffecbb0:  3fffdc20 00000000 3ffecbf4 40201d3a  
3ffecbc0:  00000000 00000000 3ffebbb0 40100378  
<<<stack<<<
```

rnplus, 2015-07-23:
I uploaded my sketch to my webserver, you can download it here: http://rnpl.us/code.zip

Maybe that helps while debugging the problem.

igrr, 2015-07-23:
/cc @Makuna

patjazz, 2015-07-23:
- I have the same problem ( http://www.esp8266.com/viewtopic.php?f=29&t=4060 ). It seems to me important as RAM is small and FLASH is the good place for all static data in a WEB/net environnement.
- I have the same problem with fonts datas... :-(
- I have the idea that 'multiple' PROGMEM is the problem ?

Regards... And thanks for all the job you are doing !

rnplus, 2015-07-23:
#patjazz: Sometimes it works with two progmem instances, but I've not yet been able to find a pattern in when it works and when it doesn't. My guess is that it fails above a certain size.

(Wild guess, big chance it's bullshit: could it be that the flash is divided in pages and that the first PROGMEM instance fits into the first page but the second does not? (Then while reading the ESP fails because it only reads the first page and then tries to read more data than what it read from the flash?)

martinayotte, 2015-07-23:
My thought about new WebServer needs : http://www.esp8266.com/viewtopic.php?f=29&t=4060&p=24039#p24039

Makuna, 2015-07-23:
With Progmem, you really need to look closely at the Arduino docs, the way it was original designed is not how I would have done it, but it is the way it is in this project for compatibility.

The core of the problem is, the compiler can't tell the difference between these two variables to decide which function to call; as the PROGMEM doesn't not effect the type, it only effects where in memory the data is stored.
```
const char stringInFlash[] PROGMEM = "in flash";
const char stringInRam[] = "in ram";
```
But the memory location they are stored at have different requirements.  Specifically, in flash, the data pointer must always be 32bit aligned and reads are always 32bits.  But the same code reads both flash and ram.  So if the code tries to read one byte at a time, it will cause an exception. 

If you use the above "stringInFlash" variable in program method (end in _P), then there is an implied knowledge that the passed variable is in flash and it does the right thing in the method.
If you use the same variable to try to call a method that has versions that can take flash memory, then the variable is best defined differently so that the right method is picked by the compiler.

So, in your example, you have
```
server.on ("/style.css", []() { 
        String content(DATA_css); 
        server.send(server.client(), "text/css", content ); 
    } );
```
since it can't tell DATA_css is in flash, you need to let it know, by casting it like this
```
server.on ("/style.css", []() { 
        String content(reinterpret_cast<const __FlashStringHelper *>(DATA_css)); 
        server.send(server.client(), "text/css", content ); 
    } );
```

FYI: I glanced at the Server code to see how hard it would be to convert to have a send_P(), unfortunately it is not simple; but in the long run I agree it would be interesting.

patjazz, 2015-07-23:
Why the compiler and/or linker can't do an ALIGN 32 ? even if we loose 3 bytes ? as they know that the data will stay in flash ? (if it is an alignment problem....)

Makuna, 2015-07-23:
@patjazz Its not just pointer alignment, you also can't read a single byte, which most string methods  assume they can do.  You can only read 32bits.

martinayotte, 2015-07-24:
Hi Makuna,
I think that adding new send_P() and sendContent_P() functions should be quite straight forward, (except if there are pitfalls somewhere I didn't saw beforehand).
Although I miss the famous "time ingredient" these days, maybe I can try to get it patched, but that won't warranty to get it into an official PR soon.

Makuna, 2015-07-24:
@martinayotte If it still has to allocate the full buffer inside to copy memory, sure, but I think the right solution is to try to "chunk" it so the memory allocation is smaller allowing a large content size to be used without taking a larger memory footprint.

Links2004, 2015-07-24:
Note: more then 1460 Byte can note be send by TCP at one. so bigger buffer make no sense.

Makuna, 2015-07-24:
p.s.  I am working on exposing a send_P();
```
void send_P(int code, PGM_P content_type, PGM_P content);
```

igrr, 2015-07-24:
Great!
IMO content_type doesn't really need to be PGM_P, it's just a few bytes.

Makuna, 2015-07-24:
If the compiler doesn't do string duplication removal, and you call send_P is many locations, those can add up quickly.  
I am sure there are some people who would want it, so rather than add one that does and one that doesn't, it seems cleaner to just require it.  Thoughts?

rnplus, 2015-07-24:
Can't you give it a pointer to a const char[] PROGMEM?
That content would always be a the location given by the pointer and thus only in flash once?

Makuna, 2015-07-24:
@mplus this is what a PGM_P is.

rnplus, 2015-07-24:
ah, okay

martinayotte, 2015-07-24:
@Makuna, Great ! 
So, you're quicker than me (I didn't started yet, so I leave you the work :-) )

Makuna, 2015-07-26:
@mplus pointer uses are not commonly used by Arduino AVR developers, so you may see code where the same literal string is used like...
'''
Serial.print (F ("dup"));

...

Serial.print (F ("dup"));
'''
In this case, the compiler isn't smart enough with esp8266 to pool these, but with the AVR compiler will pool them, this difference is why this issue is raised.

rnplus, 2015-08-21:
I tried using the send_P function but it still causes crashes.

server.on ("/style.css", []() { Serial.println("CSS"); server.send_P( 200, "text/css", DATA_css ); } );

const char DATA_css[] PROGMEM = R"=====(
*A LOT OF CSS HERE*
)=====";

It crashes with the following message:

ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

Makuna, 2015-08-22:
@rnplus the content type is also progmem string, but you are not passing one.

rnplus, 2015-08-24:
const char MIME_png[] PROGMEM = "image/png";
const char DATA_logo[] PROGMEM = { //PNG
  0x89, 0x50, 0x4e, 0x47, 0x0d, 0x0a, 0x1a, 0x0a, 0x00, 0x00, 0x00, 0x0d, //Logo
  0x49, 0x48, 0x44, 0x52, 0x00, 0x00, 0x00, 0x32, 0x00, 0x00, 0x00, 0x29,
  0x08, 0x03, 0x00, 0x00, 0x00, 0x40, 0xf0, 0x8a, 0xdb, 0x00, 0x00, 0x00,
  0x03, 0x73, 0x42, 0x49, 0x54, 0x08, 0x08, 0x08, 0xdb, 0xe1, 0x4f, 0xe0,
  0x00, 0x00, 0x03, 0x00, 0x50, 0x4c, 0x54, 0x45, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x33, 0x00, 0x00, 0x66, 0x00, 0x00, 0x99, 0x00, 0x00, 0xcc, 0x00,
  0x00, 0xff, 0x00, 0x33, 0x00, 0x00, 0x33, 0x33, 0x00, 0x33, 0x66, 0x00,
  0x33, 0x99, 0x00, 0x33, 0xcc, 0x00, 0x33, 0xff, 0x00, 0x66, 0x00, 0x00,
  0x66, 0x33, 0x00, 0x66, 0x66, 0x00, 0x66, 0x99, 0x00, 0x66, 0xcc, 0x00,
  0x66, 0xff, 0x00, 0x99, 0x00, 0x00, 0x99, 0x33, 0x00, 0x99, 0x66, 0x00,
  0x99, 0x99, 0x00, 0x99, 0xcc, 0x00, 0x99, 0xff, 0x00, 0xcc, 0x00, 0x00,
  0xcc, 0x33, 0x00, 0xcc, 0x66, 0x00, 0xcc, 0x99, 0x00, 0xcc, 0xcc, 0x00,
  0xcc, 0xff, 0x00, 0xff, 0x00, 0x00, 0xff, 0x33, 0x00, 0xff, 0x66, 0x00,
  0xff, 0x99, 0x00, 0xff, 0xcc, 0x00, 0xff, 0xff, 0x33, 0x00, 0x00, 0x33,
  0x00, 0x33, 0x33, 0x00, 0x66, 0x33, 0x00, 0x99, 0x33, 0x00, 0xcc, 0x33,
  0x00, 0xff, 0x33, 0x33, 0x00, 0x33, 0x33, 0x33, 0x33, 0x33, 0x66, 0x33,
  0x33, 0x99, 0x33, 0x33, 0xcc, 0x33, 0x33, 0xff, 0x33, 0x66, 0x00, 0x33,
  0x66, 0x33, 0x33, 0x66, 0x66, 0x33, 0x66, 0x99, 0x33, 0x66, 0xcc, 0x33,
  0x66, 0xff, 0x33, 0x99, 0x00, 0x33, 0x99, 0x33, 0x33, 0x99, 0x66, 0x33,
  0x99, 0x99, 0x33, 0x99, 0xcc, 0x33, 0x99, 0xff, 0x33, 0xcc, 0x00, 0x33,
  0xcc, 0x33, 0x33, 0xcc, 0x66, 0x33, 0xcc, 0x99, 0x33, 0xcc, 0xcc, 0x33,
  0xcc, 0xff, 0x33, 0xff, 0x00, 0x33, 0xff, 0x33, 0x33, 0xff, 0x66, 0x33,
  0xff, 0x99, 0x33, 0xff, 0xcc, 0x33, 0xff, 0xff, 0x66, 0x00, 0x00, 0x66,
  0x00, 0x33, 0x66, 0x00, 0x66, 0x66, 0x00, 0x99, 0x66, 0x00, 0xcc, 0x66,
  0x00, 0xff, 0x66, 0x33, 0x00, 0x66, 0x33, 0x33, 0x66, 0x33, 0x66, 0x66,
  0x33, 0x99, 0x66, 0x33, 0xcc, 0x66, 0x33, 0xff, 0x66, 0x66, 0x00, 0x66,
  0x66, 0x33, 0x66, 0x66, 0x66, 0x66, 0x66, 0x99, 0x66, 0x66, 0xcc, 0x66,
  0x66, 0xff, 0x66, 0x99, 0x00, 0x66, 0x99, 0x33, 0x66, 0x99, 0x66, 0x66,
  0x99, 0x99, 0x66, 0x99, 0xcc, 0x66, 0x99, 0xff, 0x66, 0xcc, 0x00, 0x66,
  0xcc, 0x33, 0x66, 0xcc, 0x66, 0x66, 0xcc, 0x99, 0x66, 0xcc, 0xcc, 0x66,
  0xcc, 0xff, 0x66, 0xff, 0x00, 0x66, 0xff, 0x33, 0x66, 0xff, 0x66, 0x66,
  0xff, 0x99, 0x66, 0xff, 0xcc, 0x66, 0xff, 0xff, 0x99, 0x00, 0x00, 0x99,
  0x00, 0x33, 0x99, 0x00, 0x66, 0x99, 0x00, 0x99, 0x99, 0x00, 0xcc, 0x99,
  0x00, 0xff, 0x99, 0x33, 0x00, 0x99, 0x33, 0x33, 0x99, 0x33, 0x66, 0x99,
  0x33, 0x99, 0x99, 0x33, 0xcc, 0x99, 0x33, 0xff, 0x99, 0x66, 0x00, 0x99,
  0x66, 0x33, 0x99, 0x66, 0x66, 0x99, 0x66, 0x99, 0x99, 0x66, 0xcc, 0x99,
  0x66, 0xff, 0x99, 0x99, 0x00, 0x99, 0x99, 0x33, 0x99, 0x99, 0x66, 0x99,
  0x99, 0x99, 0x99, 0x99, 0xcc, 0x99, 0x99, 0xff, 0x99, 0xcc, 0x00, 0x99,
  0xcc, 0x33, 0x99, 0xcc, 0x66, 0x99, 0xcc, 0x99, 0x99, 0xcc, 0xcc, 0x99,
  0xcc, 0xff, 0x99, 0xff, 0x00, 0x99, 0xff, 0x33, 0x99, 0xff, 0x66, 0x99,
  0xff, 0x99, 0x99, 0xff, 0xcc, 0x99, 0xff, 0xff, 0xcc, 0x00, 0x00, 0xcc,
  0x00, 0x33, 0xcc, 0x00, 0x66, 0xcc, 0x00, 0x99, 0xcc, 0x00, 0xcc, 0xcc,
  0x00, 0xff, 0xcc, 0x33, 0x00, 0xcc, 0x33, 0x33, 0xcc, 0x33, 0x66, 0xcc,
  0x33, 0x99, 0xcc, 0x33, 0xcc, 0xcc, 0x33, 0xff, 0xcc, 0x66, 0x00, 0xcc,
  0x66, 0x33, 0xcc, 0x66, 0x66, 0xcc, 0x66, 0x99, 0xcc, 0x66, 0xcc, 0xcc,
  0x66, 0xff, 0xcc, 0x99, 0x00, 0xcc, 0x99, 0x33, 0xcc, 0x99, 0x66, 0xcc,
  0x99, 0x99, 0xcc, 0x99, 0xcc, 0xcc, 0x99, 0xff, 0xcc, 0xcc, 0x00, 0xcc,
  0xcc, 0x33, 0xcc, 0xcc, 0x66, 0xcc, 0xcc, 0x99, 0xcc, 0xcc, 0xcc, 0xcc,
  0xcc, 0xff, 0xcc, 0xff, 0x00, 0xcc, 0xff, 0x33, 0xcc, 0xff, 0x66, 0xcc,
  0xff, 0x99, 0xcc, 0xff, 0xcc, 0xcc, 0xff, 0xff, 0xff, 0x00, 0x00, 0xff,
  0x00, 0x33, 0xff, 0x00, 0x66, 0xff, 0x00, 0x99, 0xff, 0x00, 0xcc, 0xff,
  0x00, 0xff, 0xff, 0x33, 0x00, 0xff, 0x33, 0x33, 0xff, 0x33, 0x66, 0xff,
  0x33, 0x99, 0xff, 0x33, 0xcc, 0xff, 0x33, 0xff, 0xff, 0x66, 0x00, 0xff,
  0x66, 0x33, 0xff, 0x66, 0x66, 0xff, 0x66, 0x99, 0xff, 0x66, 0xcc, 0xff,
  0x66, 0xff, 0xff, 0x99, 0x00, 0xff, 0x99, 0x33, 0xff, 0x99, 0x66, 0xff,
  0x99, 0x99, 0xff, 0x99, 0xcc, 0xff, 0x99, 0xff, 0xff, 0xcc, 0x00, 0xff,
  0xcc, 0x33, 0xff, 0xcc, 0x66, 0xff, 0xcc, 0x99, 0xff, 0xcc, 0xcc, 0xff,
  0xcc, 0xff, 0xff, 0xff, 0x00, 0xff, 0xff, 0x33, 0xff, 0xff, 0x66, 0xff,
  0xff, 0x99, 0xff, 0xff, 0xcc, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
[... 62 lines cut ...]

Results in errors like this:

Exception (2):
epc1=0x3fff34e8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff34e8 depc=0x00000000

ctx: sys 
sp: 3ffffdc0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff60:  00000000 40215a11 3ffed37c 3ffecfe0  
3fffff70:  0156d8c7 60000600 40214d4a 3ffea230  
3fffff80:  40215a56 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffecff0 3fffdc20 3ffec8bc 40201c31  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

martinayotte, 2015-08-24:
As discussed here http://www.esp8266.com/viewtopic.php?f=28&t=4688#p26962, I think the current ESP8266WebServer::sendContent_P() should be used for now only with text or html, since it doesn't have "size_t size" argument to tell the end of the content, it rely on NULL character at the end of the string. Maybe we will end up to add this "size_t size", this will allow binary too, like I did in my commit for WiFiClient::write_P() but still pending for merge.

martinayotte, 2015-09-06:
This commit waiting for PR merge should fix all the issues :
https://github.com/esp8266/Arduino/pull/753

igrr, 2015-09-30:
Released in staging

closed 2015-09-30 by igrr