esp8266/Arduino #600: DNS Server / Captive portal samples not working

luc-github, 2015-07-23:
I use 1.6.5-939-g033463a and the samples code for DNS Server / Captive portal are not working at least under windows 8.1 (Chrome/IE)  and Android (Chrome)
I open browser and type www.example.com 
I have also changed domain in sketch by www.esp8266.local and open it in browser
I always get error no DNS available
but get correct page if I type 192.168.1.1 

Are there some network configuration to be set that are not documented ?

igrr, 2015-07-24:
/cc @knovoselic

RobertCL, 2015-07-26:
I've been trying to use the DNSServer library as well from 1.6.5-947-g39819f0 and it works well from IOS, but not from Windows 7 or Android devices.  Similar issues to @luc-github, using the IP address works fine, but the dns name doesn't (except on IOS).

Using nslookup on Windows reports the correct IP for the dns name, but ping doesn't work and Chrome fails to resolve the name, so all we requests fail.

Using the code from this discussion fixes the problem for me:
http://www.esp8266.com/viewtopic.php?f=32&t=3618&start=40

knovoselic, 2015-07-27:
Sorry guys, I know about the problem, but was too busy with day job to fix it. I'll definitely fix it over the next weekend (if someone else doesn't do it before then :)).

luc-github, 2015-07-27:
It is great! Thanks a lot

luc-github, 2015-08-13:
Hi any update ?

julianh2o, 2015-08-13:
Use the link @RobertCL provided and find the comment ny SwiCago -  Sun Jun 28, 2015 11:23 am.

He lists a working version of the .cpp file, the .h file and the sample .ino file. I've just used this and it's working for me. Though you'll probably want to trim down the sample code a bit.

luc-github, 2015-08-13:
I use it - but found several variations in code in this page - all working but I do not master enough to know which one is best.
Also would be great to have correct version directly in module to avoid to apply workaround each time there is a new version.

luc-github, 2015-09-07:
Sorry I was busy lately I will test and feedback today - thanks

luc-github, 2015-09-07:
works like a charm - thanks a million

closed 2015-09-07 by luc-github