esp8266/Arduino #601: AnalogRead crashes and Reboots ESP8266

iisfaq, 2015-07-23:
I am using the latest "stable" code from today.

The following causes the ESP8266 to crash and reboot

```
void setup() {
  Serial.begin(115200);
  Serial.println("ESP8266");
}

void loop()
{
  long x = millis();
  for (int i = 0; i < 25000; i++)
    analogRead(A0);
  long y = millis();
  long  t = y - x;
  Serial.print("Time=");
  Serial.println(t);
}
````

Serial Output

ESP8266

ctx: cont 
sp: 3ffe9a60 end: 3ffe9cb0 offset: 01b0

>>>stack>>>
3ffe9c10:  40213be4 00000004 00000000 4020d111  
3ffe9c20:  4020d1b9 60000a00 3fff0d50 00000218  
3ffe9c30:  04390439 04280433 0428041f 04370435  
3ffe9c40:  3fffdc20 fe000000 fe000000 402039bc  
3ffe9c50:  00004146 000000d7 3fffdc20 3ffe9cdc  
3ffe9c60:  00000000 3fff3640 3ffe9d08 40202b84  
3ffe9c70:  40201180 00000000 3ffe9d08 40202bf0  
3ffe9c80:  3fffdc20 00000000 3ffe9cd4 40201d2d  
3ffe9c90:  00000000 00000000 3ffe9cd4 4020186e  
3ffe9ca0:  00000000 00000000 3ffe8c90 40100118  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

igrr, 2015-07-23:
This loop with 25000 iterations leads to watchdog reset. You need to call `yield()` or `delay` once in a while to keep the system running.

iisfaq, 2015-07-23:
I should have remembered that - sorry.....

On a side note and the reason for my strange loop

I find the AnalogRead quite slow.

The loop takes 9780ms running at 80Mhz

The same loop on an Arduino Pro Mini at 16Mhz takes 2800ms

Chris

igrr, 2015-07-23:
We're just calling system_adc_read function from SDK here, so not sure how this can be improved.
Keep in mind that ESP8266 ADC is implemented in software. There are a bunch of registers which are manipulated to control S&H, comparator, etc. to implement a SAR ADC. Plus, whenever you use ADC, SDK will temporarily disable some WiFi functions and re-enable them once conversion is complete.

iisfaq, 2015-07-23:
Would you therefore recommend using a dedicated hardware ADC chip? If performance is needed?  Also would that be more likely to be more accurate? Even if it is the same 10 bit resolution?

igrr, 2015-07-23:
Whether an external ADC is necessary depends on your application, but I'm sure that with an external SPI ADC you will likely be able to improve both throughput and accuracy. Even an I2C ADC might be faster.

closed 2015-07-24 by igrr