esp8266/Arduino #606: SPI overload cause to WDT reset

guyelg, 2015-07-24:
Hello 
I am trying to use library from  https://github.com/Nkawu/TFT_22_ILI9225 to operate ILI9255 display.
After changing the lines : #include <avr/pgmspace.h> to #include <pgmspace.h> in the library files, the program run just fine with the example sketch.
But when i try to write more than 10 line to lcd one after another i get wdt_reset.
I have try the same sketch with  Arduino Uno i it run just fine with more than 100 lines one after other.

igrr, 2015-07-24:
Try calling `yield();` once in a while if you have a long loop. Not only this will feed the watchdog, but also let WiFi run smoothly.

guyelg, 2015-07-27:
finally i found the problem!!!!
The problem is *Ticker*:
here example that make wdt_reset:

```
#include "TFT_22_ILI9225.h"
#include <Ticker.h>
#include <SPI.h>
#define TFT_RS  2
#define TFT_RESET  15
TFT_22_ILI9225 tft = TFT_22_ILI9225(TFT_RESET, TFT_RS, 0, 0);
Ticker flipper;
 
void DrawTest() {
   for(int i=1;i<100;i++) {
    tft.drawText(10+i, 10, "hello!");
     yield();
  }
}

void setup() {
  tft.begin();
  Serial.begin(115200);
   tft.setFont(Terminal6x8);
   flipper.attach(1,DrawTest );
}

void loop() {
  //DrawTest();
}
```
And another one without our friend Ticker:

```
#include "TFT_22_ILI9225.h"
#include <Ticker.h>
#include <SPI.h>
#define TFT_RS  2
#define TFT_RESET  15
TFT_22_ILI9225 tft = TFT_22_ILI9225(TFT_RESET, TFT_RS, 0, 0);
Ticker flipper;
 
void DrawTest() {
   for(int i=1;i<100;i++) {
    tft.drawText(10+i, 10, "hello!");
     yield();
  }
}

void setup() {
  tft.begin();
  Serial.begin(115200);
   tft.setFont(Terminal6x8);
   // flipper.attach(1,DrawTest );
}

void loop() {
   DrawTest();
}
```
The second one work without crash.
You can try that without hardware (i connected only with MOSI line)

igrr, 2015-07-27:
>It is currently not recommended to do blocking IO operations (network, serial, file) from Ticker callback functions. Instead, set a flag inside the ticker callback and check for that flag inside the loop function.

igrr, 2015-07-28:
Okay, I forgot that yield() is now checking where it is called from, so it will not work from a Ticker callback. So the original comment from Readme applies — don't do any length stuff in Ticker, instead set a flag and check for it in the main loop.

closed 2015-07-28 by igrr