esp8266/Arduino #609: ESP.deepSleep function behavior

mhanuel26, 2015-07-24:
I notice the following behavior,

when calling the function ESP.deepSleep(sleeptimeout, WAKE_RF_DEFAULT);

it cannot be use in a closing function call or if clause,  you can try the blink example, but call the function blink directly from loop.
Even you cannot put the instruction ESP.deepSleep(tmp, WAKE_RF_DEFAULT); like this 

void loop(){
       print("going to sleep");
       ESP.deepSleep(tmp, WAKE_RF_DEFAULT);
}

The sleep never execute. Instead the esp print to serial at max speed...

My workaround is to do this 

void loop(){
       print("going to sleep");
       ESP.deepSleep(tmp, WAKE_RF_DEFAULT);
}

giannoug, 2015-07-26:
I'm troubleshooting the same problem. Which board are you using?

Also, the second example you gave is exactly the same as the first one.

mhanuel26, 2015-07-26:
giannoug,

I have updated the snippet.

torntrousers, 2015-07-30:
deepSleep seems to take a while to happen after the making the call, about 40 or so milliseconds in tests i've done, and in that time the sketch continues on executing, and it gets confused if in that time another deepSleep call is executed. So i always add the delay(1000) right after the deepSleep call too.

I suppose it could be handy if the delay(1000) call was inside the ESP.deepSleep function itself.

igrr, 2015-07-30:
Or better yet, to call `esp_yield();` inside deepSleep (like we now do for `ESP.restart`).

garrettmoon, 2015-07-31:
Calling yield() didn't work for me, only delay(1000).

igrr, 2015-07-31:
`esp_yield()`, not `yield()`.

garrettmoon, 2015-07-31:
Aha, I'm new to this. Is yield not the correct function to call to, uh, yield? That's more of a general question tangentially related to this.

igrr, 2015-07-31:
That's not the best function naming, admittedly.
`yield()` will return control to OS and then start sketch.

`esp_yield()` is an internal function which will return control to OS. To resume sketch execution again there is another internal function, `esp_schedule()`. These two functions are normally used inside the core.

garrettmoon, 2015-07-31:
Cool, thank you for the explanation :)

closed 2015-09-30 by igrr