esp8266/Arduino #612: Stable version libm.a compilation error

Aedvin, 2015-07-25:
Compiling project using libm.a throws: 
packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o):(.literal+0xf8): undefined reference to `__ieee754_sqrt'

packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o): In function `__ieee754_pow':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/e_pow.c:164: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status

Some hints how to deal with this? Thanks

krulkip, 2015-07-26:
I get same problem.

Arduino: 1.6.5 (Windows 7), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)"

j:/arduino-1.6.5-r2/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o):(.literal+0xec): undefined reference to `__ieee754_sqrt'
j:/arduino-1.6.5-r2/portable/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o): In function `__ieee754_pow':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/e_pow.c:164: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status

forkineye, 2015-07-28:
Experiencing this as well / windows 8.1 and Arduino 1.6.5-r2
Arduino: 1.6.5 (Windows 8.1), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200"

```c:/users/xxxx/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o):(.literal+0x104): undefined reference to `__ieee754_sqrt'```
```c:/users/xxxx/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o): In function `__ieee754_pow':```
```d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/e_pow.c:164: undefined reference to `__ieee754_sqrt'```
collect2.exe: error: ld returned 1 exit status

igrr, 2015-07-28:
I think the workaround is to delete the downloaded package (it's somewhere
in the user folder on Windows) as well as downloaded toolchain archive.
Arduino doesn't download the file again even though the size and checksum
had changed.

I will bump toolchain version for the next release.

forkineye, 2015-07-28:
@igrr I just tried that out, removing the ESP8266 board from the Arduino IDE and then removing all ESP xtensa references locating the ```users\<user>\appdata\arduino15``` packaging and staging directories.  Same issue when trying to call ```pow()```.

Example from a gamma table generation loop: ```Serial.print((int)(255 * pow((i / 255), gamma)));```

forkineye, 2015-08-04:
I'm able to recreate this issue, but only with floats inside of loops.  Code to re-create is here - https://gist.github.com/forkineye/d05909c5c0184c7300ae. The toolchain is ```1.20.0-26-gb404fb9```.  Toolchain archive is ```win32-xtensa-lx106-elf-gb404fb9.tar.gz``` with a MD5 hash of ```7e6cf4fb104022ddee0b9ef716232439```

```arduino
void setup() {
    Serial.begin(115200);
    Serial.println("");

    float x = 2.0; 
    float y = 2.2;

    int i;
    for(i = 0; i < 10; i++) {
        Serial.println(pow(i, x));
        Serial.println(pow(i, y));    // Fails to compile
    }
    Serial.println(pow(i, y));        // Compiles fine, 10 ^ 2.2
}

void loop() {
 
}
```

krulkip, 2015-08-08:
This seems to refer to a fix for this issue, but not sure how to try
https://github.com/jcmvbkbc/crosstool-NG/commit/b404fb9ec3a5c59e6b4d90741d39f7d1937431cd

igrr, 2015-08-09:
> jcmvbkbc/crosstool-NG@b404fb9

Windows toolchain is not built using crosstool-NG, so probably not exactly
relevant.

krulkip, 2015-08-09:
I installed this libm.a file replacing the xisting one in 1.20.0-26-gb404fb9/xtensa-lx106-elf/lib/ and the error was solved.
https://files.gitter.im/esp8266/Arduino/Abqa/libm.a.tbz

Testato, 2015-08-22:
same problem here, i use the Stable version on Win7-64

Is it solved on Staging Version ?

locchi93, 2015-08-30:
same problem on windows using the current Stable version and Arduino ide 1.6.5

Testato, 2015-08-30:
@krulkip
Thanks, whit your lib there is no problem

erhome00, 2015-09-04:
Testato or igrr,

can you please tell me how exactly I use that corrected LIB? I have the same problem and am not able to solve it...

tim-in-oakton, 2015-09-06:
Same issue here, trying to port over a module that uses math - at compile I get:

c:/users/timm/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o):(.literal+0x4): undefined reference to `__ieee754_sqrt'
c:/users/timm/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o): In function `sqrt':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/w_sqrt.c:63: undefined reference to `__ieee754_sqrt'
c:/users/timm/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_asin.o): In function `__ieee754_asin':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/e_asin.c:104: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status
Error compiling.

I'm on 1.6.5 of the IDE, and did a remove/replace of the ESP8266 board support package - no change in displayed error message.  Windows dev environment.

Any idea how to mitigate this?
tim

Testato, 2015-09-06:
@erhome00 
I followed exactly what krulkip wrote

worked on Win7-64

erhome00, 2015-09-07:
Thanks for the replies. It was my mistake...got to use the Stageing version and not the Stable version. The change was not yet migrated to the Stable...

Now everything works just fine!

Thanks for that nice fix!

aniston, 2015-09-27:
Just to confirm, changing the libm.a with the link supplied by Krulkip and the error with - undefined reference to `__ieee754_sqrt' is no more, code compiles and works.

I had to change the file in this folder ::
C:\Users\aniston\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9\xtensa-lx106-elf\lib

cuillerj, 2015-09-28:
I got the same problem and did exactly what krulkip wrote.
And it works
Thanks

QuadTinnakon, 2015-09-29:
c:/users/n_sam/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o):(.literal+0x4): undefined reference to `__ieee754_sqrt'
c:/users/n_sam/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o): In function `sqrt':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/w_sqrt.c:63: undefined reference to `__ieee754_sqrt'
c:/users/n_sam/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_asin.o): In function `__ieee754_asin':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/e_asin.c:104: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status
Error compiling.

What causes.
Thanks.

mytot, 2015-10-25:
while i am downloading krulkip file, it was of 165KB, and previoius one is 2MB, and after replacing the file i am getting below error.

Arduino: 1.6.5 (Windows 8.1), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)"

c:/users/shann/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a: file not recognized: File format not recognized
collect2.exe: error: ld returned 1 exit status
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Help me in resolving the issue

aniston, 2015-10-25:
@mytot I think you have the wrong path, try downloading it into c:\users\shann\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9\xtensa-lx106-elf\lib

Testato, 2015-10-26:
collect2.exe: error: ld returned 1 exit status
It is an arduino 1.6.5 bug on some cpu and/or OS
Try to delete collect2.exe
And if it do not work ask help on arduino forum

DiegoDVG, 2015-10-27:
hi!

this is my problem

Arduino:1.6.5 (Windows 8.1), Placa:"Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)"

c:/users/diego/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o):(.literal+0x8): undefined reference to `__ieee754_sqrt'
c:/users/diego/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o): In function `sqrt':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/w_sqrt.c:63: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status
Error de compilación

  This report would have more information with
  "Show verbose output during compilation"
  activala desde Archivo > Preferencias

win 10 pro

I try to use MPU6050

nabkey, 2015-10-27:
@mytot The reason you're having the issue is that @krulkip posted a file that's in an archive - you have to extract it first, instead of just renaming the archive file.

jwschull, 2015-11-10:
Got a couple issues...  NOTE: I'm using the build from 11/10 (1.6.7)
1) Getting this error:Invalid library found in C:\Users\JerryHB\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266: C:\Users\JerryHB\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266

If I go there (i.e."C:\Users\JerryHB\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266") I see only 1 directory "Examples" and Nothing else.  No library.

2) per nabkey's suggestion I extracted the archive and replaced it.  (i.e. "C:\Users\JerryHB\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9\xtensa-lx106-elf\lib"   The file libm.a went from 2202KB -> 1027K.  

It now compiles, But I'm still getting the error #1 above and a warning (i.e. "Warning: platform.txt from core 'ESP8266 Modules' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.")

homerruma, 2015-11-12:
@krulkip
Thanks, whit your lib there is no problem

Hopperpop, 2015-11-12:
The lib from krulkip works for all functions I need, except "fmod". It gives this error:

c:/users/xxx/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x4): undefined reference to `__ieee754_remainder'
c:/users/xxx/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':
e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'
collect2.exe: error: ld returned 1 exit status

RiseUp26, 2015-11-15:
Thank you, @krulkip. After replacing the file as you suggested, it now works for me, too.

Testato, 2015-11-26:
I tried the actual stable version of the core and this problem is not present.are you tried ? I think this issue can be closed

jwschull, 2015-11-26:
I got it working..  Thanks

bmitov, 2015-11-30:
I just installed the stable version as well as rc2 - tested both.
I have the same problem:
:/users/mitov/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'

c:/users/mitov/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':
e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'
collect2.exe: error: ld returned 1 exit status

I also tested by replacing the libm.a file posted by @krulkip but the error persists.
This bug is not fixed at least in the distributions available from here:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

aquarat, 2015-12-30:
I'm also experiencing this issue, specifically when trying to use Adafruit's GPS parsing library.

ketancool203, 2016-01-06:
Hey Guys ,
I wanted to connect esp8266 with MPU6050. Such that it sends data to the particular ip address.Please help and provide me how to do it.

bmitov, 2016-01-08:
Moving here from #1385. Is there any solution for this? This is a very serious issue, and a serious roadblock, that seems to affect a number of functions.
Any workarounds?

igrr, 2016-01-08:
The missing function is not enabled in libm build. libm relies on the one
provided by libgcc, but xtensa libgcc doesn't provide one. So newlib has to
be patched, and the implementation provided by newlib enabled.

As a workaround, you can copy the source of this function from newlib and
add it to some .c file inside your sketch (if you are shipping a library,
than add this .c file to your library). You can also mark this function
with "weak" attribute so that everything will still work once upstream libm
is fixed.

Hopperpop, 2016-01-08:
I use this as workaround. It's simple and it works for me.

```c++
double floatmod(double a, double b)
{
    return (a - b * floor(a / b));
}
```

bmitov, 2016-01-08:
Thank you igrr!
Where I can find the source of the newlib for the ESP8266 ? I can open request for this with their team. Alternatively a file containing the function as source can be temporary included in the ESP8266 libraries, until newlib is fixed.
I am not sure what is the best approach considering that this has wide spread effect on many libraries, and should be fixed.
For now I will add conditional defines in the Visuino libraries to handle the ESP8266 with the workaround suggested by Hopperpop(Thank you for the workaround, it seems to work well so far!).

igrr, 2016-04-19:
#1752 addresses `fmod` issue.

umarjunaid, 2016-06-29:
Thanks to @krulkip and aniston for there comments which solve my problem

@krulkip said:
This seems to refer to a fix for this issue, but not sure how to try
jcmvbkbc/crosstool-NG@b404fb9

aniston said:
Just to confirm, changing the libm.a with the link supplied by Krulkip and the error with - undefined reference to `__ieee754_sqrt' is no more, code compiles and works.

I had to change the file in this folder ::
C:\Users\aniston\AppData\Roaming\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9\xtensa-lx106-elf\lib

Arduino: 1.6.5 (Windows 7 x64), Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS)

umarjunaid, 2016-06-29:
sorry 

@krulkip said:
I installed this libm.a file replacing the xisting one in 1.20.0-26-gb404fb9/xtensa-lx106-elf/lib/ and the error was solved.
https://files.gitter.im/esp8266/Arduino/Abqa/libm.a.tbz

anilkunchalaece, 2016-08-11:
i am Facing the Error
`
c:/users/med/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'

c:/users/med/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':

e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

I Did replace the file libm.a
But I am facing the Same Error.

I am Trying to use the Adafruit Ultimate Gps Library with NodeMCU

dkgrieshammer, 2016-08-12:
thanks Hopperpop, thats it :)

kjhulin, 2017-04-15:
I am having this same error, using Adafruit Ultimate GPS library with the ESP8266WiFi

zandoga, 2017-06-15:
I am not sure if it is related, I replaced libm.a with no success:

Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 9600, 4M (3M SPIFFS)"

Archiving built core (caching) in: C:\Users\luli\AppData\Local\Temp\arduino_cache_44067\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_9600,FlashSize_4M3M_b19f53dec49a63a208bb25ec6cc9c30f.a
c:/users/luli/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'

c:/users/luli/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':

e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'

collect2.exe: error: ld returned 1 exit status

mjs513, 2017-12-18:
I am getting a similar error for fmod:
`Archiving built core (caching) in: C:\Users\CYBERP~1\AppData\Local\Temp\arduino_cache_461718\core\core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____6f0ffaa82fb8685893767509385827e5.a
c:/users/cyberpalin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'

c:/users/cyberpalin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':

e_fmod.c:(.text+0x48): undefined reference to `__ieee754_remainder'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.
`
I am IDE1.8.5, Windows 10, stable release 2.3.0.  Is there an easy fix for this or do I have to wait for a new release?

Thanks
Mike

devyte, 2017-12-19:
@mjs513 Staged for release means the issue was found before 2.3.0, and has been already fixed in master branch. So try latest git.

closed 2018-01-02 by igrr

KevinLod, 2018-05-11:
@igrr  
Im having a problem with my IDE 
exec: "C:\Users\Doesn't matter\AppData\Local\Arduino15\\packages\\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++": file does not exist

would you please help me fixing this issue
I've tried many ways to fix it but i couldn't

commit 5c410c7c (2015-08-10): Update board manager package build

refactor shell script a bit and update toolchain version to fix #612