esp8266/Arduino #613: Error installing esptool

hrshovon, 2015-07-25:
esptool download error.
[image]

tytower, 2015-07-29:
Yes I confirm this seems to be a bug and a few others have it too . reportedly only on Linux 32 bit

tytower, 2015-07-29:
There is no /releases/  directory

igrr, 2015-07-30:
Whoops. We really don't have a linux-32 build of esptool for this version. Sorry guys.
@hrshovon since you are running linux 32bit, could you please help everyone by building esptool and providing an archive?
```bash
git clone https://github.com/igrr/esptool-ck.git
cd esptool-ck
git checkout 0.4.5
make dist
```
This will create a tar.gz archive which you can upload somewhere, and i'll make it available for download.
Thanks.

hrshovon, 2015-07-30:
Done,hope it helps :)
https://www.dropbox.com/s/hirp6j1udt5g0c5/esptool-0.4.5-linux32.tar.gz?dl=0

igrr, 2015-07-30:
Thanks!
I have pushed new staging package, version number is 1.6.5-988-gfd1d9da.
http://arduino.esp8266.com/staging/package_esp8266com_index.json
Please give it a go.

FredAnoniem, 2015-07-30:
Thanks guys! It works on my Linux 32.

tytower, 2015-07-30:
me too Thanks

closed 2015-08-05 by igrr