esp8266/Arduino #614: Takes a while to return from webserver handler.

Duality4Y, 2015-07-25:
I am having a peculiar problem.
I have some code that runs in the loop.
And I have some code that runs when a handler is called through the webserver.

It seems that it takes a while for the handler to return to the loop,

And i am wondering why this is?

me-no-dev, 2015-07-25:
currently all network is handled in the loop, even though it might look as callback.
what is your handler doing to take such long time?
what is the request that takes time?

Duality4Y, 2015-07-29:
i have temporarily fixed it by calling that code directly from the handler.
my handler looks like:
```c++
void handleStripControl()
{
  String debug = "protocol debug: <br>";
  Serial.println("received data");
  if(server.args())
  {
    for(int i = 0;i < server.args(); i++)
    {
      debug += server.argName(i) + ": " + server.arg(i) + "<br>";
    }
    stripcontrol.pincode = server.arg("pin").toInt();
    stripcontrol.effect = server.arg("effect").toInt();
    stripcontrol.brightness = server.arg("brightness").toInt();
    stripcontrol.varZero = server.arg("var0").toInt();
    stripcontrol.varOne = server.arg("var1").toInt();
    stripcontrol.varTwo = server.arg("var2").toInt();
    stripcontrol.changed = true;
    debugPrintStripControl();
  }
  handleStrips();

  String htmlStripControl = 
  "<!DOCTYPE html>"
  "<html>"
  "<title> esp-light page. </title>"
  "<head>" + css + "</head>"
  "<body>" +
  divHeader +
  divNav +
  "<h1>StripControl</h1><br>" +
  debug +
  divFooter + 
  "</body>"
  "</html>"
  ;
  server.send(200, "text/html", htmlStripControl);
}
```
The server client handling is done in the loop.
And some other function's that shouldn't take long to run.

igrr, 2015-07-30:
That's the right way to handle it. ```server.handleClient()``` won't return until all the data has been sent and ACKed. So if you need to do something really quick, do it from the handler.

closed 2015-07-30 by igrr