esp8266/Arduino #617: <ESP8266WiFi.h> errors

Smart-Dragon, 2015-07-26:
Hi!
Help me, please!
I am using <ESP8266WiFi.h>
After compile:

In file included from C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:32,
                 from wifi_dimmer.ino:1:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h: In member function 'size_t WiFiClient::write(T&, size_t)':
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: 'unique_ptr' is not a member of 'std'
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: expected primary-expression before '[' token
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: expected primary-expression before ']' token
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: there are no arguments to 'buffer' that depend on a template parameter, so a declaration of 'buffer' must be available
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:113: error: 'buffer' was not declared in this scope
In file included from wifi_dimmer.ino:1:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h: At global scope:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:240: error: ISO C++ forbids initialization of member '_smartConfigStarted'
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:240: error: making '_smartConfigStarted' static
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:240: error: ISO C++ forbids in-class initialization of non-const static member '_smartConfigStarted'

igrr, 2015-07-26:
Could you please post full verbose compilation log (enable it in Arduino
Preferences)?

Smart-Dragon, 2015-07-26:
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -IC:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\variants\standard -IC:\Program Files (x86)\Arduino\libraries\SPI -IC:\Program Files (x86)\Arduino\libraries\ESP8266WiFi -IC:\Program Files (x86)\Arduino\libraries\WiFi C:\Users\SMARTD~1\AppData\Local\Temp\build2533683805112170500.tmp\CyberLib.cpp -o C:\Users\SMARTD~1\AppData\Local\Temp\build2533683805112170500.tmp\CyberLib.cpp.o 
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avr-g++ -c -g -Os -Wall -fno-exceptions -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -MMD -DUSB_VID=null -DUSB_PID=null -DARDUINO=105 -IC:\Program Files (x86)\Arduino\hardware\arduino\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\variants\standard -IC:\Program Files (x86)\Arduino\libraries\SPI -IC:\Program Files (x86)\Arduino\libraries\ESP8266WiFi -IC:\Program Files (x86)\Arduino\libraries\WiFi C:\Users\SMARTD~1\AppData\Local\Temp\build2533683805112170500.tmp\wifi_dimmer.cpp -o C:\Users\SMARTD~1\AppData\Local\Temp\build2533683805112170500.tmp\wifi_dimmer.cpp.o 
In file included from C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:32,
                 from wifi_dimmer.ino:2:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:28:18: warning: memory: No such file or directory
In file included from C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:32,
                 from wifi_dimmer.ino:2:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h: In member function 'size_t WiFiClient::write(T&, size_t)':
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: 'unique_ptr' is not a member of 'std'
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: expected primary-expression before '[' token
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: expected primary-expression before ']' token
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: there are no arguments to 'buffer' that depend on a template parameter, so a declaration of 'buffer' must be available
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:106: error: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/WiFiClient.h:113: error: 'buffer' was not declared in this scope
In file included from wifi_dimmer.ino:2:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h: At global scope:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:240: error: ISO C++ forbids initialization of member '_smartConfigStarted'
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:240: error: making '_smartConfigStarted' static
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi/ESP8266WiFi.h:240: error: ISO C++ forbids in-class initialization of non-const static member '_smartConfigStarted'

igrr, 2015-07-26:
avr-g++ ?

Looks like you are trying to use this library while building for an AVR
board. Make sure you have one of ESP8266 boards selected.

Smart-Dragon, 2015-07-26:
надеюсь ты говоришь по русски) Если где-то затуплю прости, пару дней назад только платка пришла, разбираюсь.
Так это получается эта библиотека работает только внутри ESP8266 ?
А не знаешь какие нибудь библиотеки которые позволяют работать Ардунике вместе с ESP8266, как со стандартными модулями WiFi?

igrr, 2015-07-27:
@Smart-Dragon  https://github.com/sparkfun/SparkFun_ESP8266_AT_Arduino_Library

closed 2015-07-28 by igrr