esp8266/Arduino #621: ESP.restart not work

asetyde, 2015-07-28:
ESP.restart not work, i've tried many times to reboot from network , at the end of code ESP.restart(); but only this write and nothing do.

it's not good thing that board can not reboot

`<if (packetBuffer[0] == key && packetBuffer[1] == 0X31 ) //  T3 0x05 0x31 0x00 0x00   0x00 0x00 0x00 0x00   OK
    {
    	if(debug==true){Serial.println(F("REBOOT"));}
		byte message[5] = { 0x05, 0x99, 0x99, 0x99, 0x0A };
		sendUDPOp(remoteip, message, 15);
		ESP.restart();
    	
    }`

asetyde, 2015-07-28:
sometimes work , other not and write this :

' ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

asetyde, 2015-07-28:
if i use while(1); 

REBOOT

ctx: cont 
sp: 3ffeb7d0 end: 3ffeba00 offset: 01b0

>>>stack>>>
3ffeb980:  3fff3105 00000013 3ffeba60 4020b501  
3ffeb990:  3ffea92c 40101653 3ffe9170 9301a8c0  
3ffeb9a0:  9905dc20 3f0a9999 40201c4e 3ffea9e0  
3ffeb9b0:  4020153a 3ffe9170 9301a8c0 4020152f  
3ffeb9c0:  00000000 00000000 00000016 3ffeba2c  
3ffeb9d0:  3fffdc20 00000000 3ffeba24 40203910  
3ffeb9e0:  3fffdc20 00000000 3ffeba24 40201cca  
3ffeb9f0:  00000000 00000000 3ffea9e0 40100378  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

asetyde, 2015-07-28:
can i try with pin manual reboot ? and how can i do ?

tontito, 2015-07-28:
Gpio0 will 10k pullup resistor.

asetyde, 2015-07-28:
hi, thanks to reply , but i now that nodemcu 1.0 or 12e v2 not needed ?  and if yes ..  how can do 

gpio0 -10k- ground ?

asetyde, 2015-07-28:
Seems that problem there is when i flash and after call this function ... if i remove usb and reattach to usb power or eternal reboot work normally .. something wrong after a flash to modify code... 

@igrr  who can i ask to resolve ?
@tontito , not work on this board

tontito, 2015-07-28:
i don`t know nodemcu board. Gpio0 will 10k pullup resistor to VCC (3.3v) Generic ESP module

asetyde, 2015-07-29:
@tontito   design reference , seems not be necessary 
http://smartarduino.gitbooks.io/user-manual-for-esp-12e-devkit/content/schematics_for_esp-12e_devkit.html

asetyde, 2015-07-29:
but not says about serial part block to reboot .. i think depends from firmware . Wait to see with update sdk

closed 2015-08-17 by asetyde