esp8266/Arduino #630: CRC doesn't match. File is corrupted.

Kiran2015, 2015-07-29:
Using Arduino IDE 1.6.5
Using Additional Boards Manager URLs : http://arduino.esp8266.com/stable/package_esp8266com_index.json

I get the message "CRC doesn't match. File is corrupted." when I try to install esp8266 package from Boards Manager.

Please guide me on how to resolve this.

sticilface, 2015-07-29:
delete this folder.

/Arduino15/staging/packages

on mac it is located in /Users/xxx/Library/Arduino15/staging/packages

Kiran2015, 2015-07-30:
sticilface, I tried deleting /Arduino15/staging/packages, but I'm still getting the same message.

When I went inside /Arduino15/staging/packages folder, I found that the file esp8266-1.6.5-947-g39819f0.zip is empty

igrr, 2015-07-30:
Looks like you're having some download issues. Perhaps you can try downloading this file manually via the browser, and placing it into the staging/packages/ folder?

lrmoreno007, 2015-07-30:
I like the portable method to make a clean installation of every new version: with Arduino closed, create a folder named "Portable" in the Arduino folder, run Arduino and you have a new clean config. Now you must go to Preferences again, input the json, and download in Boards Manager, etc everything like a new installation.

When you want, you can rename or delete the Portable folder and you will have the original installation. (Sometimes you must restart two times)

With this method you can have many config folders (Portable1, Portable2, PortableStaging, PortableNightly, etc) and you are working ALWAYS with the folder named Portable. With this method the files aren't in the folder /Arduino15/, they are in the Portable folder. /Arduino15/ is the folder for no one Portable config.

It's a very very good method.

Good luck.

Kiran2015, 2015-07-30:
igrr, you are correct. There was some problem with the internet connection. Now I connected with a different internet connection and was able to install the package.

sticilface, igrr, lrmoreno007, Thank you for your responses.

closed 2015-07-30 by Kiran2015

vazquezjm, 2015-08-30:
I'm experiencing the exact same problem on Linux (Mint 32bits). Any advice will be greatly appreciated

jakorten, 2016-04-09:
I am writing my own board definitions etc for custom boards. Seems that staging / packages contains the packages as temp files. Deleting my own (older) tar.gz file solved the CRC issue for me.

AlfhaOmega, 2019-05-21:
try download https://www.arduinolibraries.info/libraries/arduino-json version what you need, 
then place the .zip file in C:\Users\XXX\AppData\Local\Arduino15\staging\libraries, 
then redo your installation in to arduino,

it's work for me!!

Kavin0511, 2019-10-20:
Sometimes you won't be able to find packages at that time you have to visit your project folder inside Arduino folder and then   there will be a file near data folder then you have to delete it and then again try to install

danishahmed219, 2020-02-22:
Try to download another version. perhaps, it will work..

kylierios, 2021-07-15:
Try re-downloading the newest version of Arduino IDE

Mani230496, 2023-10-10:
CRC doesn't match. File is corrupted
i have faced the same issue but its has solved the problem is internet concetivity , so i do wired connection in my system then downlaod is completed.

ARI-byte, 2023-12-27:
Just paste this in the Additional Boards Manager Urls
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://dl.espressif.com/dl/package_esp32_index.json