esp8266/Arduino #635: webserver in AP crashes with lmac.c

Duality4Y, 2015-07-31:
this is what i get on the serial port when trying to run a webserver in acces point mode.
```
..........
192.168.4.1
done setting up pins, and WifiMode.
done setting up server
add 1
aid 1
station: 00:1c:26:35:d2:fb join, AID = 1
lmac.c 599

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

I don't know if this is an issue with the code, or with the esp or the network.
any ideas ?

Duality4Y, 2015-07-31:
the code can be found here:
https://github.com/Duality4Y/Sketchbook/tree/master/sketchbook/webserver-esp

Duality4Y, 2015-08-01:
okay when i comment these it runs fine when i just go to root: (.ino)
```c++
  server.on("/ledsettings", handleLedSettings);
  server.on("/wifisettings", handleWiFiSettings);
  server.on("/stripcontrol", handleStripControl);
```
but clicking a link that goes to those handlers it crashes with
```
add 1
aid 1
station: 00:1c:26:35:d2:fb join, AID = 1
lmac.c 599

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

Duality4Y, 2015-08-01:
oke, Added pull-ups on the pins doesn't help.

closed 2015-08-01 by Duality4Y

Duality4Y, 2015-08-01:
turns out it's a library outside of the esp that makes it crash. (NeoPixelbus)

sticilface, 2015-08-01:
This might make two of us. I'm having an epic time with lmac.c   Using web server and neopoxelbus!

Duality4Y, 2015-08-01:
I traced my issue back to strip.show();
and it only crashes in access point mode.