esp8266/Arduino #636: Really need help. Boards breaks after some boots

emilianocasas, 2015-07-31:
Hi, i am having a huge problem and I dont know how to solve it. I have crushed 3 esp07 boards. 
Instead of using an uart adapter, I use an arduino uno to boot the esp07 (I remove the arduino's processor previously) and a mb102 protoboard power supply for the esp (The power supply is
the following: http://www.ebay.com/itm/MB102-Breadboard-Power-Supply-Module-3-3V-5V-For-Arduino-Solderless-Breadboard-/170886733144?hash=item27c9a49d58)  
Connections are 
RX --- RX (arduino) (No need to connect RX-TX )
TX ---TX (arduino)
GND --- GND
CHPD --- VCC (mb102)
VCC (ESP) --- VCC (mb102)
GPIO0 --- GND
GPIO15 --- GND
GPIO2 --- VCC (mb102)

I join the arduino's and the esp's gnd with the mb102's gnd.
At first, it works great, but the board gets really warm. After several uses, the board red light becomes weak. Then, when I connect the esp with the mb102, mb102's light becomes weak also, and the esp won't work. This happened also with esp01 and esp12.
Does anyone knows what can possibly be wrong? Maybe the mb102? The connections between esp and arduino shoul be connected with resistors?

Thanks for your time, and sorry for my english.

duncan-a, 2015-07-31:
- Ensure that the MB102 is 3.3V (jumper setting and measured voltage)
   - Use pullup resistors (10K) for CH_PD, GPIO0, GPIO2 and RESET lines -
   do not connect directly to Vcc 3.3V - and connect GPIO0 to GND only to
   enter upload mode, at other times it must be pulled high to boot from the
   ESP.
   - Use pulldown resistor (10K) for GPIO15 to GND if it is not hard-wired
   on the module.
   - The inputs of the ESP are '5V Tolerant' but it is better if they
   receive 3.3V logic signals via a level-shifter or from another 3.3V source.

I'm not familiar with the technique of using an Arduino in place of a
UART-Logic converter, but I would have expected 'Rx to Tx' and 'Tx to Rx'
connections, but I'm sure you've researched that.

I use an 'Adafruit FTDI Friend' set to 3.3V, with the same PSU as you are
using and the ESPs have never felt even slightly warm. I suspect that you
may have killed your ESPs.

jeffryr, 2015-07-31:
Ditto what duncan-a said plus
- What are you powering the MB102 with?  My experience has been that anything more than 15V or so will fry the regulators.
- Triple check your wiring
- If you want to program it with an Arduino, use a 3.3V 8MHz Pro mini to eliminate the need for level shifters

Links2004, 2015-07-31:
The Pro Mini not has an USB port :) 
GPIO2 to VCC may kill the board, GPIO2 is a output by boot mode ( alternative TX )
more info here:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#generic-esp8266-modules

emilianocasas, 2015-07-31:
Thanks a lot, I will try to implement level shifters

duncan-a, 2015-07-31:
The pullup/pulldown resistors are more important, and an easier first move.

closed 2015-09-30 by igrr